martinpitt/umockdev #47: Don't abort when Xorg cannot be launched

hadess, 2015-04-29:
umockdev 0.8.8 on Fedora:

```
$ Xorg -config xorg-dummy.conf -logfile /tmp/Xorg.log.24CDYX :5
/usr/libexec/Xorg.wrap: Only console users are allowed to run the X server
```

It should skip the X dependent tests, just like it skips gphoto2 tests if it's not available.

martinpitt, 2015-05-22:
What is the precise error message that you see? I suppose that you *don't* get "cannot call Xorg: <some error>", as it's spawned asynchronously. I assume it times out waiting for the X socket,  i. e. "Xorg failed to start up: please ensure you have the X.org dummy ...". Is that right?

martinpitt, 2015-05-22:
I tested this with `ln -s /bin/false /usr/local/bin/Xorg` and get the expected error from above.

Fixed in https://github.com/martinpitt/umockdev/commit/e03ef650fa556bbfdec972974d08f3509f425175 , it now looks like:

    umockdev-run/integration/input-touchpad: SKIP: Xorg failed to start up; please ensure you have the X.org dummy driver installed, and check the log file: /tmp/Xorg.log.MTQIYX
    OK

closed 2015-05-22 by martinpitt

hadess, 2015-05-22:
You still need me to do something here?

martinpitt, 2015-05-22:
If that fix (included in today's 0.8.9 release) works for you (I'm 95% sure it does), then it's all good.