martinpitt/umockdev #48: using umockdev to record CAN device

koenlek, 2015-05-20:
Hi all,

I have a CAN-bus radar that I want to record data from. It is connected through a CAN to usb adapter and appears as a /dev/ttyUSBx. I would like to make some low level recordings outdoors and use that to write a driver for it offline, using a simulated dev based on that recording. I think I should be able to achieve this with umockdev, but so far I haven't been able to get it right. 

To clarify my case a bit better: I can start everything, such that running this ```screen /dev/canusb 500000``` (the radar has a baud rate of 500kbit/s) echo's the messages send over the CAN bus. So actually, it is just a serial device to which data is written at a high frequency (I'll discard sending data over the serial device for now, i.e. sending messages to the radar).

I think I should be able to record data being published (the CAN messages) on the serial device using umockdev and than replay it for offline development. But how should I do that? When using ```umockdev-record``` I seem to need to add a command as well, why? what should I put there? and should I record using the -i or -s option? Both do not seem to output any recording file... 

Any hints would be very much appreciated!

Best,
Koen

martinpitt, 2015-05-22:
Hello,

You can't "attach" umockdev-record to an already running program. You have to start whatever talks to the device "under" umockdev-record, i. e. something like 

    umockdev-record -s /dev/canusb=/tmp/canusb.script -- screen /dev/canusb 500000

I assume you want to Control-C that after a few seconds, depending on how much data you want to record.

This is described at https://github.com/martinpitt/umockdev#command-line-record-and-replay-tty-devices.

koenlek, 2015-06-16:
Hi Martin,

I found another way around it, I just dumped all the data to a csv and gave my 'CAN driver' two modes, one from csv and one directly talking with the device. 

Thanks for your help!

closed 2015-06-16 by koenlek

masterkorp, 2015-06-16:
@koenlek while not related to this issue, can you elaborate how did you solved the problem ?

koenlek, 2015-06-16:
I was able to use candump from can-utils to read out and dump the device. I downloaded the source code [here](https://github.com/linux-can/can-utils/) and adapted it to make my own csv dump tool. My [ROS](http://ros.org/) driver then reads this csv and deserializes the messages (which I had to implement manually).