martinpitt/umockdev #50: src/libumockdev-preload.c:1381: possible use of dead pointer ?

dcb314, 2016-03-02:
[src/libumockdev-preload.c:1381]: (error) Dead pointer usage. Pointer 'addr' is dead if it has been assigned '&trapped_addr' at line 1375.

martinpitt, 2017-02-20:
That's quite deliberate. The whole point of the wrapper is to "divert" the `addr` argument and replace it with a local Unix socket. Where do you see this warning? I don't see it with gcc 6.3 at least, is that from coverity or similar? Is there a way to mark the code as "this is correct", or maybe a trick like setting `addr` to `NULL` before reassigning it or similar? Thanks!

dcb314, 2017-02-20:
I suspect you need to upgrade your knowledge of C programming language.
Local variable trapped_addr is out of scope by line 1381.
Having addr pointing at random stack data is seriously bad news at line 1381.

martinpitt, 2017-02-20:
Oh, this is just about the scope of `trapped_addr`? That's easy enough to fix (should mostly be cosmetical, but it indeed looks more plausible for moving it up). The error message is quite misleading, though. Thanks!

commit a84256fb (2017-02-20): preload: Fix variable scope

In connect(), trapped_addr is being used throughout connect(), so
declare it at the top.

Fixes #50
closed 2017-02-20 by martinpitt