martinpitt/umockdev #53: Aborts in UPower test suite

hadess, 2017-01-24:
Using umockdev 0.8.22 and UPower master, the linux/integration-test aborts during the test_bluetooth_mouse_reconnect() test:
```
sendmsg_one: cannot connect to client's event socket: Connection refused
```

With this backtrace:
```
                Stack trace of thread 25391:
                #0  0x00007f393297891f raise (libc.so.6)
                #1  0x00007f393297a51a abort (libc.so.6)
                #2  0x00007f39253b8b03 sendmsg_one (libumockdev.so.0)
                #3  0x00007f392774dc58 ffi_call_unix64 (libffi.so.6)
                #4  0x00007f392774d6ba ffi_call (libffi.so.6)
                #5  0x00007f392833b9cc pygi_invoke_c_callable (_gi.cpython-35m-x86_64-linux-gnu.so)
                #6  0x00007f392833d478 pygi_function_cache_invoke (_gi.cpython-35m-x86_64-linux-gnu.so)
                #7  0x00007f3928331229 _callable_info_call (_gi.cpython-35m-x86_64-linux-gnu.so)
```

*Not* unref'ing the GUdevClient in up_device_supply_get_sibling_with_subsystem() fixes the assertion.

I've disabled the test for now, rather than leaking the client.

martinpitt, 2017-01-26:
I can reproduce this, thanks for the report! Will look into that today.
```
$ LD_LIBRARY_PATH=~/upstream/umockdev/.libs/:libupower-glib/.libs/ GI_TYPELIB_PATH=~/upstream/umockdev/:libupower-glib PATH=$PATH:~/upstream/umockdev/src src/linux/integration-test Tests._test_bluetooth_mouse_reconnect
Testing binaries from local build tree
sendmsg_one: cannot connect to client's event socket: Connection refused
cleaning up pid 11684
Aborted (core dumped)
```

martinpitt, 2017-01-26:
Turned out to be quite simple and straightforward. Fixed in commit ebd1576612cba with a test case.

With this your upower `Tests._test_bluetooth_mouse_reconnect()` test works fine.

closed 2017-01-26 by martinpitt

martinpitt, 2017-01-26:
Released 0.8.13 with that fix: https://launchpad.net/umockdev/trunk/0.8.13

hadess, 2017-01-26:
Great, thanks!

commit ebd15766 (2017-01-26): Fix crash when client sends uevents to closed listeners

We lazily create the unix sockets which mock netlink sockets for
uevents, but so far never cleaned them. It may happen that the client
closes the connection to a previous netlink socket and sends an uevent
after that, so that our server-side socket fails with "Connection
refused". Intercept that and clean up the socket instead of bailing out
with an assertion.

Fixes #53