martinpitt/umockdev #54: Add sential annotation to umockdev_testbed_add_device()

hadess, 2017-02-20:
So that missing a NULL sentinel as in https://bugzilla.gnome.org/show_bug.cgi?id=778734 doesn't crash on some systems.

See:
http://stackoverflow.com/questions/12031059/gcc-double-null-sentinel-attribute

martinpitt, 2017-02-20:
Yeah, that would be nice. This seems to be rather complicated in C already, I need to see if that can be translated to Vala somehow.

martinpitt, 2018-08-18:
I searched the interwebs for this, and I think it's not possible to declare sentinels in Vala. There is a [sentinel CCode attribute](https://wiki.gnome.org/Projects/Vala/Manual/Attributes), but it's the "wrong way around" (it specifies a sentinel for calling variadic C functions, for Vala bindings).

It could be hacked into the generated umockdev.h, though. Adding a simple `__attribute__((sentinel))` at least helps against forgetting both `NULL`s, but this is an unrealistic degenerate case. But even the stackoverflow question from above doesn't really fit this: I don't want to enforce two NULLs in a row, just two lists. I don't think this can be expressed using the `sentinel` attribute.

martinpitt, 2019-06-10:
Closing, I have no idea how to fix this properly, unfortunately.

closed 2019-06-10 by martinpitt