martinpitt/umockdev #57: Tarball releases and link from github

shevegen, 2017-09-01:
Hello Martin,

The main readme contains a link to the launchpad site, e. g. then to the tarball:

  https://launchpad.net/umockdev/trunk/0.9.2/+download/umockdev-0.9.2.tar.xz

I was wondering whether there could additionally be a link to the tarball from the
github project site, from the main readme?

I am not entirely sure how this is done on github, but other projects manage to
do so such as https://github.com/MatzeB/libfirm/archive/libfirm-1.22.0.tar.gz or
https://github.com/libgit2/libgit2/archive/libgit2-0.25.0.tar.gz.

The reason why I would like to have that is mostly convenience; it's not a big
deal to wget from launchpad, or from github, but just quite convenient if we can
do so from github as well and also use so via simple scripts (via wget) and then
update to a new version automatically. Anyway, thanks for reading, please do
feel free to close this issue request at any moment in time either way!

martinpitt, 2017-09-07:
GitHub automatically generates tarballs from tags on https://github.com/martinpitt/umockdev/releases . These do exist for umockdev, and they are the same kind like the ones you referenced (libfirm). These are pure tarballs of the git checkout, i. e. they do not have any autogenerated files from autoconf or cmake in those. So you cannot just start with `./configure`, you have to install automake, autoconf etc. and run `./autogen.sh` first. If that's okay with you, you are of course welcome to use the autogenerated tarballs (or zip files).

The "official" release tarballs of umockdev on launchpad do include the autoconf pregenerated files, so that you don't need to run `autogen.sh` nor need its dependencies. I keep this open, it should not be that hard to extend umockdev's [do-release script](https://github.com/martinpitt/umockdev/blob/master/do-release) to upload the generated tarball (and relevant section of NEWS) to GitHub too, I'm just not familiar with the API. If you want to have a go at it, be my guest of course :-)

martinpitt, 2018-03-24:
With 0.11.1 I moved to GitHub as the primary download source: https://github.com/martinpitt/umockdev/releases/tag/0.11.1

This now contains "proper" `make dist` tarballs and news.

closed 2018-03-24 by martinpitt