martinpitt/umockdev #64: Updating to 0.9.4 causes umockdev_in_mock_environment to fail for fwupd

superm1, 2017-10-18:
In fwupd we run CI in docker containers across Fedora, Debian, Ubuntu, and Arch in various configurations.
fwupd uses umockdev with the thunderbolt plugin.
Debian and Ubuntu both recently moved to umockdev 0.9.4-1.

Debian things seem to continue to work, but in Ubuntu the test suite fails (https://github.com/hughsie/fwupd/issues/280).

```
--- command ---
LD_PRELOAD='libumockdev-preload.so.0' /build/build/obj-x86_64-linux-gnu/plugins/thunderbolt/thunderbolt-self-test
--- stderr ---
(/build/build/obj-x86_64-linux-gnu/plugins/thunderbolt/thunderbolt-self-test:1065): FuPluginThunderbolt-WARNING **: Need to run within umockdev wrapper (umockdev-wrapper thunderbolt-self-test)!
```
The test for this that shows the warning is simple:
https://github.com/hughsie/fwupd/blob/master/plugins/thunderbolt/fu-self-test.c#L1266

That worked fine in the previous version in Ubuntu before 0.9.4-1.

superm1, 2017-10-18:
Debian testing hasn't moved to 0.9.4, so that's probably why it's not failing there yet.  In debian testing (where CI runs) 0.9.3 works properly.

gicmo, 2017-10-19:
The problem is that the tbt fwupd plugin uses `umockdev_in_mock_env...` before we created the `UMockdevTestbed`, which then reports `FALSE` and we fail. See [here](https://github.com/hughsie/fwupd/issues/280#issuecomment-337855991) for more details. The behaviour of the function clearly changed, question is should we adapt or should umockdev change (maybe doing both, the LD_PRELOAD check and the sysfs check)?

superm1, 2017-10-19:
Thanks for clarifying what was going on @gicmo.
So I switched this around in fwupd: https://github.com/hughsie/fwupd/commit/fecfe3b8db3846be366d2916d47f5b9462e2ebd4 to test for after the env was setup.

I do agree if possible it's probably worth checking both LD_PRELOAD and sysfs check though so as to not cause regressions for anyone else from the 0.9.4 changeover.

martinpitt, 2017-11-14:
Sorry for the late reply. The previous behaviour of `umockdev_in_mock_env()` was really wrong, as you can see with trying to run `umockdev-run` with < 0.9.4 - you would get confusing warnings about nonexisting devices in `uevent_sender_send()`. It previously only tested if `$LD_PRELOAD` contained `libumockdev-preload`, not if the process was *actually* using it. This would go wrong if a process (like umockdev-run or another test suite entry point) would set `$LD_PRELOAD` - then this applies to child processes, but *not* to that process itself.

So the commit you referenced looks good - this should also work with previous versions. As such I'd declare this a "won't fix" and close. Is that okay with you, or do you still have trouble with the current implementation?

superm1, 2017-11-14:
fine by me.  thanks!

closed 2017-11-14 by superm1