martinpitt/umockdev #67: Build using CMake

radhikapraveen, 2017-12-20:
Hi Martin,

I would like to include  umockdev in our project. But our projects use cmake. Is there a version of umockdev built using cmake? Am trying to convert to cmake but finding it difficult.

martinpitt, 2017-12-20:
There's no version using cmake, to my knowledge. Also, I really don't want to convert umockdev itself to cmake - the only thing I could be talked into would be meson.

But what do you mean with "include"? If you build the whole thing as-is, then linking to the shared library and just including the headers seems enough (and you can do that with any build system). And if you want to copy some portions of the code, you most likely can't reuse umockdev's build system anyway.

martinpitt, 2018-02-09:
No response in two months, closing. Feel free to reopen if you want to follow up. Thanks!

closed 2018-02-09 by martinpitt