martinpitt/umockdev #73: umockdev-testbed/proc fails on sparc64 due to "processor" keyword not found

glaubitz, 2018-02-14:
On some architectures like sparc64, the keyword "processor" isn't actually found in /proc/cpuinfo and the testsuite fails:

```
/umockdev-testbed/proc: 'processor' not found in /proc/cpuinfo: ----------
cpu		: UltraSparc T1 (Niagara)
fpu		: UltraSparc T1 integrated FPU
pmu		: niagara
prom		: OBP 4.30.4.e 2013/09/23 16:08
type		: sun4v
ncpus probed	: 16
ncpus active	: 16
D$ parity tl1	: 0
I$ parity tl1	: 0
cpucaps		: flush,stbar,swap,muldiv,v9,blkinit,mul32,div32,v8plus,ASIBlkInit
Cpu0ClkTck	: 000000003b9aca00
Cpu1ClkTck	: 000000003b9aca00
Cpu2ClkTck	: 000000003b9aca00
Cpu3ClkTck	: 000000003b9aca00
Cpu4ClkTck	: 000000003b9aca00
Cpu5ClkTck	: 000000003b9aca00
Cpu6ClkTck	: 000000003b9aca00
Cpu7ClkTck	: 000000003b9aca00
Cpu8ClkTck	: 000000003b9aca00
Cpu9ClkTck	: 000000003b9aca00
Cpu10ClkTck	: 000000003b9aca00
Cpu11ClkTck	: 000000003b9aca00
Cpu12ClkTck	: 000000003b9aca00
Cpu13ClkTck	: 000000003b9aca00
Cpu14ClkTck	: 000000003b9aca00
Cpu15ClkTck	: 000000003b9aca00
MMU Type	: Hypervisor (sun4v)
MMU PGSZs	: 8K,64K,4MB,256MB
```

In general, different architectures have unfortunately different output in /proc/cpuinfo and it should not be assumed that all keywords are found on all architectures.

On m68k, processor doesn't exist either:

```
root@elgar:~> cat /proc/cpuinfo     
CPU:            68060
MMU:            68060
FPU:            68060
Clocking:       48.8MHz
BogoMips:       97.68
Calibration:    488448 loops
root@elgar:~>
```
Let me know if you need a comprehensive list of /proc/cpuinfo output across different architectures, but in general it can be retrieved from the kernel sources.

CC @jrtc27

martinpitt, 2018-02-14:
Thanks @glaubitz for your report! It seems there is nothing reliable across architectures, so I figure I'll just use a different file for the tests;  as you seem to have easy access to these arches, how about /proc/meminfo? Can I rely on `MemTotal:`  being in there?

glaubitz, 2018-02-14:
> as you seem to have easy access to these arches, how about /proc/meminfo? Can I rely on MemTotal: being in there?

Yes, I just checked all architectures we have in Debian and ```MemTotal:``` is present on all of them.

martinpitt, 2018-02-14:
@glaubitz: Yay, many thanks for checking! Fixed in #74.

PR #74 (2018-02-14): Fix architecture specific /proc test
`/proc/cpuinfo` is very architecture specific and has no reliable
strings in it. Use `/proc/meminfo` instead.

Thanks to John Paul Adrian Glaubitz!

Fixes #73
commit 2ea66d6c (2018-02-14): Fix architecture specific /proc test (#74)

`/proc/cpuinfo` is very architecture specific and has no reliable
strings in it. Use `/proc/meminfo` instead.

Thanks to John Paul Adrian Glaubitz!

Fixes #73
closed 2018-02-14 by martinpitt