martinpitt/umockdev #75: Build failure in GNOME Continuous

ebassi, 2018-02-14:
umockdev fails to build in the GNOME continuous integration server inside umockdev-run:

```
src/umockdev-run.vala:185.24-185.34: error: Argument 1: Cannot pass value to reference or output parameter
    Posix.sigemptyset (act.sa_mask);
```

Full build log available [here](http://build.gnome.org/continuous/buildmaster/builds/2018/02/14/59/build/log-umockdev.txt).

tankoslinux, 2018-03-20:
Same here.

Build log:

```
lcov not installed, not enabling code coverage
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:35: installing 'build-aux/ar-lib'
configure.ac:29: installing 'build-aux/compile'
configure.ac:38: installing 'build-aux/config.guess'
configure.ac:38: installing 'build-aux/config.sub'
configure.ac:32: installing 'build-aux/install-sh'
configure.ac:32: installing 'build-aux/missing'
Makefile.am: installing 'build-aux/depcomp'
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking how to create a pax tar archive... gnutar
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... dlltool
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
[... 77 lines cut ...]
make  all-recursive
make[1]: Entering directory '/build/umockdev-0.11.1'
Making all in .
make[2]: Entering directory '/build/umockdev-0.11.1'
  CC       src/libumockdev_preload_la-ioctl_tree.lo
  CC       src/libumockdev_preload_la-debug.lo
  CC       src/libumockdev_preload_la-libumockdev-preload.lo
  CC       src/tests_test_ioctl_tree-ioctl_tree.o
  VALAC    src_umockdev_run_vala.stamp
  VALAC    src_umockdev_record_vala.stamp
  VALAC    tests_test_umockdev_vala_vala.stamp
  VALAC    tests_test_umockdev_run_vala.stamp
src/umockdev-run.vala:185.24-185.34: error: Argument 1: Cannot pass value to reference or output parameter
    Posix.sigemptyset (act.sa_mask);
                       ^^^^^^^^^^^
src/umockdev-run.vala:186.30-186.42: warning: Posix.SIGTERM has been deprecated since vala-0.40. Use Posix.Signal.TERM
src/umockdev-run.vala:187.30-187.41: warning: Posix.SIGHUP has been deprecated since vala-0.40. Use Posix.Signal.HUP
src/umockdev-run.vala:188.30-188.41: warning: Posix.SIGINT has been deprecated since vala-0.40. Use Posix.Signal.INT
src/umockdev-run.vala:189.30-189.42: warning: Posix.SIGQUIT has been deprecated since vala-0.40. Use Posix.Signal.QUIT
src/umockdev-run.vala:190.30-190.42: warning: Posix.SIGABRT has been deprecated since vala-0.40. Use Posix.Signal.ABRT
Compilation failed: 1 error(s), 5 warning(s)
src/umockdev-record.vala:275.17-275.24: warning: GLib.HashTable.add has been deprecated since vala-0.26. Use GenericSet.add
make[2]: *** [Makefile:1401: src_umockdev_run_vala.stamp] Error 1
make[2]: *** Waiting for unfinished jobs....
```

finikorg, 2018-03-23:
I have similar issue here: https://travis-ci.org/intel/thunderbolt-software-user-space

```
make[2]: Entering directory '/usr/local/src/umockdev'
  CC       src/libumockdev_preload_la-ioctl_tree.lo
  CC       src/libumockdev_preload_la-debug.lo
  CC       src/libumockdev_preload_la-libumockdev-preload.lo
  VALAC    src_umockdev_run_vala.stamp
src/umockdev-run.vala:185.24-185.34: error: Argument 1: Cannot pass value to reference or output parameter
    Posix.sigemptyset (act.sa_mask);
                       ^^^^^^^^^^^
```

martinpitt, 2018-03-24:
Thanks for the report! Fixed in master now.

commit 2b89e02a (2018-03-24): Fix build failures with glibc 2.27

    src/umockdev-record.c:355:10: error: called object ‘major’ is not a function or function pointer
      major = major (_tmp5_);

    src/umockdev_record-umockdev-record.o: In function `resolve':
      src/umockdev-record.c:355: undefined reference to `major'

Add a hack to posix_extra.vapi to force #inclusion of sys/sysmacros.h
(work around https://bugzilla.gnome.org/show_bug.cgi?id=794651)

Fixes #75
closed 2018-03-24 by martinpitt