martinpitt/umockdev #76: Build failure in 0.11.1

hadess, 2018-03-01:
Tried to build 0.11.1 in Fedora Rawhide:
```
src/umockdev-record.c: In function 'resolve':
src/umockdev-record.c:354:10: error: called object 'major' is not a function or function pointer
  major = major (_tmp5_);
          ^~~~~
src/umockdev-record.c:325:8: note: declared here
  guint major = 0U;
        ^~~~~
src/umockdev-record.c:357:10: error: called object 'minor' is not a function or function pointer
  minor = minor (_tmp7_);
          ^~~~~
src/umockdev-record.c:328:8: note: declared here
  guint minor = 0U;
        ^~~~~
```

See for full log:
https://kojipkgs.fedoraproject.org//work/tasks/5025/25395025/build.log

hadess, 2018-03-01:
Builds correctly on Fedora 27, likely a problem with glibc or gcc changes.

hadess, 2018-03-01:
I have half a fix (rename the variable to major_int/minor_int, so they can't be confused with the function calls), but the generated vala is missing `#include <sys/sysmacros.h>`, which would require adding a new .vapi...

martinpitt, 2018-03-24:
Indeed, I can easily replicate the build failures in a rawhide mock. I filed https://bugzilla.gnome.org/show_bug.cgi?id=794651 with a small reproducer about the missing `#include` for `sysmacros.h`. I'll add a workaround for now. Once doing that, I also run into issue #75, looking at that now.

commit 15cff99e (2018-03-24): Fix build failure with vala 0.40

    src/umockdev-run.vala:185.24-185.34: error: Argument 1: Cannot pass value to reference or output parameter
        Posix.sigemptyset (act.sa_mask);

Fixes #76
commit 73f38e0e (2018-03-24): Fix build failures with glibc 2.27

    src/umockdev-record.c:355:10: error: called object ‘major’ is not a function or function pointer
      major = major (_tmp5_);

    src/umockdev_record-umockdev-record.o: In function `resolve':
      src/umockdev-record.c:355: undefined reference to `major'

Add a hack to posix_extra.vapi to force #inclusion of sys/sysmacros.h
(work around https://bugzilla.gnome.org/show_bug.cgi?id=794651)

Fixes #76
commit 067f3c21 (2018-03-24): Fix build failures with glibc 2.27

    src/umockdev-record.c:355:10: error: called object ‘major’ is not a function or function pointer
      major = major (_tmp5_);

    src/umockdev_record-umockdev-record.o: In function `resolve':
      src/umockdev-record.c:355: undefined reference to `major'

Add a hack to posix_extra.vapi to force #inclusion of sys/sysmacros.h
(work around https://bugzilla.gnome.org/show_bug.cgi?id=794651)

Fixes #76
commit 74824341 (2018-03-24): Fix build failure with vala 0.40

    src/umockdev-run.vala:185.24-185.34: error: Argument 1: Cannot pass value to reference or output parameter
        Posix.sigemptyset (act.sa_mask);

Fixes #76
closed 2018-03-24 by martinpitt