martinpitt/umockdev #81: umockdev-record hangs when reading some driver's sysfs attribute files

dkl, 2018-06-15:
Hi,

`umockdev-record --all` can sometimes hang when trying to read certain sysfs files, if the read blocks.

For example I noticed this with the `fwlog` attribute of a `wl12xx` device. It appears as empty regular file, but when doing `cat` on it, it hangs in `read()` until CTRL+C (same with umockdev-record). I tried using `Posix.open(... O_NONBLOCK)` instead of `FileUtils.get_data()` but that didn't help.

It would be nice to have a work-around for this, for example...
- use a timeout when reading, if possible -- maybe read in a subprocess and send SIGTERM after a while?
- have command line options for identifying and skipping such files, for example `--debug` (showing accessed files) and then `--skip-path=...`.
- blacklist the known cases and skip them automatically
- check file size and don't `read()` at all if it's zero (that would help in my case at least)

dkl, 2018-08-17:
Yea, I can do more tests (in about 2 weeks); trying `select()` sounds like a good idea.

martinpitt, 2018-08-18:
As a first step, I rewrote the attribute reading using a single non-blocking POSIX `read()`. This should already be a bit better, and it's now easy to add a `select()` or other additional check into it.

I'm doing this in https://github.com/martinpitt/umockdev/tree/record-all .

martinpitt, 2018-08-18:
I pushed the `select()` check to the branch. I don't have access to a machine which exhibits this behaviour (I tried my own and all Debian porter boxes), so when you have a chance, can you please give this a try? Thanks!

dkl, 2018-08-18:
Ok thanks, will do.

dkl, 2018-09-11:
I've tested it again today, but this [test program](https://gist.github.com/dkl/168935259d5cf931ed1214f5d169d0ba) using O_NONBLOCK+select() even hangs, with strace showing this:
```
[...]
open("/sys/devices/soc0/soc/2100000.aips-bus/2194000.usdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/wl12xx.0.auto/fwlog", O_RDONLY|O_NONBLOCK) = 3
select(4, [3], NULL, NULL, {10, 0})     = 1 (in [3], left {9, 999992})
read(3, 
[hangs here]
```
It seems to be the same issue for `umockdev-record --all` built from the record-all branch:
```
[...]
P: /devices/soc0/soc/2100000.aips-bus/2194000.usdhc/mmc_host/mmc1/mmc1:0001/mmc1:0001:2/wl12xx.0.auto
E: DRIVER=wl12xx_driver
E: MODALIAS=platform:wl12xx
E: SUBSYSTEM=platform
A: bt_coex_state=1\n\n0 - off\n1 - on
L: driver=../../../../../../../../../../bus/platform/drivers/wl12xx_driver
A: driver_override=(null)
[hangs here]
```

martinpitt, 2018-09-14:
Thanks for testing. It seems outright impossible from userspace then to not fall into this trap. I suggest filing this a a kernel bug instead, as this attribute is really just completely broken then :-(

dkl, 2018-10-03:
Ok, that makes sense. I'll just close this then since there's no good way to work around it.

closed 2018-10-03 by dkl