martinpitt/umockdev #82: Replay USB control transfers?

superm1, 2018-07-13:
I was hoping to use umockdev to mock a USB device characteristics.  I normally interact with the device via HID reports.

I've found that trying to capture the interactions with both `--ioctl` and `--script` don't actually work though.

For the purpose of showing this, I wrote a really simple sample application and wrapper that could interact with two of the USB endpoints here:
https://github.com/superm1/umockdev-hid

I use it to try to pull out two bytes of data from the report embedded in the USB control transfer.

```
Running: umockdev-record /dev/bus/usb/001/016

Running: umockdev-record --ioctl=/dev/bus/usb/001/016=001.016.ioctl ./hidreport b06e
Querying 413c:b06e
Major: 1, Minor: 17

Running: umockdev-run --device=001.016.umockdev --ioctl=/dev/bus/usb/001/016=001.016.ioctl ./hidreport b06e
Querying 413c:b06e
ERROR: libumockdev-preload: failed to load ioctl record file for /dev/bus/usb/001/016: empty or invalid format?

Running: umockdev-record /dev/bus/usb/001/018

Running: umockdev-record --ioctl=/dev/bus/usb/001/018=001.018.ioctl ./hidreport b06f
Querying 413c:b06f
Major: 1, Minor: 5

Running: umockdev-run --device=001.018.umockdev --ioctl=/dev/bus/usb/001/018=001.018.ioctl ./hidreport b06f
Querying 413c:b06f
ERROR: libumockdev-preload: failed to load ioctl record file for /dev/bus/usb/001/018: empty or invalid format?
```

I looked and the files didn't really contain anything useful indeed.  So trying to manually run without the ioctl to see what happens I notice that the claiming action fails.
```
$ sudo umockdev-run --device=001.016.umockdev ./hidreport b06e
Querying 413c:b06e
failed to claimUSB error on device 413c:b06e : Other error [-99]
```

That's fine, I can try to do it without a claim, but then it hangs (I guess because there is no script or ioctl data to feed?).
```
$ sudo umockdev-run --device=001.016.umockdev ./hidreport b06e
Querying 413c:b06e

```

So my question is how to handle this situation?  Is there a proper way to capture those control transfers for replay?

superm1, 2018-07-13:
Also all of the sample data I've saved in a commit on that repo with the sample program.

superm1, 2018-07-13:
One more thing to add in case it helps to understand what's going on, here is what libusb is actually doing during those calls.

```
$ sudo LIBUSB_DEBUG=10 ./create_report.py 
libusb: debug [libusb_init] created default context
libusb: debug [libusb_init] libusb v1.0.21.11156
libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
libusb: debug [op_init] bulk continuation flag supported
libusb: debug [op_init] zero length packet flag supported
libusb: debug [op_init] sysfs can relate devices
libusb: debug [op_init] sysfs has complete descriptors
libusb: debug [linux_udev_event_thread_main] udev event thread entering.
libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
libusb: debug [linux_get_device_address] scan usb1
libusb: debug [linux_get_device_address] bus=1 dev=1
libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
libusb: debug [linux_get_device_address] scan 1-1
libusb: debug [linux_get_device_address] bus=1 dev=29
libusb: debug [linux_enumerate_device] busnum 1 devaddr 29 session_id 285
libusb: debug [linux_enumerate_device] allocating new device for 1/29 (session 285)
libusb: debug [linux_get_parent_info] Dev 0x55844431f100 (1-1) has parent 0x55844431bd90 (usb1) port 1
libusb: debug [linux_get_device_address] getting address for device: 1-1.3 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3
libusb: debug [linux_get_device_address] bus=1 dev=30
libusb: debug [linux_enumerate_device] busnum 1 devaddr 30 session_id 286
libusb: debug [linux_enumerate_device] allocating new device for 1/30 (session 286)
libusb: debug [linux_get_parent_info] Dev 0x558444314280 (1-1.3) has parent 0x55844431f100 (1-1) port 3
libusb: debug [linux_get_device_address] getting address for device: 1-1.3.4 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3.4
libusb: debug [linux_get_device_address] bus=1 dev=32
libusb: debug [linux_enumerate_device] busnum 1 devaddr 32 session_id 288
libusb: debug [linux_enumerate_device] allocating new device for 1/32 (session 288)
libusb: debug [linux_get_parent_info] Dev 0x558444313900 (1-1.3.4) has parent 0x558444314280 (1-1.3) port 4
libusb: debug [linux_get_device_address] getting address for device: 1-1.3.5 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3.5
libusb: debug [linux_get_device_address] bus=1 dev=33
libusb: debug [linux_enumerate_device] busnum 1 devaddr 33 session_id 289
libusb: debug [linux_enumerate_device] allocating new device for 1/33 (session 289)
libusb: debug [linux_get_parent_info] Dev 0x558444316870 (1-1.3.5) has parent 0x558444314280 (1-1.3) port 5
libusb: debug [linux_get_device_address] getting address for device: 1-1.5 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.5
libusb: debug [linux_get_device_address] bus=1 dev=31
libusb: debug [linux_enumerate_device] busnum 1 devaddr 31 session_id 287
libusb: debug [linux_enumerate_device] allocating new device for 1/31 (session 287)
libusb: debug [linux_get_parent_info] Dev 0x558444319cb0 (1-1.5) has parent 0x55844431f100 (1-1) port 5
libusb: debug [linux_get_device_address] getting address for device: 1-5 detached: 0
libusb: debug [linux_get_device_address] scan 1-5
libusb: debug [linux_get_device_address] bus=1 dev=3
libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
libusb: debug [linux_get_parent_info] Dev 0x55844431bfd0 (1-5) has parent 0x55844431bd90 (usb1) port 5
libusb: debug [linux_get_device_address] getting address for device: 1-7 detached: 0
libusb: debug [linux_get_device_address] scan 1-7
libusb: debug [linux_get_device_address] bus=1 dev=5
libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
libusb: debug [linux_get_parent_info] Dev 0x558444314920 (1-7) has parent 0x55844431bd90 (usb1) port 7
libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
libusb: debug [linux_get_device_address] scan usb2
libusb: debug [linux_get_device_address] bus=2 dev=1
libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
libusb: debug [linux_get_device_address] scan 2-1
libusb: debug [linux_get_device_address] bus=2 dev=77
libusb: debug [linux_enumerate_device] busnum 2 devaddr 77 session_id 589
libusb: debug [linux_enumerate_device] allocating new device for 2/77 (session 589)
libusb: debug [linux_get_parent_info] Dev 0x558444208b30 (2-1) has parent 0x558444316250 (usb2) port 1
[... 446 lines cut ...]
libusb: debug [linux_get_device_address] getting address for device: 1-1.3.5 detached: 0
libusb: debug [linux_get_device_address] scan 1-1.3.5
libusb: debug [linux_get_device_address] bus=1 dev=33
libusb: debug [linux_enumerate_device] busnum 1 devaddr 33 session_id 289
libusb: debug [linux_enumerate_device] allocating new device for 1/33 (session 289)
libusb: debug [linux_get_parent_info] Dev 0x5633ad1fed70 (1-1.3.5) has parent 0x5633ad1fd1c0 (1-1.3) port 5
libusb: debug [usbi_add_pollfd] add fd 6 events 1
[... cut ...]
libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
libusb: debug [libusb_get_device_descriptor] 
Querying 413c:b06f
libusb: debug [libusb_open] open 1.33
ERROR: libumockdev-preload: failed to load ioctl record file for /dev/bus/usb/001/033: empty or invalid format?
```

superm1, 2018-07-14:
Ah, I think I might have figured out what's going on.

Was looking through `UMOCKDEV_DEBUG=ioctl-tree` output and matching to code and encountered this snippet: 
```
    /* we assume that every SUBMITURB is followed by a REAPURB and that
     * ouput EPs don't change the buffer, so we ignore USBDEVFS_SUBMITURB */
    I_CUSTOM(USBDEVFS_REAPURB, 0, usbdevfs_reapurb),
```

From my understanding that sounds like it's exactly the problem here.  The buffer does change between `SUBMITURB` and `REAPURB`.  That's actually the exact data I'm interested in here.

benzea, 2021-01-05:
I might look into this some time soon. I kinda need it for libfprint testing.

Though, tbh. for libfprint testing I would prefer a pure trace/replay approach in general. i.e. I keep needing to manually restructure the ioctl tree because the devices repeat the same commands with differing responses and umockdev incorrectly groups them.

benzea, 2021-01-05:
> Ah, I think I might have figured out what's going on.
>
> Was looking through `UMOCKDEV_DEBUG=ioctl-tree` output and matching to code and encountered this snippet:
>
> ```
>     /* we assume that every [...]

I am thinking that we need to do some sort of partial matching on the submit buffer. e.g. for a register read match everything except for the value data (rather than nothing).

benzea, 2021-03-25:
So, my solution to this is to restructure the `ioctl` replaying to run in the `umockdev` main process and then implement a replayer on top of `pcap` (or `pcapng`) packet USB monitor dumps in vala.

See https://github.com/benzea/umockdev/tree/benzea/pcap-usb-replay and https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/258 for an example.

For me, this works great. It provides a purely linear playback order, which works much better for the libfprint CI and will result in e.g. timeouts to be correctly replayed. Also, I believe that control transfers from the userspace application will simply work in that codebase, but I have not tested that part yet.

All this works by pushing the `ioctl` emulation through a socket in order to provide a more flexibility for emulation (you could e.g. implement ioctl emulation in python for the test suite). The disadvantages are that this is relatively slow (scheduling overhead) and that it is likely hard to adapt for fast fuzzying (i.e. you can't just fork to copy the process state in order to inject errors at different points).

superm1, 2021-03-25:
Cool!  @martinpitt are you open to merging this?  
@hughsie this can potentially be really interesting for fwupd fuzzing..

martinpitt, 2021-03-28:
@superm1 : Yes, in general -- I expect #118 will still need a few iterations, but I hope that can land relatively soon.

closed 2021-06-15 by martinpitt

benzea, 2021-07-12:
For reference: #137 finalises the support and makes control transfer replay work correctly.

PR #124 (2021-06-15): Implement pcap based USB replay
The ioctl-tree based replay is mostly useful for protocols that are fundamentally state-less. With state-full protocols, the ioctl-tree may randomly jump around, cause confusing situations and gets the order wrong during recording.

The pcap based replay is also able to play-back control transfers adding this feature to umockdev, even if outside of the original ioctl-tree implementation.
    
Note that umockdev does not support creating a recording. Recordings should be done using a separate tool such as wireshark.
    
Closes: #82, #117
commit 86d4809c (2021-06-15): Implement pcap based USB replay

The ioctl-tree based replay is mostly useful for protocols that are
fundamentally state-less. With state-full protocols, the ioctl-tree may
randomly jump around, cause confusing situations and gets the order
wrong during recording.

The pcap based replay is also able to play-back control transfers adding
this feature to umockdev, even if outside of the original ioctl-tree
implementation.

Note that umockdev does not support creating a recording. Recordings
should be done using a separate tool such as wireshark.

Closes: #82, #117