martinpitt/umockdev #83: Compilation failure on tests

PureTryOut, 2018-07-31:
I'm trying to compile umockdev on a Musl based system (not sure if relevant), but it fails to compile some tests which breaks compilation entirely.

```
  CC       src/libumockdev_la-umockdev.lo
  CC       src/libumockdev_la-uevent_sender.lo
  CCLD     libumockdev.la
ar: `u' modifier ignored since `D' is the default (see `U')
make  all-recursive
make[1]: Entering directory '/home/pmos/build/src/umockdev-0.11.3'
Making all in .
make[2]: Entering directory '/home/pmos/build/src/umockdev-0.11.3'
  VALAC    src_umockdev_run_vala.stamp
  CC       src/umockdev_run-umockdev-run.o
  CCLD     src/umockdev-run
  VALAC    src_umockdev_record_vala.stamp
  CC       src/umockdev_record-umockdev-record.o
  CCLD     src/umockdev-record
  CC       tests/chatter-chatter.o
  CCLD     tests/chatter
  CC       tests/chatter_socket_stream-chatter-socket-stream.o
  CCLD     tests/chatter-socket-stream
  CC       tests/test_umockdev-test-umockdev.o
tests/test-umockdev.c: In function 't_testbed_uevent_error':
tests/test-umockdev.c:757:12: error: assignment of read-only variable 'stderr'
     stderr = tmpfile();
            ^
tests/test-umockdev.c:765:12: error: assignment of read-only variable 'stderr'
     stderr = orig_stderr;
            ^
tests/test-umockdev.c: In function 't_testbed_libc':
tests/test-umockdev.c:1023:12: error: implicit declaration of function 'canonicalize_file_name' [-Werror=implicit-function-declaration]
     path = canonicalize_file_name("/sys/devices/dev1");
            ^~~~~~~~~~~~~~~~~~~~~~
tests/test-umockdev.c:1023:5: error: nested extern declaration of 'canonicalize_file_name' [-Werror=nested-externs]
     path = canonicalize_file_name("/sys/devices/dev1");
     ^~~~
tests/test-umockdev.c:1023:10: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     path = canonicalize_file_name("/sys/devices/dev1");
          ^
tests/test-umockdev.c:1028:10: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     path = canonicalize_file_name("/sys/bus/pci/devices/dev1");
          ^
In file included from /usr/lib/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from tests/test-umockdev.c:25:
tests/test-umockdev.c:1033:63: warning: comparison between pointer and integer
     g_assert(canonicalize_file_name("/sys/devices/xxnoexist") == NULL);
                                                               ^
/usr/include/glib-2.0/glib/gmacros.h:379:8: note: in definition of macro '_G_BOOLEAN_EXPR'
    if (expr)                                    \
        ^~~~
/usr/include/glib-2.0/glib/gtestutils.h:118:49: note: in expansion of macro 'G_LIKELY'
                                              if G_LIKELY (expr) ; else \
                                                 ^~~~~~~~
tests/test-umockdev.c:1033:5: note: in expansion of macro 'g_assert'
     g_assert(canonicalize_file_name("/sys/devices/xxnoexist") == NULL);
     ^~~~~~~~
cc1: some warnings being treated as errors
```

martinpitt, 2018-08-17:
I just pushed some changes which should fix these two issues. Can you please try to build current git master to confirm?

PureTryOut, 2018-08-17:
Sadly it does not. You added a `#ifdef __MUSL__` statement but that's not actually something Musl defines. Musl has no way to differentiate itself from other libc's by itself, so normally `#ifndef __GLIBC__` is used.

Also, when adding that instead, compilation fails on some now unused variables. Based on your commits I made a working patch for Musl (applied on top of 0.11.3, **not** the latest git master):

```
diff --git a/src/libumockdev-preload.c b/src/libumockdev-preload.c
index 94f2d16..0b06deb 100644
--- a/src/libumockdev-preload.c
+++ b/src/libumockdev-preload.c
@@ -1274,6 +1274,8 @@ ssize_t readlinkat(int dirfd, const char *pathname, char *buf, size_t bufsiz)
     return r;
 }
 
+#ifdef __GLIBC__
+
 char *canonicalize_file_name(const char *path)
 {
     const char *p;
@@ -1300,6 +1302,8 @@ char *canonicalize_file_name(const char *path)
     return r;
 }
 
+#endif
+
 ssize_t
 read(int fd, void *buf, size_t count)
 {
diff --git a/tests/test-umockdev.c b/tests/test-umockdev.c
index 8318f25..6990688 100644
--- a/tests/test-umockdev.c
+++ b/tests/test-umockdev.c
@@ -743,8 +743,7 @@ t_testbed_uevent_error(UMockdevTestbedFixture * fixture, gconstpointer data)
 {
     struct udev *udev;
     struct udev_monitor *mon;
-    FILE *orig_stderr;
-    char buf[1000];
+    FILE *orig_stderr; 
 
     /* set up monitor */
     udev = udev_new();
@@ -754,6 +753,10 @@ t_testbed_uevent_error(UMockdevTestbedFixture * fixture, gconstpointer data)
 
     /* unknown device, shouldn't crash but print an error message */
     orig_stderr = stderr;
+
+	/* musl libc has readonly stderr */
+#ifdef __GLIBC__
+	char buf[1000];
     stderr = tmpfile();
     g_assert(stderr != NULL);
     umockdev_testbed_uevent(fixture->testbed, "/devices/unknown", "add");
@@ -763,6 +766,7 @@ t_testbed_uevent_error(UMockdevTestbedFixture * fixture, gconstpointer data)
     g_assert_cmpstr(buf, ==, "ERROR: uevent_sender_send: No such device /devices/unknown\n");
     fclose(stderr);
     stderr = orig_stderr;
+#endif
 
     /* should not trigger an actual event */
     g_assert(udev_monitor_receive_device(mon) == NULL);
@@ -1006,7 +1010,6 @@ t_testbed_libc(UMockdevTestbedFixture * fixture, gconstpointer data)
 {
     gboolean success;
     GError *error = NULL;
-    char *path;
     int dirfd, fd;
 
     /* start with adding one device */
@@ -1017,6 +1020,8 @@ t_testbed_libc(UMockdevTestbedFixture * fixture, gconstpointer data)
     g_assert_no_error(error);
     g_assert(success);
 
+#ifdef __GLIBC__
+    char *path;
     /* canonicalize_file_name */
 
     /* dir */
@@ -1032,6 +1037,7 @@ t_testbed_libc(UMockdevTestbedFixture * fixture, gconstpointer data)
     /* nonexisting */
     g_assert(canonicalize_file_name("/sys/devices/xxnoexist") == NULL);
     g_assert_cmpint(errno, ==, ENOENT);
+#endif
 
     /* openat */
```

closed 2018-08-18 by martinpitt

z3ntu, 2018-08-18:
@PureTryOut I think that we have a patch for that (some include change) in the pmos branch

martinpitt, 2018-08-18:
@PureTryOut : These `_IOC_*` macros don't come from libc, but from the Linux kernel headers (/usr/include/asm-generic/ioctl.h here), and umockdev really needs those. But it seems @z3ntu already has a fix for that :-)

z3ntu, 2018-08-18:
That should be https://gitlab.com/postmarketOS/pmbootstrap/blob/feature/unity8/aports/main/umockdev/fix-ioctl.patch
You can also look at the other `.patch` files in that directory which contains what we needed.
The 64-bit patch is because musl defines some of the things that were commented out as an alias to the non-64-bit variant which then resulted in the same thing being registered twice and failing.

martinpitt, 2018-08-19:
Thanks, this `#include` change actually looks correct. I pushed it to master: https://github.com/martinpitt/umockdev/commit/62261c55ceb2

So fix-musl-errors.patch, fix-failing-tests.patch, and fix-ioctl.patch should now be obsolete with current master.

fix-64bit-errors.patch needs a closer look. `man ioctl(2)` really shows the declaration as `unsigned long request`, changing that to `int` sounds wrong for upstream. If the various `*64()` functions don't exist on your platform, then they might be `#ifdef`ed out, but this needs some more research if these are really glibc and/or architecture specific, i. e. what the correct conditions are.

martinpitt, 2018-08-19:
Most of your [fix-64bit-errors.patch](https://gitlab.com/postmarketOS/pmbootstrap/blob/feature/unity8/aports/main/umockdev/fix-64bit-errors.patch) should now be obsolete with commit https://github.com/martinpitt/umockdev/commit/741753653f197659bdc0496a33a6afb757686870 .

This leaves the `ioctl()` part. If you want to craft a configure.ac check for it, I'm happy to include that upstream, but it must stay compatible with glibc.

z3ntu, 2018-08-19:
https://pastebin.com/eU9et6hi is the current build log; and also thanks for trying to make it work more elegantly than our patches :)

martinpitt, 2018-08-19:
@z3ntu : Argh, sorry for the `stat()` vs. `stat64()` confusion, I fixed that on current master (I force-pushed, sorry). As for the `fopen64()` and `__xstat()` failures, the configure.ac test was supposed to be "no" for you. Can you verify?
```
checking for fopen64 in -lc... yes
checking for __xstat in -lc... yes
```
If they are "yes", then your libc actually *does* define them, but they are not in the header files. At this point it becomes too cumbersome for me to figure this out just by guesswork - can you work on a patch?

z3ntu, 2018-08-19:
```
checking for fopen64 in -lc... yes
checking for __xstat in -lc... yes
```
:(

z3ntu, 2018-08-19:
```diff
diff --git a/configure.ac b/configure.ac
index 780d16d..2383535 100644
--- a/configure.ac
+++ b/configure.ac
@@ -38,8 +38,28 @@ LT_PREREQ(2.2)
 LT_INIT
 
 # check which functions to wrap in preload
-AC_CHECK_LIB([c], [fopen64], [AC_DEFINE([HAVE_FILE64], [1], [libc has fopen64() and similar functions]) ])
-AC_CHECK_LIB([c], [__xstat], [AC_DEFINE([HAVE_XSTAT], [1], [libc has __xstat() and similar functions]) ])
+# Check for open64 to know if the current system does have open64() and similar functions
+AC_CHECK_FUNCS_ONCE([open64])
+
+# With Glibc ioctl takes an 'unsigned long' instead of the POSIX 'int'
+# for the second parameter.
+AC_CACHE_CHECK([for ioctl with POSIX signature],
+    [gl_cv_func_ioctl_posix_signature],
+    [AC_COMPILE_IFELSE(
+        [AC_LANG_PROGRAM(
+        [[#include <sys/ioctl.h>]],
+        [[extern
+            #ifdef __cplusplus
+            "C"
+            #endif
+            int ioctl (int, int, ...);
+        ]])
+        ],
+        [gl_cv_func_ioctl_posix_signature=yes],
+        [gl_cv_func_ioctl_posix_signature=no])
+    ])
+AS_IF([test "x$gl_cv_func_ioctl_posix_signature" = "xyes"], AC_DEFINE([HAVE_POSIX_IOCTL], 1, [Have a POSIX ioctl?]))
+
 
 # check for gtk-doc
 m4_ifdef([GTK_DOC_CHECK], [
diff --git a/src/libumockdev-preload.c b/src/libumockdev-preload.c
index dddf7f3..1f24fdf 100644
--- a/src/libumockdev-preload.c
+++ b/src/libumockdev-preload.c
@@ -1531,7 +1531,11 @@ fclose(FILE * stream)
 }
 
 int
+#ifdef HAVE_POSIX_IOCTL
+ioctl(int d, int request, ...)
+#else
 ioctl(int d, unsigned long request, ...)
+#endif
 {
     libc_func(ioctl, int, int, unsigned long, ...);
     int result;
@@ -1549,13 +1553,21 @@ ioctl(int d, unsigned long request, ...)
 
     result = ioctl_emulate(d, request, arg);
     if (result != UNHANDLED) {
+#ifdef HAVE_POSIX_IOCTL
+       DBG(DBG_IOCTL, "ioctl fd %i request %i: emulated, result %i\n", d, request, result);
+#else
        DBG(DBG_IOCTL, "ioctl fd %i request %lX: emulated, result %i\n", d, request, result);
+#endif
        return result;
     }
 
     /* call original ioctl */
     result = _ioctl(d, request, arg);
+#ifdef HAVE_POSIX_IOCTL
+    DBG(DBG_IOCTL, "ioctl fd %i request %i: original, result %i\n", d, request, result);
+#else
     DBG(DBG_IOCTL, "ioctl fd %i request %lX: original, result %i\n", d, request, result);
+#endif
 
     if (result != -1 && ioctl_record_fd == d)
        record_ioctl(request, arg, result);
```

This works for me. Mostly copied from http://lists.busybox.net/pipermail/buildroot/2015-July/135135.html

z3ntu, 2018-08-19:
Maybe with this change in `configure.ac` the previous patch has to be reverted partially or adjusted?

martinpitt, 2018-08-19:
@z3ntu : Thanks, this looks mostly good. But it drops the check for `__xstat()`. Your libc has it, so apparently it's not part of a header file?

It would be nice to put that under CI, so that it doesn't break again in the future. musl is actually packaged in Debian and Ubuntu, so I gave that a try. Documenting the steps here for future reference:

 * `apt install -y musl-tools`
 * `./autogen.sh CC=musl-gcc LD=musl-ldd` → this fails on not finding `linux/input.h`. There is no linux-musl-dev package, even though musl-dev recommends it.
 * I created a symlink farm to work around this:
```
$ ls -l include/
total 0
lrwxrwxrwx 1 martin martin 34 19. Aug 21:32 asm -> /usr/include/x86_64-linux-gnu/asm/
lrwxrwxrwx 1 martin martin 24 19. Aug 21:24 asm-generic -> /usr/include/asm-generic
lrwxrwxrwx 1 martin martin 22 19. Aug 21:33 libudev.h -> /usr/include/libudev.h
lrwxrwxrwx 1 martin martin 19 19. Aug 21:23 linux -> /usr/include/linux/
```
and build with `CFLAGS="-I `pwd`/include"`.

 * This now at least reproduces the fopen() errors. This confirms that the musl libc.so *does* export all the `__xstat()` and `open64()` glory, so the previous configure check does not work. The header files just alias them like `#define open64 open`, which explains the `redefinition of ‘fopen’` errors.

 * Linking fails like below. I haven't yet found a workaround for this.
```
/bin/bash ./libtool  --tag=CC   --mode=link musl-gcc -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I ./src -I src -w -I /home/martin/upstream/umockdev/include    -o src/umockdev-run src/umockdev_run-umockdev-run.o -lgio-2.0 -lgobject-2.0 -lglib-2.0 libumockdev.la 
libtool: link: musl-gcc -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I ./src -I src -w -I /home/martin/upstream/umockdev/include -o src/.libs/umockdev-run src/umockdev_run-umockdev-run.o  -lgio-2.0 -lgobject-2.0 -lglib-2.0 ./.libs/libumockdev.so -pthread
cc: error: ./.libs/libumockdev.so: No such file or directory
```

martinpitt, 2018-08-19:
BTW, your patch "works" for you because it moves `HAVE_FILE64` to the new `HAVE_OPEN64`, which isn't being checked anywhere. Thus it effectively just disables these functions.

martinpitt, 2018-08-19:
As I'm now at least able to compile with musl headers, I fixed the `*64()` detection in commit and the ioctl request type in commit . As I wrote below, I can't link though, and thus don't run tests either. Without further hacks, it fails with
```
  CCLD     libumockdev.la
/usr/bin/ld: cannot find -lgio-2.0
/usr/bin/ld: cannot find -lgobject-2.0
/usr/bin/ld: cannot find -lglib-2.0
/usr/bin/ld: cannot find -ludev
collect2: error: ld returned 1 exit status
```
Probably because musl-gcc only looks in /usr/lib/x86_64-linux-musl/. When I use

    ./autogen.sh  CC=musl-gcc CFLAGS="-I `pwd`/include -L/lib/x86_64-linux-gnu -L/usr/lib/x86_64-linux-gnu/"

it fails with
```
/usr/bin/ld: src/.libs/libumockdev_la-umockdev.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: src/.libs/libumockdev_la-uevent_sender.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: nonrepresentable section on output
```
but when I actually add `-fPIC` it still fails the same way. So I'm still unable to put this into CI.

martinpitt, 2018-08-19:
> AC_CHECK_FUNCS_ONCE call should do the job at detecting whether it's available or not.

Not really, as musl actually *does* export symbols like `__xstat()`, it just doesn't provide headers for it. So that check won't work either. Either way, the guarding with `__GLIBC__` should do for now.

z3ntu, 2018-08-19:
```
src/ioctl_tree.c:48:22: error: format '%lX' expects argument of type 'long unsigned int', but argument 3 has type 'int' [-Werror=format=]
```

PureTryOut, 2018-08-20:
Awesome job, it now completely compiles! There are some warnings left, but at least we don't need any custom patches any more. Also thanks for setting up automated testing with Musl, I wish more software projects gave that much care to alternative setups to what they're used too!

martinpitt, 2018-08-20:
Nice! Note that I didn't actually set up CI for it, as I'm unable to get over that linker error (see above).

z3ntu, 2018-08-20:
I'm trying to set up a Docker image that builds it in an Alpine Linux environment, hopefully that can be used in the CI.
The reason you get linker errors is that the dependencies for umockdev are compiled for glibc and not for musl so it's probably a pain to get everything running under a non-musl system except you want to compile everything yourself.

z3ntu, 2018-08-20:
`Dockerfile`:
```Dockerfile
FROM alpine:3.8

RUN mkdir -p /app
WORKDIR /app

RUN apk add --no-cache autoconf automake libtool gcc musl-dev glib-dev eudev-dev libgudev-dev make vala linux-headers
```
Commands to run:
```sh
$ docker build -t umockdev-env .
$ docker run --rm -it -v /path/to/umockdev:/app umockdev-env /bin/sh -c './autogen.sh && make'
```

martinpitt, 2018-08-20:
Awesome, thanks @z3ntu! I am adding that in a [branch](https://github.com/martinpitt/umockdev/tree/alpine-test), and got a first result: https://travis-ci.org/martinpitt/umockdev/jobs/418396791

This reproduces the warnings, and I get a test failure in `t_testbed_dev_access()`, so some function isn't properly wrapped for musl. I can reproduce this just fine locally, so it's now fairly easy for me to hack on this. I'll land the branch once tests become green.

martinpitt, 2018-08-20:
I fixed all the warnings and figured out the wrong `stat()` result in the branch, now checks are getting [much further](https://travis-ci.org/martinpitt/umockdev/jobs/418421205). Now some ioctl related tests fail, but everything else succeeds:
```
ERROR:/source/tests/test-umockdev.c:1127:t_testbed_usb_lsusb: assertion failed: (g_str_has_prefix(out, "\nBus 001 Device 001: ID 04a9:31c0 Canon, Inc. PowerShot SX200 IS\n"))

/umockdev-ioctl-tree/type_get_by: **
ERROR:/source/tests/test-ioctl-tree.c:105:t_type_get_by: assertion failed (t->id == EVIOCGABS(0)): (-9223372036854775808 == 2149074240)
```
This actually sounds very closely related to the `int` vs. `unsigned long` type change. Debugging this is not a  small thing, too late for tonight. I'll try to look at it tomorrow.

martinpitt, 2018-08-22:
There, I just had to disable one test, and work around #19. CI is there now: https://travis-ci.org/martinpitt/umockdev/builds/419057274

So building and `make check` ought to work for you now.

Thanks for your help! I'll look into the remaining issues soon.

PureTryOut, 2018-08-22:
Awesome job! Thanks for your attention to the problem!

However I am still getting a failed test somehow.

```
/umockdev-testbed/add_from_string_dev_char: **
ERROR:src/umockdev.c:5076:umockdev_testbed_create_node_for_device: assertion failed: (Linux.openpty (out ptym, out ptys, ptyname_array, null, null) == 0)
/bin/bash: line 1: 25541 Aborted                 TOP_BUILDDIR=. TOP_SRCDIR=. LD_LIBRARY_PATH=./.libs:$LD_LIBRARY_PATH GI_TYPELIB_PATH=. PATH=./src:$PATH G_SLICE=debug-blocks MALLOC_PERTURB_=85 MALLOC_CHECK_=3 ./src/umockdev-wrapper $f
make[3]: *** [Makefile:2149: check-local] Error 134
make[3]: Leaving directory '/home/pmos/build/src/umockdev-66f8cdac6b77eff89a24ba981de39a3e693740b6'
make[2]: *** [Makefile:1913: check-am] Error 2
make[2]: Leaving directory '/home/pmos/build/src/umockdev-66f8cdac6b77eff89a24ba981de39a3e693740b6'
make[1]: *** [Makefile:1623: check-recursive] Error 1
make[1]: Leaving directory '/home/pmos/build/src/umockdev-66f8cdac6b77eff89a24ba981de39a3e693740b6'
make: *** [Makefile:1915: check] Error 2
```

martinpitt, 2018-08-22:
I worked on the two test workarounds mentioned above tonight, and am happy to report that with latest master there are now no disabled tests any more, and they 
[all succeed](https://travis-ci.org/martinpitt/umockdev/builds/419361571). So from now on, no regressions on Alpine any more :)

@PureTryOut : In what environment are you testing this? That `openpty()` failure smells like /dev/pts/ptmx does not exist (missing `/dev/pts` mount?) or is not accessible (it should have 0666 permissions). I tried chmod'ing the file to 600, that didn't reproduce it here.

Indeed that failure mode is a bit unhelpful, though. In my [fixes branch](https://github.com/martinpitt/umockdev/tree/fixes) I added a commit to show the actual error from openpty - can you please try that one and see what it says?

PureTryOut, 2018-08-22:
Ah missing mount point might be right. I'm testing it in a chroot automatically made by our [pmbootstrap](https://gitlab.com/postmarketos/pmbootstrap) tool. I'll report this to our own issue tracker then.

martinpitt, 2018-08-23:
@PureTryOut : if you re-run the test in the mentioned fixes branch, you should get a more helpful error message to point to the problem.

martinpitt, 2018-08-26:
This is now released in [0.12](https://github.com/martinpitt/umockdev/releases/tag/0.12), for easy downstream packaging.

z3ntu, 2018-10-05:
Hey again, the following fix is needed to compile it on armhf. I don't know if it's correct and does what is intended, but it compiles on armhf which should be good enough ;)
```
diff --git a/src/ioctl_tree.c b/src/ioctl_tree.c
index 3a5d877..6d8a949 100644
--- a/src/ioctl_tree.c
+++ b/src/ioctl_tree.c
@@ -474,7 +474,7 @@ id_matches_type(IOCTL_REQUEST_TYPE id, const ioctl_type *type)
 static void
 ioctl_simplestruct_init_from_bin(ioctl_tree * node, const void *data)
 {
-    DBG(DBG_IOCTL_TREE, "ioctl_simplestruct_init_from_bin: %s(%X): size is %lu bytes\n", node->type->name, (unsigned) node->id, NSIZE(node));
+    DBG(DBG_IOCTL_TREE, "ioctl_simplestruct_init_from_bin: %s(%X): size is %X bytes\n", node->type->name, (unsigned) node->id, NSIZE(node));
     node->data = malloc(NSIZE(node));
     memcpy(node->data, data, NSIZE(node));
 }
@@ -489,7 +489,7 @@ ioctl_simplestruct_init_from_text(ioctl_tree * node, const char *data)
     node->data = malloc(data_len);
 
     if (NSIZE(node) != data_len) {
-	DBG(DBG_IOCTL_TREE, "ioctl_simplestruct_init_from_text: adjusting ioctl ID %X (size %lu) to actual data length %zu\n",
+	DBG(DBG_IOCTL_TREE, "ioctl_simplestruct_init_from_text: adjusting ioctl ID %X (size %X) to actual data length %zu\n",
 	    (unsigned) node->id, NSIZE(node), data_len);
 	node->id = _IOC(_IOC_DIR(node->id), _IOC_TYPE(node->id), _IOC_NR(node->id), data_len);
     }

```

commit bf67e4d3 (2018-08-17): Only wrap and test canonicalize_file_name() with glibc

This is a GNU extension.

Fixes #83
commit 7078dc16 (2018-08-18): test: Fix building with musl libc

musl libc doesn't actually `#define __MUSL__`, so turn this around and
only verify the error on glibc.

Fixes #83