martinpitt/umockdev #91: Build of 0.13 fails in coverage rules

rossburton, 2019-07-22:
configure reports:
```
checking for GNOME code coverage support... yes
checking whether to build with code coverage support... no
```
(remove the GNOME mention, as the GNOME rules are not used anymore)

Then make:
```
Makefile:2233: *** missing separator.  Stop.
```
The Makefile looks like this:
```
2231   │ check-valgrind:
2232   │     EXEC_NAME="valgrind --leak-check=full --show-possibly-lost=no -q" G_DEBUG=gc-friendly $(MAKE) check
2233   │ @CODE_COVERAGE_RULES@
```

According to [the documentation](https://www.gnu.org/software/autoconf-archive/ax_code_coverage.html) `@CODE_COVERAGE_RULES@` should be replaced with `  include $(top_srcdir)/aminclude_static.am`.  Personally I'd also drop the `m4_ifdef` and just copy the file from autoconf-archive.

martinpitt, 2019-07-22:
I can't reproduce this in Debian sid, I tried all combinations of "no autoconf-archive", "autoconf-archive but no lcov", and "autoconf-archive and lcov". Your configure output looks like you do  have autoconf-archive installed, but  did not configure with `--enable-code-coverage`.

So apparently your version of autoconf-archive does define `AX_CODE_COVERAGE`, but does not define `CODE_COVERAGE_RULES`. This  seems to coincide with your doc reference vs. the local documentation in /usr/share/aclocal/ax_code_coverage.m4 which still does document `@CODE_COVERAGE_RULES@`.

Curiously, with the tarball (instead of git with ./autogen.sh), `checking for GNOME code coverage support...` says "no" for me even with autoconf-archive installed.

Indeed this is too unwieldy (meh, there haven't been problems with the GNOME version..). The disted tarball ought to ship ax_code_coverage.m4, but I most definitively don't want to track it in upstream git. I'll check how to convince autogen/autoconf to copy it.

martinpitt, 2019-07-22:
Oh, the tarball says "no" because that's hardcoded into the built configure -- the tarball gets generated by Travis, in an environment  without autoconf-archive.

So how and where (OS, installed packages) exactly are you experiencing this?

rossburton, 2019-07-22:
Built in Yocto Project, from your tarball, after a fresh autoreconf, with autoconf-archive 2019.01.06.

rossburton, 2019-07-22:
Google suggests that AX_CODE_COVERAGE changed its API: https://gitlab.freedesktop.org/dbus/dbus/commit/2938c2125ebcd001e470aeac1ffac45b6b1ebe89

martinpitt, 2019-07-22:
Should be fixed with https://github.com/martinpitt/umockdev/commit/9b9b31b32952, I'll make a new 0.13.1 release now.

closed 2019-07-22 by martinpitt

martinpitt, 2019-07-22:
Released: https://github.com/martinpitt/umockdev/releases/tag/0.13.1

commit 2fdc45e0 (2019-07-22): Ship code coverage autoconf macros in dist tarball

This will enable code coverage for releases as well (install
autoconf-archive into the release tarball build environment for that),
and avoids incompatibilities between the version installed on the target
system and release build system.

Fixes #91
commit 94dbb0eb (2019-07-22): Ship code coverage autoconf macros in dist tarball

This will enable code coverage for releases as well (install
autoconf-archive into the release tarball build environment for that),
and avoids incompatibilities between the version installed on the target
system and release build system.

Fixes #91
commit 9b9b31b3 (2019-07-22): Ship code coverage autoconf macros in dist tarball

This will enable code coverage for releases as well (install
autoconf-archive into the release tarball build environment for that),
and avoids incompatibilities between the version installed on the target
system and release build system.

Provide a dummy fallback for the Alpine and CentOS CI containers.

Fixes #91