martinpitt/umockdev #96: I couldn't record to *.ioctl file anything

msergenergen, 2019-09-20:
I want to mock touchpad. But I didn't record anything into ioctl file.
When I was checking to mouse.ioctl file, I saw only this part
> @DEV /dev/input/event5
Probably, I didn't record anything. But I didn't get any error messages after following step;
`sudo umockdev-record -i /dev/input/event5=mouse.ioctl -e /dev/input/event5=mouse.events -- evtest /dev/input/event5`

 But I got this error message after following step;
`umockdev-run -d mouse.umockdev -i /dev/input/event5=mouse.ioctl -e /dev/input/event5=mouse.events -- evtest /dev/input/event5`

> ERROR: libumockdev-preload: failed to load ioctl record file for /dev/input/event5: empty or invalid format

Do you want to recommend anything else?

msergenergen, 2019-09-20:
I tried running on another pc and devices (keyboard, external mouse, touchpad) (umockdev installation). But I didn't get any result.

msergenergen, 2019-09-27:
Also I tried install and run steps on another Ubuntu versions. (16,04 and 18,04) I got same error.

DanielMalangu, 2020-04-28:
Yes, I have this exact issue. I'm not sure why no one has attempted to provide a solution. Is this a problem that is not common at all?

msergenergen, 2020-04-30:
Probably many people did not occur this problem.

hadess, 2020-11-03:
You're missing one half of the commands, you also need to create the `.umockdev` file:
https://github.com/martinpitt/umockdev/blob/master/README.rst#record-and-replay-input-devices

(even though that doesn't work for me either)

hadess, 2020-11-04:
Nothing is getting saved to disk:
```
$ sudo umockdev-record /dev/input/event18 > lap-proximity-switch.umockdev
$ sudo UMOCKDEV_DEBUG=all umockdev-record -i /dev/input/event18=lap-proximity-switch.ioctl -e /dev/input/event18=lap-proximity-switch.events -- evtest /dev/input/event18 
testbed wrapped open(/dev/input/event18) -> /dev/input/event18
ioctl_record_open: Starting new record lap-proximity-switch.ioctl
ioctl_record_open: starting ioctl recording of fd 3 into lap-proximity-switch.ioctl
init_script_dev_logfile_map: will record script of device 13:82 into lap-proximity-switch.events
script_record_open: start recording fd 3 on device 13:82 into lap-proximity-switch.events (format 1)
script_start_record: Starting new record of format 1
isatty(1): is a terminal, ttyname /dev/pts/2
isatty(1): readlink((null)/dev/.ptymap/_dev_pts_2) failed: No such file or directory
ioctl fd 3 request 80044501: original, result 0
ioctl_simplestruct_init_from_bin: EVIOCGVERSION(80044501): size is 4 bytes
Input driver version is 1.0.1
ioctl fd 3 request 80084502: original, result 0
ioctl_simplestruct_init_from_bin: EVIOCGID(80084502): size is 8 bytes
Input device ID: bus 0x6 vendor 0x0 product 0x0 version 0x0
ioctl fd 3 request 81004506: original, result 28
ioctl_simplestruct_init_from_bin: EVIOCGNAME(81004506): size is 256 bytes
Input device name: "Thinkpad proximity switches"
ioctl fd 3 request 801F4520: original, result 8
ioctl_simplestruct_init_from_bin: EVIOCGBIT(801F4520): size is 31 bytes
Supported events:
  Event type 0 (EV_SYN)
ioctl fd 3 request 9800451B: original, result 8
ioctl_simplestruct_init_from_bin: EVIOCGSW(9800451B): size is 6144 bytes
  Event type 5 (EV_SW)
ioctl fd 3 request 82FF4525: original, result 8
ioctl_simplestruct_init_from_bin: EVIOCGBIT(82FF4525): size is 767 bytes
    Event code 17 (?) state 0
ioctl fd 3 request 80F84509: original, result 8
ioctl_simplestruct_init_from_bin: EVIOCGPROP(80F84509): size is 248 bytes
Properties:
Testing ... (interrupt to exit)
ioctl fd 3 request 40044590: original, result 0
ioctl fd 3 request 40044590: original, result 0
script_record_op r: got 48 bytes on fd 3 (format 1)
Event: time 1604491485.311818, type 5 (EV_SW), code 17 (?), value 1
Event: time 1604491485.311818, -------------- SYN_REPORT ------------
script_record_op r: got 48 bytes on fd 3 (format 1)
Event: time 1604491487.536091, type 5 (EV_SW), code 17 (?), value 0
Event: time 1604491487.536091, -------------- SYN_REPORT ------------
^Cioctl fd 3 request 40044590: original, result -1
$ cat *ioctl
@DEV /dev/input/event18
```

Looks like `DBG(DBG_IOCTL, "ioctl_record_sigint_handler: got signal %i, flushing record\n", signum);` is never getting triggered. I also tested running in a root shell, rather than under `sudo`, but the signal handler is not getting triggered.

hadess, 2020-11-04:
The interrupt handler in evtest is however getting triggered. Removing the signal handlers in evtest fixes it.

It might be useful to find a way to install the signal handler in a more robust way in `libumockdev-preload.c`. Right now, the signal handler in the preload is put in place before the one in evtest, which overrides it:
```
$ sudo UMOCKDEV_DEBUG=all LD_LIBRARY_PATH=/home/hadess/Projects/gnome-install/lib/ umockdev-record -i /dev/input/event18=lap-proximity-switch.ioctl -e /dev/input/event18=lap-proximity-switch.events -- /home/hadess/.cache/scratch/jhbuild/build/evtest/evtest /dev/input/event18
<snip>
ioctl_record PUTTING SIGNAL HANDLER IN PLACE
<snip>
PUTTING EVTEST SIGNAL HANDLERS IN PLACE
^Cinterrupt_handler called
ioctl fd 3 request 40044590: original, result -1
```

hadess, 2020-11-04:
See also https://gitlab.freedesktop.org/libevdev/evtest/-/issues/1

martinpitt, 2020-11-12:
I'm of course open to suggestions how else umockdev-record (i.e. the preload library) could be told to stop recording and start writing the file. SIGINT seemed like an obvious choice back then, but indeed signal handlers don't chain well. Another possibility is to add some timeout to evtest and just let that exit cleanly?

georgiablanco, 2021-05-10:
Hello. I am having the same problem. Has this issue been resolved? Or has anyone got a work around?

martinpitt, 2021-05-10:
@georgiablanco : It has not been resolved, no. This will require some bigger reworking which didn't land yet.

(I tried to reply to your email, and your hotmail server rejected it :shrug: )

ethanfowler, 2021-05-10:
Just to check my understanding of the situation, correct me if I'm wrong:

Because `evtest` registers its own SIGINT handler, `umockdev-record` doesn't close cleanly and write the recorded data to disk?

Does this mean that https://github.com/martinpitt/umockdev#record-and-replay-input-devices does not currently work?

Is there an `evtest` alternative or workaround at current?

commit e4ea3a71 (2021-06-01): Port ioctl recording to go through the main process

Note that this changes the ioctl_tree_insert API to be more vala
friendly. i.e. it will return the new tree pointer, and it will always
take the passed tree node, free'ing it if a duplicate has been found.

This fixes storing the record if the process overrides SIGINT for
ioctl's.

Fixes #96
closed 2021-06-01 by martinpitt