martinpitt/umockdev #97: umockdev-run hits assert with device file containing storage device

dkl, 2020-01-02:
Hi,

The latest umockdev-run aborts when run with the attached example device file [example.txt](https://github.com/martinpitt/umockdev/files/4016537/example.txt). (This is actually part of a bigger list produced by umockdev-record, but it seems to be only the entries for the USB mass storage device that trigger the problem.)

```
$ src/umockdev-run -d example.txt 
**
ERROR:src/umockdev.c:5200:umockdev_testbed_add_dev_from_string: assertion failed: (FileUtils.symlink(devnode_path, devnode_links[i]) == 0)
Bail out! ERROR:src/umockdev.c:5200:umockdev_testbed_add_dev_from_string: assertion failed: (FileUtils.symlink(devnode_path, devnode_links[i]) == 0)
Aborted (core dumped)
```

This used to work in version 0.12; git bisect pointed to 7c0671cc142cab8101f27b4a8711cca47a74972a.

martinpitt, 2020-01-05:
Thanks for the report! The device dump is indeed invalid, as it's a duplicated symlink:
```
P: /devices/soc0/soc/2100000.aips-bus/2184200.usb/ci_hdrc.1/usb1/1-1/1-1.3/1-1.3:1.0/host9/target9:0:0/9:0:0:0/block/sda/sda1
S: disk/by-label/Xubuntu\x2016.04.3\x20LTS\x20amd64

P: /devices/soc0/soc/2100000.aips-bus/2184200.usb/ci_hdrc.1/usb1/1-1/1-1.3/1-1.3:1.0/host9/target9:0:0/9:0:0:0/block/sda
S: disk/by-label/Xubuntu\x2016.04.3\x20LTS\x20amd64
```
I. e. the `disk/by-label/Xubuntu\x2016.04.3\x20LTS\x20amd64` symlink can only point to either sda or (much more plausibly) sda1. It's very unusual that the raw block device has a file system label, as in your case the first partition is the actual file system. So I wonder how that happened in the first place?

This could have gotten there due to manual editing, did you do that? Or because of a bug in umockdev-record, or in udev itself. First, can you please check `sudo blkid -p /dev/sda /dev/sda1`? sda should not have a label. And presumably your real /dev/disk/by-label/Xubuntu\x2016.04.3\x20LTS\x20amd64 points to sda1, not sda?

Either way, umockdev should warn about this, not crash. I have a fix for that queued up, I just need to sort out some (unrelated) CI problems until I can land it.

martinpitt, 2020-01-05:
Reopening, as I'm still interested in the root cause of this.

dkl, 2020-01-07:
Thanks for the fix! I did some more tests. It was reproducible with a hybrid ISO image (a Debian live image in this case) on the USB stick, but not with a "normally" partitioned USB stick.

```
$ sudo lsblk -f /dev/sde /dev/sdf
NAME   FSTYPE  LABEL                  UUID                                 FSAVAIL FSUSE% MOUNTPOINT
sde    iso9660 d-live 10.2.0 st amd64 2019-11-16-10-15-29-00
├─sde1 iso9660 d-live 10.2.0 st amd64 2019-11-16-10-15-29-00
└─sde2 vfat    d-live 10.2.0 st amd64 DEB0-0001
sdf
└─sdf1 vfat    MYPARTLABEL            6A89-A80A
$ ls -l /dev/disk/by-label/MYPARTLABEL /dev/disk/by-label/d-live\\x2010.2.0\\x20st\\x20amd64 
lrwxrwxrwx 1 root root 10 Jan  7 15:44  /dev/disk/by-label/MYPARTLABEL -> ../../sdf1
lrwxrwxrwx 1 root root 10 Jan  7 15:57 '/dev/disk/by-label/d-live\x2010.2.0\x20st\x20amd64' -> ../../sde1
$ sudo blkid -p /dev/sde /dev/sde1
/dev/sde: UUID="2019-11-16-10-15-29-00" BOOT_SYSTEM_ID="EL TORITO SPECIFICATION" LABEL="d-live 10.2.0 st amd64" TYPE="iso9660" USAGE="filesystem" PTUUID="1bf56f01" PTTYPE="dos"
/dev/sde1: UUID="2019-11-16-10-15-29-00" BOOT_SYSTEM_ID="EL TORITO SPECIFICATION" LABEL="d-live 10.2.0 st amd64" TYPE="iso9660" USAGE="filesystem" PTUUID="1bf56f01" PTTYPE="dos" PART_ENTRY_SCHEME="dos" PART_ENTRY_UUID="1bf56f01-01" PART_ENTRY_TYPE="0x0" PART_ENTRY_FLAGS="0x80" PART_ENTRY_NUMBER="1" PART_ENTRY_OFFSET="0" PART_ENTRY_SIZE="1738752" PART_ENTRY_DISK="8:64"
$ sudo blkid -p /dev/sdf /dev/sdf1
/dev/sdf: PTUUID="44056f5c" PTTYPE="dos"
/dev/sdf1: LABEL_FATBOOT="MYPARTLABEL" LABEL="MYPARTLABEL" UUID="6A89-A80A" VERSION="FAT32" TYPE="vfat" USAGE="filesystem" PART_ENTRY_SCHEME="dos" PART_ENTRY_UUID="44056f5c-01" PART_ENTRY_TYPE="0xb" PART_ENTRY_NUMBER="1" PART_ENTRY_OFFSET="2048" PART_ENTRY_SIZE="15130624" PART_ENTRY_DISK="8:80"
$ umockdev-record /dev/sde /dev/sde1 >sde.txt
$ umockdev-record /dev/sdf /dev/sdf1 >sdf.txt
$ ./src/umockdev-run -d sde.txt echo foo

** (umockdev-run:17300): WARNING **: 15:59:51.227: umockdev.vala:1205: failed to create /tmp/umockdev.OYG5D0/dev/disk/by-uuid/2019-11-16-10-15-29-00 -> /tmp/umockdev.OYG5D0/dev/sde1 symlink for device /devices/pci0000:00/0000:00:07.1/0000:27:00.3/usb4/4-4/4-4.4/4-4.4.2/4-4.4.2:1.0/host12/target12:0:0/12:0:0:0/block/sde/sde1: File exists

** (umockdev-run:17300): WARNING **: 15:59:51.227: umockdev.vala:1205: failed to create /tmp/umockdev.OYG5D0/dev/disk/by-label/d-live\x2010.2.0\x20st\x20amd64 -> /tmp/umockdev.OYG5D0/dev/sde1 symlink for device /devices/pci0000:00/0000:00:07.1/0000:27:00.3/usb4/4-4/4-4.4/4-4.4.2/4-4.4.2:1.0/host12/target12:0:0/12:0:0:0/block/sde/sde1: File exists
foo
$ ./src/umockdev-run -d sdf.txt echo foo
foo
```

[sde.txt](https://github.com/martinpitt/umockdev/files/4031169/sde.txt)
[sdf.txt](https://github.com/martinpitt/umockdev/files/4031170/sdf.txt)

martinpitt, 2020-01-07:
Interesting, thanks! So udev is Not Wrong™, this is indeed a weird situation. So I think the fix here is adequate, one gets a warning, but it doesn't crash any more.

commit 80c0e0c8 (2020-01-05): Fix crash on duplicate S: link targets

These are usually a bug in the device description, not an internal
programming error.

Fixes #97
commit 7a348ab5 (2020-01-05): Fix crash on duplicate S: link targets

These are usually a bug in the device description, not an internal
programming error.

Fixes #97
closed 2020-01-07 by martinpitt