martinpitt/umockdev #98: Errors building umockdev on s390x rawhide

hadess, 2020-02-07:
Full build log of 0.13.2 at https://kojipkgs.fedoraproject.org//work/tasks/4866/41324866/build.log

```
In file included from /usr/include/stdio.h:867,
                 from src/libumockdev-preload.c:37:
In function 'fprintf',
    inlined from 'openat' at src/libumockdev-preload.c:1401:1:
/usr/include/bits/stdio2.h:100:10: error: '%s' directive argument is null [-Werror=format-overflow=]
  100 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  101 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
In function 'fprintf',
    inlined from 'openat64' at src/libumockdev-preload.c:1406:1:
/usr/include/bits/stdio2.h:100:10: error: '%s' directive argument is null [-Werror=format-overflow=]
  100 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  101 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
  CCLD     tests/test-umockdev
In function 'fprintf',
    inlined from 'readlinkat' at src/libumockdev-preload.c:1434:5:
/usr/include/bits/stdio2.h:100:10: error: '%s' directive argument is null [-Werror=format-overflow=]
  100 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  101 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
/usr/include/bits/unistd.h: In function 'readlinkat':
src/libumockdev-preload.c:1434:55: note: format string is defined here
 1434 |     DBG(DBG_PATH, "testbed wrapped readlinkat (%s) -> %s\n", pathname, p);
      |                                                       ^~
g-ir-compiler -l libumockdev.so.0 -o UMockdev-1.0.typelib UMockdev-1.0-hacked.gir
In file included from /usr/include/stdio.h:867,
                 from src/libumockdev-preload.c:37:
In function 'fprintf',
    inlined from 'script_start_record' at src/libumockdev-preload.c:802:2,
    inlined from 'script_start_record' at src/libumockdev-preload.c:782:1,
    inlined from 'script_record_connect' at src/libumockdev-preload.c:927:3,
    inlined from 'connect' at src/libumockdev-preload.c:1594:5:
/usr/include/bits/stdio2.h:100:10: error: '%s' directive argument is null [-Werror=format-overflow=]
  100 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  101 |    __va_arg_pack ());
      |    ~~~~~~~~~~~~~~~~~
/usr/include/sys/socket.h: In function 'connect':
src/libumockdev-preload.c:802:91: note: format string is defined here
  802 |  DBG(DBG_SCRIPT, "script_start_record: Appending to existing record of format %i for path %s\n", fmt, recording_path);
      |                                                                                           ^~
  CC       src/umockdev_run-umockdev-run.o
cc1: some warnings being treated as errors
```

martinpitt, 2020-02-08:
Thanks for the report! 0.13.12 still [built fine in September](https://koji.fedoraproject.org/koji/buildinfo?buildID=1370137), this looks like the compiler got stricter. This doesn't happen on any other arch, so it's a bit hard for me to reproduce.

However, it seems quite obvious that gcc 9 enables some static code checking that complains when printed `%s` values can be NULL. That it indeed the case and was intended, but I'll change the code to avoid it and print "(null)" explicitly as a fallback.

martinpitt, 2020-02-08:
I found similar issues in [systemd](https://github.com/systemd/systemd/issues/12534) and [bcal](https://github.com/jarun/bcal/issues/30). In between these it seemed that building with `-O3` and/or with various options like `CFLAGS="-g -O3 -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong` would enable this, but unfortunately not.

This is with gcc 10.0.1 20200130 as of today's rawhide. So I suppose I need to fix this "blindly".

martinpitt, 2020-02-08:
I tried cross-compiling with
```
./configure  CC=s390x-linux-gnu-gcc  CFLAGS="-g -O3 -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1  -fasynchronous-unwind-tables -fstack-clash-protection" --host=s390x-linux-gnu
```
but that fails very early on
```
configure:2665: checking whether the C compiler works
configure:2687: s390x-linux-gnu-gcc -g -O3 -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1  -fasynchronous-unwind-tables -fstack-clash-protection   conftest.c  >&5
/usr/bin/s390x-linux-gnu-ld: cannot find crt1.o: No such file or directory
/usr/bin/s390x-linux-gnu-ld: cannot find crti.o: No such file or directory
/usr/bin/s390x-linux-gnu-ld: cannot find -lc
/usr/bin/s390x-linux-gnu-ld: cannot find crtn.o: No such file or directory
```
and neither in the Fedora wiki nor Google I can find how to do this. Only recipes for the kernel, which doesn't need libc.

martinpitt, 2020-02-08:
Took a few attempts, but commit db7278c fixes it: [scratch build](https://koji.fedoraproject.org/koji/taskinfo?taskID=41435866)

closed 2020-02-08 by martinpitt

martinpitt, 2020-02-08:
@hadess: I did a new release: https://github.com/martinpitt/umockdev/releases/tag/0.14

hadess, 2020-02-09:
Sorry I didn't get to fixing this. The new version works though:
https://koji.fedoraproject.org/koji/taskinfo?taskID=41435962
even on s390x

martinpitt, 2020-02-09:
@hadess: Nice, thanks for packaging it!

commit d7ede8e0 (2020-02-08): preload: Avoid printing NULL strings in debug statements

Recent gcc on s390x failed on these with

    error: '%s' directive argument is null [-Werror=format-overflow=]

Fixes #98
commit 7d155a58 (2020-02-08): preload: Avoid printing NULL strings in debug statements

Recent gcc on s390x failed on these with

    error: '%s' directive argument is null [-Werror=format-overflow=]

Fixes #98
commit db7278ca (2020-02-08): preload: Avoid printing NULL strings in debug statements

Recent gcc on s390x failed on these with

    error: '%s' directive argument is null [-Werror=format-overflow=]

Fixes #98