martinpitt/umockdev #99: Having issues trying to simulate a ttyACM0 device

joaofl, 2020-02-14:
I've been trying to use the umockdev (v14.1) lib to mock a ttyACM0 device on an ARM based platform with Yocto. I first included the recipe, compiled, deployed and installed it on the remote host, and so far it seems to work just fine.

The target device is connected to a second device through USB, and a C program interfaces /manages  a scanner of RFIDs connected to /dev/ttyACM0. 

In order to get the traces a first run:
```
umockdev-record /dev/ttyACM0 > ttyACM0.umockdev
umockdev-record -s /dev/ttyACM0=ttyACM0.script -i /dev/ttyACM0=ttyACM0.ioctl -- /home/root/scanner-reader
```
and after recording the main operations I want to reproduce, I try to run it with:
```
umockdev-run -d ttyACM0.umockdev -s /dev/ttyACM0=ttyACM0.script  -- /home/root/scanner-reader
```
However, as the `scanner-reader` progrtam loads,  I'm getting errors. Mainly in two parts of the code:
```
if (ioctl(fd, FIONREAD, &bytesAvailable) < 0) {
        LOG_WARN("ioctl(FIONREAD) for %s failed with errno %d: %s", portName, errno, strerror(errno));
``` 
throws the error:
`ioctl(FIONREAD) for /dev/ttyACM0 failed with errno 25: Inappropriate ioctl for device`

And when I try to read the file descriptor fd with ```read(fd, buffer, maxLength)```

I get the error:
`Read error on /dev/ttyACM0 (9): Bad file descriptor`

Do you have any Idea why, and or suggestion on how to overcome this?

Thanks in advance!
By the way, congrats for the nice tool!

martinpitt, 2020-04-25:
I'm afraid the `FIONREAD` ioctl isn't implemented in umockdev.  This also isn't trivial to add, it would essentially require reimplementing a significant part of the tty layer. ioctl emulation is rather difficult for a program like umockdev, and I only really did it for usbdevfs and evdev (as I needed these for my projects). Contributions appreciated of course :wink: 

I suppose your recorded ttyACM0.ioctl doesn't actually contain anything, or much of value?

script recording on a TTY (without ioctls) should work though. Does your ttyACM0.script look reasonable? Can you run the `umockdev-run scanner-reader` call with `UMOCKDEV_DEBUG=script` and put the output here? Adding an strace in between also can't hurt to see what it's trying to do.

benzea, 2021-06-25:
You could look at #125, or, if you are not dealing with a lot of ioctl's also possibly emulating it yourself once there is public API to do so in 0.16, see #132.

martinpitt, 2021-12-12:
@joaofl Do the two comments above help? Is this still relevant with current versions?

joaofl, 2021-12-13:
Thanks @benzea for the tip, and @martinpitt for checking this out.
Back when I was working with that, I could not emulate the device successful, with umockdev, so I implemented a tool to parse and log the data input, such that I could play it back later on. So honestly speaking, I did not check if @benzea tips could address that problem.

martinpitt, 2021-12-13:
Thanks @joaofl . I close this for now, I'm happy to reopen if/when you ever get back to this, just yell here please.

closed 2021-12-13 by martinpitt