martinpitt/umockdev #100: Using umockdev on USB3Vision devices

EmmanuelP, 2020-04-09:
Hi,

I'm trying to mock a USB3Vision device for the testing of a library I'm developing. The library is aravis: https://github.com/AravisProject/aravis

This library relies on the iManufacturer, iProduct and iSerial fields in the USB device description. It looks like this fields are not recorded properly, as seen in this lsusb output with the mocked device:

```
umockdev-run --device pointgrey.umockdev -- lsusb -vvvv -d 1e10:4000
Bus 003 Device 013: ID 1e10:4000 Point Grey Research, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1e10 Point Grey Research, Inc.
  idProduct          0x4000 
  bcdDevice            0.00
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              152mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               4 (error)
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface              5 (error)
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 06 07 08 09 0a 0b 03 0c 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
[... 172 lines cut ...]
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat          32 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled
```

Is this a known issue ?

rgetz, 2020-05-24:
I can confirm the issues on master.

```
rgetz@brain:~/github/umockdev/src$ lsusb | grep -i analog
Bus 003 Device 045: ID 0456:b673 Analog Devices, Inc. 
rgetz@brain:~/github/umockdev/src$ ./umockdev-record /dev/bus/usb/003/045 > pluto.umockdev
rgetz@brain:~/github/umockdev/src$ ./umockdev-record --ioctl /dev/bus/usb/003/045=pluto.ioctl -- lsusb -vvvv -d 0456:b673 > lusb.rec
can't get debug descriptor: Resource temporarily unavailable
```
unplug the device
```
rgetz@brain:~/github/umockdev/src$ ./umockdev-run --device pluto.umockdev --ioctl /dev/bus/usb/003/045=pluto.ioctl -- lsusb -vvvv -d 0456:b673 > lusb.run
can't get device qualifier: Inappropriate ioctl for device
can't get debug descriptor: Resource temporarily unavailable
```

and the diff between the runs is:

```
rgetz@brain:~/github/umockdev/src$ diff -u ./lusb.run ./lusb.rec 
--- ./lusb.run	2020-05-24 13:16:47.951583083 -0400
+++ ./lusb.rec	2020-05-24 13:16:14.011378233 -0400
@@ -11,9 +11,9 @@
   idVendor           0x0456 Analog Devices, Inc.
   idProduct          0xb673 
   bcdDevice            4.14
-  iManufacturer           1 (error)
-  iProduct                2 (error)
-  iSerial                 3 (error)
+  iManufacturer           1 Analog Devices Inc.
+  iProduct                2 PlutoSDR (ADALM-PLUTO)
+  iSerial                 3 104473222a87000618000600473ed57ae0
   bNumConfigurations      1
 OTG Descriptor:
   bLength                 3
@@ -27,7 +27,7 @@
     wTotalLength       0x00da
     bNumInterfaces          6
     bConfigurationValue     1
-    iConfiguration          4 (error)
+    iConfiguration          4 RNDIS/MSD/ACM/IIOUSBD
     bmAttributes         0x80
       (Bus Powered)
     MaxPower              500mA
@@ -39,7 +39,7 @@
       bFunctionClass          2 Communications
       bFunctionSubClass       6 Ethernet Networking
       bFunctionProtocol       0 
-      iFunction               7 (error)
+      iFunction               7 RNDIS
     Interface Descriptor:
       bLength                 9
       bDescriptorType         4
@@ -49,7 +49,7 @@
       bInterfaceClass         2 Communications
       bInterfaceSubClass      2 Abstract (modem)
       bInterfaceProtocol    255 Vendor Specific (MSFT RNDIS?)
-      iInterface              5 (error)
+      iInterface              5 RNDIS Communications Control
       CDC Header:
         bcdCDC               1.10
       CDC Call Management:
@@ -79,7 +79,7 @@
       bInterfaceClass        10 CDC Data
       bInterfaceSubClass      0 
       bInterfaceProtocol      0 
-      iInterface              6 (error)
+      iInterface              6 RNDIS Ethernet Data
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -109,7 +109,7 @@
       bInterfaceClass         8 Mass Storage
       bInterfaceSubClass      6 SCSI
       bInterfaceProtocol     80 Bulk-Only
-      iInterface              9 (error)
+      iInterface              9 Mass Storage
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -138,7 +138,7 @@
       bFunctionClass          2 Communications
       bFunctionSubClass       2 Abstract (modem)
       bFunctionProtocol       1 AT-commands (v.25ter)
-      iFunction              13 (error)
+      iFunction              13 CDC Serial
     Interface Descriptor:
       bLength                 9
       bDescriptorType         4
@@ -148,7 +148,7 @@
       bInterfaceClass         2 Communications
       bInterfaceSubClass      2 Abstract (modem)
       bInterfaceProtocol      1 AT-commands (v.25ter)
-      iInterface             11 (error)
+      iInterface             11 CDC Abstract Control Model (ACM)
       CDC Header:
         bcdCDC               1.10
       CDC Call Management:
@@ -179,7 +179,7 @@
       bInterfaceClass        10 CDC Data
       bInterfaceSubClass      0 
       bInterfaceProtocol      0 
-      iInterface             12 (error)
+      iInterface             12 CDC ACM Data
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -209,7 +209,7 @@
       bInterfaceClass         2 Communications
       bInterfaceSubClass      0 
       bInterfaceProtocol      0 
-      iInterface             15 (error)
+      iInterface             15 IIO
       Endpoint Descriptor:
         bLength                 7
         bDescriptorType         5
@@ -270,5 +270,14 @@
           Usage Type               Data
         wMaxPacketSize     0x0200  1x 512 bytes
         bInterval               0
+Device Qualifier (for other device speed):
+  bLength                10
+  bDescriptorType         6
+  bcdUSB               2.00
+  bDeviceClass            0 
+  bDeviceSubClass         0 
+  bDeviceProtocol         0 
+  bMaxPacketSize0        64
+  bNumConfigurations      1
 Device Status:     0x0000
   (Bus Powered)
```

which, like you - causes issues in the library I'm working on, which verifies the usb device by the `iInterface` type.

rgetz, 2020-05-25:
Just trying to dig into things a little bit...

It looks like the libusb function: `libusb_get_string_descriptor` isn't being recorded properly. and then returns error on playback.

https://github.com/libusb/libusb/blob/master/libusb/libusb.h#L1815-L1835

I don't know if this is a limitation, or just a bug...

-Robin

martinpitt, 2021-06-25:
This got fixed a while ago in commit 1b58d24fb78e8297f2b0e96abb99fcbee7f37784 , and thus is a duplicate of #115.

closed 2021-06-25 by martinpitt