martinpitt/umockdev #102: i2c ioctl

prakash56755, 2020-07-20:
Hi MArtin, 

Im trying to emulate i2c ioctl calls but when I run "umockdev-record -s /dev/i2c-0=re.script -- /sbin/manger -f "  finally the script contains only  "d 0 /dev/i2c-0"  But there is nothing related to ioctl calls. do you know what is the reason?

hadess, 2020-11-04:
Does `manger` install its own signal handler for SIGINT? If so, could be the same bug as https://github.com/martinpitt/umockdev/issues/96

benzea, 2021-06-25:
The I2C ioctl's are not supported currently.

You can have a look at #125 which implements support for SPI. Something very similar should be possible to do for I2C. You will also be able to work around the lack of support in umockdev by using the public API that will likely be part of 0.16, see #132.