martinpitt/umockdev #105: SocketServer ctrl_r sometimes can't read control byte

ddstreet, 2020-07-31:
For some reason the SocketServer ctrl_r pipe can't read a byte, after select indicated data ready on ctrl_r; for example:

https://objectstorage.prodstack4-5.canonical.com/v1/AUTH_77e2ada1e7a84929a74ba3b87153c0ac/autopkgtest-groovy-ddstreet-lp1831467/groovy/armhf/u/umockdev/20200731_174931_3daf3@/log.gz

> Bail out! ERROR:tests/test-umockdev-run.c:330:check_program_out: assertion failed (_tmp7_ == expected_out): ("Bail out! ERROR:src/umockdev.c:7574:umockdev_socket_server_run: assertion failed: (Posix.read (this.ctrl_r, &buf, 1) == 1)\n" == "Got name: Joe Tester\n\nGot recv: somejunk\n")

This fails in src/umockdev.vala in:

```
            // if we got triggered by our control fd, consume the data                                                                                                                                                                                                        
            if (Posix.FD_ISSET (this.ctrl_r, fds) > 0) {                                                                                                                                                                                                                      
                debug ("socket server thread: woken up by control fd");                                                                                                                                                                                                       
                char buf;                                                                                                                                                                                                                                                     
                assert (Posix.read (this.ctrl_r, &buf, 1) == 1);                                                                                                                                                                                                              
                continue;                                                                                                                                                                                                                                                     
            }                                                                                                                                                                                                                                                                 

```
I don't see why select() would indicate there's data to read, but then a read can't read the data. I'm thinking of just removing the assert() to avoid the testcase failure, unless you can see why it's failing?

martinpitt, 2020-08-07:
According to select(2), an fd in `readfds` either has data or is EOF (when it returns 0). I don't immediately see how this could happen, though -- `this.ctrl_w` is never closed (in fact, that's a bug), so it should never EOF.

I'm able to reproduce this when I hack the `run()` thread a bit.

martinpitt, 2020-08-07:
I sent PR #106 which should fix that. Thanks for the report!

PR #106 (2020-08-07): Fix SocketServer control EOF handling
Handle EOF on the control pipe. This sometimes seems to happen in slow
test environments even though neither end of the control pipe got
closed.

Now close them properly, and use the EOF event as notification instead
of this.running and sending a control byte.

Fixes #105
commit b5cf0a58 (2020-08-07): Fix SocketServer control EOF handling (#106)

Handle EOF on the control pipe. This sometimes seems to happen in slow
test environments even though neither end of the control pipe got
closed.

Now close them properly, and use the EOF event as notification instead
of this.running and sending a control byte.

Fixes #105
closed 2020-08-07 by martinpitt