martinpitt/umockdev #111: Header installed in wrong place?

hadess, 2020-11-17:
Building 0.15:
```sh
[...]
RPM build errors:
    File not found: /home/hadess/Projects/packages/BUILDROOT/umockdev-0.15-1.fc34.x86_64/usr/include/umockdev-1.0
Could not execute local: rpmbuild --define '_sourcedir /home/hadess/Projects/Fedora/merged/umockdev' --define '_specdir /home/hadess/Projects/Fedora/merged/umockdev' --define '_builddir /home/hadess/Projects/Fedora/merged/umockdev' --define '_srcrpmdir /home/hadess/Projects/Fedora/merged/umockdev' --define '_rpmdir /home/hadess/Projects/Fedora/merged/umockdev' --define 'dist %{?distprefix}.fc34' --define 'fedora 34' --eval '%undefine rhel' --define 'fc34 1' --eval '%undefine fc33' -ba /home/hadess/Projects/Fedora/merged/umockdev/umockdev.spec 2>&1 | tee .build-0.15-1.fc34.log
$ ls /home/hadess/Projects/packages/BUILDROOT/umockdev-0.15-1.fc34.x86_64/usr/include/umockdev.h 
/home/hadess/Projects/packages/BUILDROOT/umockdev-0.15-1.fc34.x86_64/usr/include/umockdev.h
```

`umockdev.h` should be installed in `/usr/include/umockdev-1.0`, or the pkg-config file updated:
```
Cflags: -I${includedir}/umockdev-1.0
```

martinpitt, 2020-11-17:
Argh, je suis désolé ! I just pushed a fix in commit 91f77b8, I'll do a 0.15.1 release today.

closed 2020-11-17 by martinpitt

hadess, 2020-11-17:
Kein problem, danke sehr!

martinpitt, 2020-11-17:
Voilà: https://github.com/martinpitt/umockdev/releases/tag/0.15.1

commit 91f77b8a (2020-11-17): Fix installation directory of umockdev.h

In previous versions this used to be in /usr/include/umockdev-1.0/, and
that's also still what our pkg-config says.

Fixes #111