martinpitt/umockdev #117: fingerprint readers don't work with umockdev-record ioctl trees

martinpitt, 2021-01-24:
@benzea and @3v1n0, the [libfprint](https://gitlab.freedesktop.org/libfprint/libfprint) authors, currently have to employ some [hackish post-processing](https://gitlab.freedesktop.org/libfprint/libfprint/-/blob/cb332c/tests/convert-ioctl.py) ([docs](https://gitlab.freedesktop.org/libfprint/libfprint/-/blob/cb332c/tests/README.md#possible-issues)) of recorded ioctls to make them work. There is also some related [#fprint IRC conversation](https://www.pastery.net/zsaxcz/) about this.

Filing this issue to collect debugging information and some discussion about it.

martinpitt, 2021-01-24:
Without having delved into this yet (it's been many years since I wrote this code, and it was mostly experimental), there are two lines of investigation:

 * Check if the "next sibling" iteration can be salvaged to support this kind of "interleaved stateful" ioctls, where the request is identical, but the response is different. My first hunch is that the interleaving of A and B breaks the state keeping.
 * If that will break compatibility or is too involved, consider adding an option to umockdev-record like `--stateful` or `--linear` which will not do the grouping.

3v1n0, 2021-01-24:
FYI the IRC conversation is liked also here https://www.pastery.net/matzzg/ as the previous link is going to expire...

benzea, 2021-01-26:
> My first hunch is that the interleaving of A and B breaks the state keeping.

Yes, that would likely work. In a way it does feel wrong to me though as these devices have a more complex internal state. I would prefer just having things in-order and being able to throw an error immediately when something unexpected happens (rather than trying to recover and potentially causing infinite loops).

martinpitt, 2021-06-25:
Done in #124, thanks @benzea !

closed 2021-06-25 by martinpitt