martinpitt/umockdev #120: time64 issue on 32-bit arches

z3ntu, 2021-03-01:
umockdev fails to compile on 32-bit Alpine Linux (x86 & armv7) with the following error:
```
[1/9] Compiling C object libumockdev.so.0.3.0.p/meson-generated_src_umockdev.c.o
[2/9] Compiling C object libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o
ninja: job failed: cc -Ilibumockdev-preload.so.0.0.0.p -I. -I.. -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -Werror -Werror=missing-prototypes -Werror=strict-prototypes -Werror=nested-externs -Werror=pointer-arith -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format-security -Werror=format=2 -Werror=unused-variable -Werror=return-type -Werror=uninitialized -fPIC -fvisibility=default -MD -MQ libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -MF libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o.d -o libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -c ../src/libumockdev-preload.c
../src/libumockdev-preload.c: In function 'script_record_op':
../src/libumockdev-preload.c:1045:12: error: 'const struct input_event' has no member named 'time'
 1045 |    (long) e->time.tv_sec, (long) e->time.tv_usec, e->type, e->code, e->value);
      |            ^~
../src/libumockdev-preload.c:1045:35: error: 'const struct input_event' has no member named 'time'
 1045 |    (long) e->time.tv_sec, (long) e->time.tv_usec, e->type, e->code, e->value);
      |                                   ^~
ninja: job failed: cc -Ilibumockdev.so.0.3.0.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -w -g -Werror -Werror=missing-prototypes -Werror=strict-prototypes -Werror=nested-externs -Werror=pointer-arith -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format-security -Werror=format=2 -Werror=unused-variable -Werror=return-type -Werror=uninitialized -fPIC -pthread -MD -MQ libumockdev.so.0.3.0.p/meson-generated_src_umockdev.c.o -MF libumockdev.so.0.3.0.p/meson-generated_src_umockdev.c.o.d -o libumockdev.so.0.3.0.p/meson-generated_src_umockdev.c.o -c libumockdev.so.0.3.0.p/src/umockdev.c
../src/umockdev.vala: In function 'umockdev_testbed_load_evemu_events':
../src/umockdev.vala:919:21: error: 'struct input_event' has no member named 'time'
  919 |                 delay = (int) (ev_sec - ev.time.tv_sec) * 1000 + (int) (ev_usec - ev.time.tv_usec) / 1000;
      |                     ^
../src/umockdev.vala:919:21: error: 'struct input_event' has no member named 'time'
  919 |                 delay = (int) (ev_sec - ev.time.tv_sec) * 1000 + (int) (ev_usec - ev.time.tv_usec) / 1000;
      |                     ^
../src/umockdev.vala:923:5: error: 'struct input_event' has no member named 'time'
  923 |             ev.time.tv_sec = ev_sec;
      |     ^
../src/umockdev.vala:924:5: error: 'struct input_event' has no member named 'time'
  924 |             ev.time.tv_usec = ev_usec;
      |     ^
ninja: subcommands failed
```

The first one can be fixed relatively simply by replacing `time.tv_sec` with `input_event_sec` and `time.tv_usec` with `input_event_usec` (see e.g. https://github.com/gvalkov/python-evdev/commit/bcafe723e489e00441bbd0c86d252df63c422e2c), but it seems vala has kind of duplicated the time structure into the vala source? https://gitlab.gnome.org/GNOME/vala/-/blob/0.51.3/vapi/linux.vapi#L1647

See also https://wiki.adelielinux.org/wiki/Project:Time64 & https://www.linaro.org/blog/the-end-of-an-era/

martinpitt, 2021-03-14:
Thanks for the report! Indeed this is a bit tricky to do from Vala now, I reported this as https://gitlab.gnome.org/GNOME/vala/-/issues/1152 . I'll see if I can do that with a local vapi.

martinpitt, 2021-03-15:
I backported the upstream vapi fix, and enabled CI on i386/alpine -- [success](https://github.com/martinpitt/umockdev/runs/2110152029?check_suite_focus=true)!

closed 2021-03-15 by martinpitt

z3ntu, 2021-03-15:
Awesome, works great! Thanks a lot!

commit 8376f0c8 (2021-03-15): vala: Stop using deprecated input_event.time

Backport the vapi fix [1] for the input_event API change [2], to fix
building on 32 bit Alpine (and presumably other distributions).

[1] https://gitlab.gnome.org/GNOME/vala/-/commit/dbe4b716a8d79704
[2] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit?id=152194fe9c3f

Fixes #120