martinpitt/umockdev #122: 0.15.4: build fails with latest vala

kloczek, 2021-05-03:
vala 0.52.3
```console
+ /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/libexec --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload --auto-features=enabled . x86_64-redhat-linux-gnu -D gtk_doc=true
The Meson build system
Version: 0.58.0
Source dir: /home/tkloczko/rpmbuild/BUILD/umockdev-0.15.4
Build dir: /home/tkloczko/rpmbuild/BUILD/umockdev-0.15.4/x86_64-redhat-linux-gnu
Build type: native build
Project name: umockdev
Project version: 0.15.4
C compiler for the host machine: /usr/bin/gcc (gcc 11.1.1 "gcc (GCC) 11.1.1 20210428 (Red Hat 11.1.1-1)")
C linker for the host machine: /usr/bin/gcc ld.bfd 2.36.1-8
Vala compiler for the host machine: valac (valac 0.52.3)
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program g-ir-compiler found: YES (/usr/bin/g-ir-compiler)
Library dl found: YES
Found pkg-config: /usr/bin/pkg-config (1.7.4)
Run-time dependency glib-2.0 found: YES 2.68.1
Run-time dependency gobject-2.0 found: YES 2.68.1
Run-time dependency gio-2.0 found: YES 2.68.1
Run-time dependency gio-unix-2.0 found: YES 2.68.1
Run-time dependency libudev found: YES 247
Run-time dependency gudev-1.0 found: YES 236
Program python3 found: YES (/usr/bin/python3)
Library posix found: YES
Library linux found: YES
Library udev found: YES
Library util found: YES
Library config found: YES
Library posix_extra found: YES
Library ioctl found: YES
Library assertions found: YES
Checking if "ioctl request parameter type is int" compiles: NO
Program sed found: YES (/usr/bin/sed)
Program tests/test-static-code found: YES (/home/tkloczko/rpmbuild/BUILD/umockdev-0.15.4/tests/test-static-code)
Configuring config.h using configuration
Configuring version.xml using configuration
Program gtkdoc-scan found: YES (/usr/bin/gtkdoc-scan)
Program gtkdoc-scangobj found: YES (/usr/bin/gtkdoc-scangobj)
Program gtkdoc-mkdb found: YES (/usr/bin/gtkdoc-mkdb)
Program gtkdoc-mkhtml found: YES (/usr/bin/gtkdoc-mkhtml)
Program gtkdoc-fixxref found: YES (/usr/bin/gtkdoc-fixxref)
Build targets in project: 16

Found ninja-1.10.2 at /usr/bin/ninja
+ /usr/bin/meson compile -C x86_64-redhat-linux-gnu -j 48 --verbose
ninja: Entering directory `x86_64-redhat-linux-gnu'
[1/38] /usr/bin/gcc -Ilibumockdev-preload.so.0.0.0.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -fPIC -fvisibility=default -MD -MQ libumockdev-preload.so.0.0.0.p/src_debug.c.o -MF libumockdev-preload.so.0.0.0.p/src_debug.c.o.d -o libumockdev-preload.so.0.0.0.p/src_debug.c.o -c ../src/debug.c
[2/38] /usr/bin/gcc -Ireadbyte.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -MD -MQ readbyte.p/tests_readbyte.c.o -MF readbyte.p/tests_readbyte.c.o.d -o readbyte.p/tests_readbyte.c.o -c ../tests/readbyte.c
[3/38] /usr/bin/gcc -Itest-ioctl-tree.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -MD -MQ test-ioctl-tree.p/src_debug.c.o -MF test-ioctl-tree.p/src_debug.c.o.d -o test-ioctl-tree.p/src_debug.c.o -c ../src/debug.c
[4/38] /usr/bin/gcc -Ichatter.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -MD -MQ chatter.p/tests_chatter.c.o -MF chatter.p/tests_chatter.c.o.d -o chatter.p/tests_chatter.c.o -c ../tests/chatter.c
[5/38] /usr/bin/gcc -Ichatter-socket-stream.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -MD -MQ chatter-socket-stream.p/tests_chatter-socket-stream.c.o -MF chatter-socket-stream.p/tests_chatter-socket-stream.c.o.d -o chatter-socket-stream.p/tests_chatter-socket-stream.c.o -c ../tests/chatter-socket-stream.c
[6/38] /usr/bin/gcc -Itest-ioctl-tree.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -MD -MQ test-ioctl-tree.p/src_ioctl_tree.c.o -MF test-ioctl-tree.p/src_ioctl_tree.c.o.d -o test-ioctl-tree.p/src_ioctl_tree.c.o -c ../src/ioctl_tree.c
[7/38] /usr/bin/gcc -Ilibumockdev-preload.so.0.0.0.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -fPIC -fvisibility=default -MD -MQ libumockdev-preload.so.0.0.0.p/src_ioctl_tree.c.o -MF libumockdev-preload.so.0.0.0.p/src_ioctl_tree.c.o.d -o libumockdev-preload.so.0.0.0.p/src_ioctl_tree.c.o -c ../src/ioctl_tree.c
[8/38] /usr/bin/gcc -Ilibumockdev-preload.so.0.0.0.p -I. -I.. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -fcommon -fPIC -fvisibility=default -MD -MQ libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -MF libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o.d -o libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -c ../src/libumockdev-preload.c
[9/38] /usr/bin/gcc -Itest-ioctl-tree.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-h [... cut ...]
          ^^^^^^^^^^^^^
../tests/test-umockdev-vala.vala:714.3-714.16: error: `assert_cmpuint' is an ambiguous reference between `GLib.assert_cmpuint' and `Assertions.assert_cmpuint'
  assert_cmpuint (add_count, Op.EQ, 1);
  ^^^^^^^^^^^^^^
../tests/test-umockdev-vala.vala:715.3-715.16: error: `assert_cmpuint' is an ambiguous reference between `GLib.assert_cmpuint' and `Assertions.assert_cmpuint'
  assert_cmpuint (change_count, Op.EQ, num_changes);
  ^^^^^^^^^^^^^^
Compilation failed: 123 error(s), 0 warning(s)
```

kloczek, 2021-05-03:
OK. I see that issue is fixed in git.
Any plans to make new release? :)

martinpitt, 2021-05-03:
@kloczek : Thanks for the reminder! None of the distros that I have in my CI (Ubuntu, Debian, CentOS, Fedora, Alpine, NixOS) already has a vala > 0.50, so I didn't notice. Here we go! https://github.com/martinpitt/umockdev/releases/tag/0.15.5

closed 2021-05-03 by martinpitt

kloczek, 2021-05-03:
I've just sucessfully build my rpm package with updated umockdev😀

Thank you very much for your time.