martinpitt/umockdev #133: Update build, test, run instructions for new build system

patrickyeon, 2021-06-24:
As you'd expect, having moved away from autotools means `./configure && make` etc doesn't work. I'm not familiar with either of meson, ninja, nor valac, so please forgive my bumbling through. What has worked for me (Ubuntu 20.04 focal) is
```
sudo apt install meson ninja-build valac libudev-dev libpcap-dev libgudev-1.0-dev gobject-introspection libgirepository1.0-dev
meson build
cd build && ninja
```
There might be other requirements that aren't commonly on peoples' machines but I've already luckily installed. And I wouldn't be surprised that there's an accepted way to use `meson` and `ninja` that isn't that, but it gave me executables.

Hopefully that's enough breadcrumbs for someone else running into this until you've got a chance to update the README.

martinpitt, 2021-06-25:
Oops, thanks for pointing out! I updated the README accordingly.

martinpitt, 2021-06-25:
See https://github.com/martinpitt/umockdev#build-test-run

commit d743c794 (2021-06-25): README: Update for meson build system

Fixes #133
commit 2ebd5159 (2021-06-25): README: Update for meson build system

Fixes #133
commit 7903427e (2021-06-25): README: Update for meson build system

Fixes #133
closed 2021-06-25 by martinpitt