martinpitt/umockdev #138: unused `UMockdev.IoctlBase.unregister_{path,all}

martinpitt, 2021-08-25:
Building gives these two warnings:
```
../src/umockdev-ioctl.vala:771.5-771.33: warning: method `UMockdev.IoctlBase.unregister_path' never used
    internal void unregister_path(string devnode)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

../src/umockdev-ioctl.vala:777.5-777.32: warning: method `UMockdev.IoctlBase.unregister_all' never used
    internal void unregister_all()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

@benzea , are these supposed to be used in some destructors, or are they just leftovers and can be dropped?

(I'm currently cleaning up the plethora of warnings, and these two I'm not sure about)

benzea, 2021-08-25:
Neither :-/

It is a side effect of umockdev-ioctl.vala being included in the build of multiple binaries.

benzea, 2021-08-25:
IIRC, the reason was that I would have needed to make some more API public otherwise. Though, at this point, it is likely possible to split the file into multiple parts in order to avoid doing so.

martinpitt, 2021-09-12:
I proposed a fix in #148