martinpitt/umockdev #147: Build failure on sparc with -Wcast-align

floppym, 2021-09-12:
This was reported by a user on Gentoo Linux (sparc).

https://bugs.gentoo.org/812461

The user has `-Wcast-align` in CFLAGS, and this gets converted to an error by `-Werror` parameter configured in the project arguments.

It might be worth addressing the alignment issue.
```
FAILED: libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o 
sparc-unknown-linux-gnu-gcc -Ilibumockdev-preload.so.0.0.0.p -I. -I../umockdev-0.16.1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -Werror=missing-prototypes -Werror=strict-prototypes -Werror=nested-externs -Werror=pointer-arith -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format-security -Werror=format=2 -Werror=unused-variable -Werror=return-type -Werror=uninitialized -O2 -mcpu=niagara2 -pipe -Wcast-align -fPIC -fvisibility=default -MD -MQ libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -MF libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o.d -o libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o -c ../umockdev-0.16.1/src/libumockdev-preload.c
../umockdev-0.16.1/src/libumockdev-preload.c: In function ?netlink_recvmsg?:
../umockdev-0.16.1/src/libumockdev-preload.c:418:27: error: cast increases required alignment of target type [-Werror=cast-align]
  418 |      struct ucred *cred = (struct ucred *)CMSG_DATA(cmsg);
      |                           ^
cc1: all warnings being treated as errors
```

martinpitt, 2021-09-12:
On Debian sparc64, [tests fail with SIGBUS](https://buildd.debian.org/status/fetch.php?pkg=umockdev&arch=sparc64&ver=0.16.2-1&stamp=1631117362&raw=0). This may or may not be related.

martinpitt, 2021-09-12:
FTR, the Debian sparc failure is unrelated. Getting this code wrong fails tests/test-umockdev.c `t_testbed_add_devicev()`, not test-umockdev-vala.

This is annoying to "fix": the code can't just use `memcpy()` into an actually declared struct, as "real" code would do, as it really needs to modify the existing `msg` from the wrapped `netlink_recvmsg()`. I'll try to come up with something using memcpy()'ing an uid_t into the message with `offsetof()`.

commit 3e3fb703 (2021-09-12): preload: Avoid unaligned access into struct ucred

Casting the netlink CMSG_DATA into a `struct ucred` increases the
required alignment on at least sparc. Move to memcpy() instead.

Fixes #147
commit 64f5a484 (2021-09-12): preload: Avoid unaligned access into struct ucred

Casting the netlink CMSG_DATA into a `struct ucred` increases the
required alignment on at least sparc. Move to memcpy() instead.

Fixes #147
closed 2021-09-12 by martinpitt