martinpitt/umockdev #153: Is it possible to hide the KVM/QEMU environment from a program which detects which platform it's running on?

ethaniel, 2021-11-29:
I have a program which gathers system information (CPU, Motherboard, Hard drive models, etc) via udevadm to generate license information. Unfortunately, it's built so it won't allow itself to be run in a KVM/QEMU environment (which my whole system is based on). 

Is there a way to make it think that it's running on a bare metal system?

benzea, 2021-11-29:
I think that use-case is out of scope for umockdev. I am sure it is technically possible to manipulate the environment to the point that you can fool such a tool. But, the purpose of umockdev is to just provide enough in order to enable CI testing for software.

It is impossible to answer the question without knowing what that tool does.

ethaniel, 2021-11-29:
The software is querying:
1. udevadm info --query=all --name=/dev/sda | grep ID_SERIAL=
2. /proc/cpuinfo

I was thinking if there is a way to override the request to these 2 items and replace the output that the software gets?

ethaniel, 2021-11-29:
Solved it myself. Replaced /bin/udevadm with a bash script which outputs data that I need.
And also did `sudo mount --bind /cpuinfo.txt /proc/cpuinfo` to fake the /proc/cpuinfo data

martinpitt, 2021-11-30:
@ethaniel : Thanks for the clarification -- these parts are indeed what umockdev does. See the documented examples [how to fake files in /proc](https://github.com/martinpitt/umockdev/#command-line-mock-file-in-proc), you can dump your desired /dev/sda  with `umockdev-record` and load it in the test environment.

However, of course you need to run your program under `umockdev-run` for that. If you want *everything* in your system to believe the above fake data, without running under umockdev-run, then you need root privs as you did.

closed 2021-11-30 by martinpitt