martinpitt/umockdev #154: install problems

ethaniel, 2021-11-30:
On debian, after running `sudo meson install`, I'm getting:
```
ninja: Entering directory `.'
[23/55] Compiling Vala source ../src/uevent_sender.vapi ../src/ioctl_tree.vapi .....octl.vala ../src/umockdev-pcap.vala ../src/umockdev-spi.vala umockdev-utils.vapi.
FAILED: umockdev@sha/src/umockdev.c umockdev@sha/src/umockdev-ioctl.c umockdev@sha/src/umockdev-pcap.c umockdev@sha/src/umockdev-spi.c umockdev.h umockdev-1.0.vapi UMockdev-1.0.gir
valac -C --debug --debug --pkg libpcap /tmp/umockdev/src/ioctl.vapi /tmp/umockdev/src/linux_fixes.vapi --pkg linux --pkg posix --pkg gio-unix-2.0 --pkg gio-2.0 --pkg gobject-2.0 --target-glib ' 2.32.0' --pkg glib-2.0 --color=always --directory umockdev@sha --basedir ../ --library umockdev --header umockdev.h --vapi ../umockdev-1.0.vapi --gir ../UMockdev-1.0.gir --define=INTERNAL_REGISTER_API --define=INTERNAL_UNREGISTER_PATH_API --vapidir=/tmp/umockdev/src ../src/uevent_sender.vapi ../src/ioctl_tree.vapi ../src/umockdev.vala ../src/umockdev-ioctl.vala ../src/umockdev-pcap.vala ../src/umockdev-spi.vala umockdev-utils.vapi
../src/umockdev.vala:61.5-67.11: error: Return: Cannot convert from `GLib.Thread<weak void>' to `GLib.Thread<void>'
../src/umockdev.vala:102.9-102.67: error: Assignment: Cannot convert from `GLib.Thread<weak void>' to `GLib.Thread<weak void>'
        this.worker_thread = create_worker_thread(this.worker_loop);
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
../src/umockdev.vala:1082.32-1082.63: error: 1 extra arguments for `ulong ulong.parse (string)'
            ev.type = (uint16) ulong.parse (match.fetch(3), 16);
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
../src/umockdev.vala:1083.32-1083.63: error: 1 extra arguments for `ulong ulong.parse (string)'
            ev.code = (uint16) ulong.parse (match.fetch(4), 16);
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Compilation failed: 4 error(s), 0 warning(s)
[24/55] Compiling Vala source ../src/ioctl_tree.vapi ../src/umockdev-record.vala ...octl.vala ../src/umockdev-pcap.vala ../src/umockdev-spi.vala umockdev-utils.vapi.
ninja: build stopped: subcommand failed.
Could not rebuild
```

ethaniel, 2021-12-02:
Solved for now by doing `apt-get install umockdev` and not compiling from source.

martinpitt, 2021-12-07:
CI regularly tests on Debian testing (currently vala 0.54), and I just tested this on Debian 11 stable (vala 0.48). This also does not make much sense -- [ulong.parse()](https://valadoc.org/glib-2.0/ulong.parse.html) is documented to have two arguments; the second one is optional for sure, but it absolutely needs to accept a second one. Did you perhaps install some custom valac version or run on an older Debian or so?

martinpitt, 2022-01-21:
Any further information about this? right now this is not actionable, I'm afraid.

closed 2022-07-02 by martinpitt