martinpitt/umockdev #155: missing /sys/dev/block entries for block device

ethaniel, 2021-12-02:
Hello!

I'm doing: `umockdev-record /dev/nvme0n1  > umockdev.txt` on one host, and then transfer `umockdev.txt` to another host.
Then I run `umockdev-run --device umockdev.txt -- udevadm info --query=all --name=/dev/nvme0n1` and get the `device node not found` error.

Strace shows that udevadm is trying to find the `/tmp/umockdev.S1ZSD1/sys/dev/block/259:0` path which is missing:
```
access("/tmp/umockdev.S1ZSD1/dev/nvme0n1", F_OK) = 0
stat("/tmp/umockdev.S1ZSD1/dev/nvme0n1", {st_mode=S_IFREG|S_ISVTX|0644, st_size=0, ...}) = 0
readlink("/tmp/umockdev.S1ZSD1/dev/.node/nvme0n1", "259:0", 4096) = 5
access("/tmp/umockdev.S1ZSD1/disabled", F_OK) = -1 ENOENT (No such file or directory)
readlink("/tmp/umockdev.S1ZSD1/sys/dev/block/259:0", 0x7ffc5eb2c330, 1024) = -1 ENOENT (No such file or directory)
access("/tmp/umockdev.S1ZSD1/disabled", F_OK) = -1 ENOENT (No such file or directory)
stat("/tmp/umockdev.S1ZSD1/sys/dev/block/259:0", 0x7ffc5eb2c780) = -1 ENOENT (No such file or directory)
write(2, "device node not found\n", 22device node not found
```

Am I missing something?

ethaniel, 2021-12-02:
The dirty hack that solved this for me was simply recreating the symlink as it was on the old machine. Since umockdev does all the heavy lifting, it was enough:

```
umockdev-run --device /mrz/umockdev.txt -- sh -c 'mkdir -p $UMOCKDEV_DIR/sys/dev/block ; ln -s ../../devices/pci0000:16/0000:16:01.0/0000:18:00.0/nvme/nvme0/nvme0n1 $UMOCKDEV_DIR/sys/dev/block/259:0 ; udevadm info --query=all --name=/dev/nvme0n1'
```

martinpitt, 2021-12-12:
What's your umockdev version? It looks like one from before 0.15.2, as none of your recorded attributes are `\n` terminated.

I can reproduce this on the current version, but for a different reason. There is some fallout from commit 1b58d24fb78e8297f2b0e96abb99fcbee7f37784. The recorded attribute now looks like this:

    A: dev=7:23\n

and it creates a broken symlink:

    '7:23'$'\n' -> ../../devices/virtual/block/loop23

martinpitt, 2021-12-12:
Fixed in PR #158. It's still a little unclear to me how exactly it failed for you (as you seem to have an old umockdev version), but I tested it end to end with an NVME block device now.

commit 8257fe39 (2021-12-12): Fix /sys/dev/* symlinks

Commit 1b58d24fb78e regressed these, as it included the trailing \n into
the dev symlinks, like

    '7:23'$'\n' -> ../../devices/virtual/block/loop23

Strip off whitespace from the recorded `dev` attribute to fix this.

Fixes #155
commit d0b9e0e1 (2021-12-12): Fix /sys/dev/* symlinks

Commit 1b58d24fb78e regressed these, as it included the trailing \n into
the dev symlinks, like

    '7:23'$'\n' -> ../../devices/virtual/block/loop23

Strip off whitespace from the recorded `dev` attribute to fix this.

Fixes #155
PR #158 (2021-12-12): Fix /sys/dev/* symlinks
Commit 1b58d24fb78e regressed these, as it included the trailing \n into
the dev symlinks, like

    '7:23'$'\n' -> ../../devices/virtual/block/loop23

Strip off whitespace from the recorded `dev` attribute to fix this.

Fixes #155
commit 9e966e89 (2021-12-12): Fix /sys/dev/* symlinks

Commit 1b58d24fb78e regressed these, as it included the trailing \n into
the dev symlinks, like

    '7:23'$'\n' -> ../../devices/virtual/block/loop23

Strip off whitespace from the recorded `dev` attribute to fix this.

Fixes #155
closed 2021-12-12 by martinpitt