martinpitt/umockdev #159: Build started to fail with NixOS

martinpitt, 2021-12-21:
At some time in the [last four days](https://github.com/martinpitt/umockdev/actions), umockdef failed to build and test on NixOS. [Most recent run](https://github.com/martinpitt/umockdev/actions/runs/1600420074):
```
>>> MALLOC_CHECK_=3 TOP_SRCDIR=/tmp/nix-build-umockdev-0.17.0.drv-0/source G_DEBUG=fatal-warnings,fatal-criticals,gc-friendly MALLOC_PERTURB_=5 LD_LIBRARY_PATH=/tmp/nix-build-umockdev-0.17.0.drv-0/source/build PATH=/tmp/nix-build-umockdev-0.17.0.drv-0/source/build:/nix/store/mqbify3fap3rv1iz4s8z3jxpb5p57x7s-gobject-introspection-1.70.0-dev/bin:/nix/store/xx643xl761fryig8f2pljhhnhdn27k4p-glib-2.70.1-dev/bin:/nix/store/sk8ihyhk4gmx031nnw98h818k47wa9hl-gettext-0.21/bin:/nix/store/jaz6dmzddb1vyrncfyisphna4ly92rag-glib-2.70.1-bin/bin:/nix/store/2gkr4mrndxidax8i05i3wv3ps4nirv6j-gtk-doc-1.33.2/bin:/nix/store/b1p9wa97fakdwmg7yyp1mancwg1icc2f-meson-0.57.1/bin:/nix/store/dxw81a1l5sxibqbc489cd4k9m3j387c5-ninja-1.10.2/bin:/nix/store/d3c4y6ssahavbjgisvvldc9kdnfjryyk-pkg-config-wrapper-0.29.2/bin:/nix/store/qsmafsw510l84zfjd4xgy70278jyrz4i-vala-0.54.2/bin:/nix/store/k0z9n599k02hab8qjjp3ljw065iwjcvg-python3-3.9.6/bin:/nix/store/wj4g3zl2n4y99wa31qkqhm6w2i9s1k13-which-2.21/bin:/nix/store/f3zw0i5mnqd816135nvzqjbi37g3vp7j-usbutils-014/bin:/nix/store/pjba17qx8vh7dln8m5vnkqdbbjahvg9b-patchelf-0.13/bin:/nix/store/xiq6j4jsyj351p8q3yw9cg1hdqp9m685-gcc-wrapper-10.3.0/bin:/nix/store/iyssx2arl8bc40pjimyfxyknj06swjdx-gcc-10.3.0/bin:/nix/store/q6prgn66s4achzrrnvcvyjgnm94c1bn3-glibc-2.33-56-bin/bin:/nix/store/l6f4z8mmcnnxba8w004xn28y0vr4gdkf-coreutils-9.0/bin:/nix/store/v7w0pspwq8r2b7k2sndxq3db843z7xm5-binutils-wrapper-2.35.2/bin:/nix/store/lbxfixyw1yk099pjyaiy3xj5dl7kxm1g-binutils-2.35.2/bin:/nix/store/aqwqh3drn9bf6qrs70wiyrxaawh1wcnz-systemd-249.5/bin:/nix/store/bf00gj7xcmbvhdhl1c5lgpwabr3p93c2-libpcap-1.10.1/bin:/nix/store/l6f4z8mmcnnxba8w004xn28y0vr4gdkf-coreutils-9.0/bin:/nix/store/v3lvq9hqshyldc4i6f5jy0zs0k5psbws-findutils-4.8.0/bin:/nix/store/gxj3wkwc5m5vb8b3rv6h029ky9nan4bf-diffutils-3.8/bin:/nix/store/vklvyr82ajbz7jm7g8dbkh62k20b0dpr-gnused-4.8/bin:/nix/store/nkwls56wcfwi1r0jnkqkvwx2zk7w3qrz-gnugrep-3.7/bin:/nix/store/21nzalbq7ay1vzanri1dl4845s3cl72i-gawk-5.1.1/bin:/nix/store/wny483nz9q52xq8azwpqna3pq98zcsgy-gnutar-1.34/bin:/nix/store/mp7lwrwr0nl69rznln13j3k4zf69s80i-gzip-1.11/bin:/nix/store/m9jxbbbxfzi6052yw3kgz95wvz76mbb1-bzip2-1.0.6.0.2-bin/bin:/nix/store/315glp1si526x60nhhkbqmip4m7b25a7-gnumake-4.3/bin:/nix/store/a54wrar1jym1d8yvlijq0l2gghmy8szz-bash-5.1-p12/bin:/nix/store/h2xzp4a764hn5n9br63ilh9qglz3baxn-patch-2.7.6/bin:/nix/store/3q4xj02ijqgfxffndvg28nlv741sx6qx-xz-5.2.5-bin/bin GI_TYPELIB_PATH=/tmp/nix-build-umockdev-0.17.0.drv-0/source/build:/nix/store/wdgxlwarjah6jhhf7jbc8rmb9hj97fnr-gobject-introspection-1.70.0/lib/girepository-1.0:/nix/store/7nx6s739lxz2bjk5hygbdpd08vhw9jqv-libgudev-237/lib/girepository-1.0 /tmp/nix-build-umockdev-0.17.0.drv-0/source/src/umockdev-wrapper /tmp/nix-build-umockdev-0.17.0.drv-0/source/tests/test-static-code
 ✀  
stderr:
/bin/sh: /nix/store/jsp3h3wpzc842j0rz61m5ly71ak6qgdn-glibc-2.32-54/lib/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/nix-build-umockdev-0.17.0.drv-0/source/build/libumockdev-preload.so.0)

3/7 umockdev / ioctl-tree                            FAIL             0.11s   killed by signal 6 SIGABRT
Bail out! ERROR:../tests/test-ioctl-tree.c:257:t_write: assertion failed: (f != NULL)
stderr:
**
ERROR:../tests/test-ioctl-tree.c:257:t_write: assertion failed: (f != NULL)

ok 13 /umockdev-record/script-log-chatter
Bail out! FATAL-ERROR: test-umockdev-record.vala:707: Error: Error binding to address (GUnixSocketAddress): Permission denied
stderr:
```

These reproduce perfectly well in local podman with `tests/run-nix`.

@flokli @jtojnar  I'll need some help with debugging this.

martinpitt, 2021-12-21:
nixos wants a super-privileged container, running the "attaching" bit as user does not work.
```
$ sudo DEBUG=1 tests/run-nix

FAILED: meson-test 
/nix/store/b1p9wa97fakdwmg7yyp1mancwg1icc2f-meson-0.57.1/bin/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
build failed in checkPhase with exit code 1
To attach install cntr and run the following command as root:

   cntr attach -t command cntr-/nix/store/b7dmz2q3cy53r0zpcnn5zjiys2kw4jvd-umockdev-0.17.0
```

So I attached to the running container, then ran that `cntr attach` command, and `cd /tmp/nix-build-umockdev-0.17.0.drv-0/source`. However, this is still not the "right" environment -- neither `meson` nor `ninja` commands are available. @flokli, @jtojnar, what else do I need to do to get in the real env where meson and the tests run? Thanks!

martinpitt, 2021-12-21:
I. e. I can run meson from that private path as in the build log, but it doesn't find anything else:
```
$ /nix/store/b1p9wa97fakdwmg7yyp1mancwg1icc2f-meson-0.57.1/bin/meson test -C build     
Can't find ninja, can't rebuild test.
```

I. e. there must be some magic to define `$PATH` to all the installed nix packages?

jtojnar, 2021-12-21:
I am surprised by `/bin/sh: /nix/store/jsp3h3wpzc842j0rz61m5ly71ak6qgdn-glibc-2.32-54/lib/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/nix-build-umockdev-0.17.0.drv-0/source/build/libumockdev-preload.so.0)`, will try to look.

I cannot reproduce it locally so it might be something with containers or the glibc weirdness.

You can enter the environment by replacing `nix-build` with `nix-shell` (i.e. `nix-shell --keep-failed /tmp/default.nix`).

jtojnar, 2021-12-21:
`https://github.com/NixOS/nixpkgs/archive/master.tar.gz` contains  `/nix/store/563528481rvhc5kxwipjmg6rqrl95mdx-glibc-2.33-56` and it is fetched so that should not be an issue.

Sounds like the old glibc version comes from `/bin/sh`, which should not be accessible in sandbox.

Actually, looks like there is no sed: `/bin/sh: line 4: sed: command not found` so sandboxing does not get enabled?

martinpitt, 2021-12-21:
@jtojnar : Ugh, yes, well spotted! I feel a bit embarrassed now.

Not having sed nor awk is a bit awkward, but I replaced it with shell string manipulation now, and also added the missing `set -e`.

Thanks, and have some nice EOY holidays!

commit 4f22f02d (2021-12-21): tests: Work around unavailable sed in NixOS container

sed is not available in the nix container any more, so replace it with
shell. Also add the missing `set -e` to spot such errors right away.

Fixes #159
closed 2021-12-21 by martinpitt