martinpitt/umockdev #167: Property buffer overflow starting at 0.17.4

wmww, 2022-01-20:
Commit b781cf65d49dc18d6319fb783c5617a71134bbbd broke Mir's umockdev tests (https://github.com/MirServer/mir/issues/2285). We get this error:
```
ERROR: uevent_sender_send: Property buffer overflow
Aborted (core dumped)
```
In GDB, I tracked down the problem to calling `append_property()` with a `name` that is too long for the `size`. On my machine, the name is (note that snaps I have installed seem to be included, no idea why):
```
(gdb) p name
$2 = 0x555557184f80 "DEVNAME=dri/card0DEVTYPE=drm_minor\nMAJOR=226\nMINOR=0\nSUBSYSTEM=drm\nID_PATH=pci-0000:00:02.0\nID_PATH_TAG=pci-0000_00_02_0\nID_FOR_SEAT=drm-pci-0000_00_02_0\nDEVLINKS=/dev/dri/by-path/pci-0000:00:02.0-card\nTAGS=:uaccess:snap_glimpse-editor_glimpse-editor:master-of-seat:snap_qemu-virgil_arm:snap_snap-store_ubuntu-software-local-file:snap_chromium_chromedriver:snap_super-cool-app_super-cool-app:snap_fluffychat_fluffychat:snap_flokk-contacts_flokk-contacts:snap_krita_krita:snap_qemu-virgil_qemu-virgil:snap_chromium_chromium:seat:snap_qemu-virgil_arm64:snap_firefox_firefox:snap_vlc_vlc:snap_snap-store_snap-store:snap_snap-store_ubuntu-software:snap_google-play-music-desktop-player_google-play-music-desktop-player:snap_discord_discord:\nCURRENT_TAGS=:snap_flokk-contacts_flokk-contacts:snap_chromium_chromium:snap_discord_discord:uaccess:snap_super-cool-app_super-cool-app:snap_fluffychat_fluffychat:snap_vlc_vlc:snap_snap-store_snap-store:snap_qemu-virgil_arm64:snap_qemu-virgil_arm:master-of-seat:snap_google-play-music-desktop-player_google-play-music-desktop-player:snap_glimpse-editor_glimpse-editor:snap_chromium_chromedriver:seat:snap_qemu-virgil_qemu-virgil:snap_firefox_firefox:snap_snap-store_ubuntu-software:snap_krita_krita:snap_snap-store_ubuntu-software-local-file:\n"
```
The length of that is 1281, just above the 1024 cap.

martinpitt, 2022-01-21:
Ugh, thanks for the report! The /umockdev-testbed/uevent/action_overflow check is supposed to catch this, but doesn't. I'll investigate.

martinpitt, 2022-01-21:
Ah wait, it does not actually segfault -- it's doing exactly as desired, aborting with that message *before* it segfaults *phew*. So indeed I only need to bump the buffer size.

closed 2022-01-21 by martinpitt

martinpitt, 2022-01-21:
Released as [0.17.6](https://github.com/martinpitt/umockdev/releases/tag/0.17.6). I uploaded it to Debian unstable, so it should reach Ubuntu jammy over the weekend.

wmww, 2022-01-21:
Thanks for the fast response!

commit a65b2b53 (2022-01-21): uevent_sender: Increase buffer size

Some tests need more than 1 KiB of property data.

Fixes #167
PR #170 (2022-01-21): uevent_sender: Increase buffer size
Some tests need more than 1 KiB of property data.
Fixes #167
commit 542fcb71 (2022-01-21): uevent_sender: Increase buffer size

Some tests need more than 1 KiB of property data.

Fixes #167