martinpitt/umockdev #169: Failed to build on Arch Linux RISC-V

XieJiSS, 2022-01-21:
`pkgver=0.17.5`

I notice that the failing test case is named `umockdev:fails-valgrind / umockdev-vala`. Does this indicate that the test case requires valgrind to success? Currently, valgrind is not usable on RISC-V architecture.

Logs attached:

```
Bail out! ERROR:test-umockdev-vala.p/tests/test-umockdev-vala.c:2617:t_detects_not_running_in_testbed: assertion failed ((gint) read (_tmp12_, &buf, (gsize) 1) == 1): (0 == 1)

stderr:
**
ERROR:test-umockdev-vala.p/tests/test-umockdev-vala.c:2617:t_detects_not_running_in_testbed: assertion failed ((gint) read (_tmp12_, &buf, (gsize) 1) == 1): (0 == 1)
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

4/7 umockdev / static-code                    OK               1.04s
5/7 umockdev:fails-valgrind / umockdev-record OK               2.88s
6/7 umockdev / umockdev                       OK               3.25s
7/7 umockdev / umockdev-run                   OK               3.38s

Summary of Failures:

3/7 umockdev:fails-valgrind / umockdev-vala   FAIL             0.63s   killed by signal 6 SIGABRT

Ok:                 6
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /build/umockdev/src/build/meson-logs/testlog.txt
```

Related full log:

```
3/7 umockdev:fails-valgrind / umockdev-vala   FAIL             0.60s   killed by signal 6 SIGABRT
06:31:48 PATH=/build/umockdev/src/build:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl MALLOC_CHECK_=3 LD_LIBRARY_PATH=/build/umockdev/src/build TOP_SRCDIR=/build/umockdev/src/umockdev-0.17.5 MALLOC_PERTURB_=235 GI_TYPELIB_PATH=/build/umockdev/src/build G_DEBUG=fatal-warnings,fatal-criticals,gc-friendly /build/umockdev/src/umockdev-0.17.5/src/umockdev-wrapper /build/umockdev/src/build/test-umockdev-vala
----------------------------------- output -----------------------------------
stdout:
# random seed: R02Sdfd29916d55826bb129e7974b303e5c0
1..19
# Start of umockdev-testbed-vala tests
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.YNM6F1
# GLib-DEBUG: Failed to get thread scheduler attributes: Function not implemented
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.YNM6F1
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 1 /umockdev-testbed-vala/empty
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.OJ96F1
# DEBUG: umockdev.vala:752: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:756: umockdev_testbed_uevent: sending uevent add for device /sys/devices/extkeyboard1
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.OJ96F1
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 2 /umockdev-testbed-vala/add_devicev
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.31Q6F1
# DEBUG: umockdev.vala:1281: parsing device description for /devices/myusbhub/cam
# DEBUG: umockdev.vala:1350: creating device /devices/myusbhub/cam (subsystem usb)
# DEBUG: umockdev.vala:752: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:756: umockdev_testbed_uevent: sending uevent add for device /sys/devices/myusbhub/cam
# DEBUG: umockdev.vala:1392: create_node_for_device: creating file device /tmp/umockdev.31Q6F1/dev/bus/usb/001/002
# DEBUG: umockdev.vala:1281: parsing device description for /devices/myusbhub
# DEBUG: umockdev.vala:1350: creating device /devices/myusbhub (subsystem usb)
# DEBUG: umockdev.vala:756: umockdev_testbed_uevent: sending uevent add for device /sys/devices/myusbhub
# DEBUG: umockdev.vala:1392: create_node_for_device: creating file device /tmp/umockdev.31Q6F1/dev/bus/usb/001/001
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.31Q6F1
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 3 /umockdev-testbed-vala/gudev-query-list
[... 138 lines cut ...]
# DEBUG: umockdev.vala:756: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb/tty/ttyUSB1
# DEBUG: umockdev.vala:1411: create_node_for_device: creating pty device /tmp/umockdev.64CAG1/dev/ttyUSB1: got pty /dev/pts/10
# DEBUG: umockdev.vala:1432: create_node_for_device: creating ptymap symlink /tmp/umockdev.64CAG1/dev/.ptymap/_dev_pts_10
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.64CAG1
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 13 /umockdev-testbed-vala/tty_stty
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.VLABG1
# DEBUG: umockdev.vala:1281: parsing device description for /devices/serial/ttyS10
# DEBUG: umockdev.vala:1350: creating device /devices/serial/ttyS10 (subsystem tty)
# DEBUG: umockdev.vala:752: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:756: umockdev_testbed_uevent: sending uevent add for device /sys/devices/serial/ttyS10
# DEBUG: umockdev.vala:1411: create_node_for_device: creating pty device /tmp/umockdev.VLABG1/dev/ttyS10: got pty /dev/pts/11
# DEBUG: umockdev.vala:1432: create_node_for_device: creating ptymap symlink /tmp/umockdev.VLABG1/dev/.ptymap/_dev_pts_11
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.VLABG1
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 14 /umockdev-testbed-vala/tty_data
ok 15 /umockdev-testbed-vala/detects_running_in_testbed
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
Bail out! ERROR:test-umockdev-vala.p/tests/test-umockdev-vala.c:2617:t_detects_not_running_in_testbed: assertion failed ((gint) read (_tmp12_, &buf, (gsize) 1) == 1): (0 == 1)
stderr:
**
ERROR:test-umockdev-vala.p/tests/test-umockdev-vala.c:2617:t_detects_not_running_in_testbed: assertion failed ((gint) read (_tmp12_, &buf, (gsize) 1) == 1): (0 == 1)
------------------------------------------------------------------------------
```

XieJiSS, 2022-01-21:
I can confirm that this error does not occur on x86_64 platforms.

martinpitt, 2022-01-21:
> Does this indicate that the test case requires valgrind to success?

No, to the contrary. It's a test which is known to *not* work under valgrind, so my CI calls the valgrind tests with `--no-suite fails-valgrind` to skip these.

Indeed `t_detects_not_running_in_testbed` is a bit problematic, it fails in emulators. That's why in commit 0cffcea655 I already disabled it for the RPM architectures which are emulated (s390x and arm). 

[Debian's](https://buildd.debian.org/status/logs.php?pkg=umockdev&arch=riscv64) and [Ubuntu's](https://launchpad.net/ubuntu/+source/umockdev/0.17.5-1/+build/23072842) riscv64 builds have been happy for a long time -- I figure they will use QEMU emulation as well instead of actual hardware. But maybe they are just "fast enough" to make that test pass?

There's most definitively a race condition somewhere, I just didn't track it down yet (as it's hard to reproduce locally). I'm happy to do something similar to commit 0cffcea655, is there an arch counterpart for `$RPM_ARCH`? (pun not intended..)

Otherwise I could just generally skip the test on RISCv64, as these days pretty much every builder is emulated still.

XieJiSS, 2022-01-21:
@martinpitt 

Hi, the Arch Linux build system specifies `arch` environment variable while building. For riscv64, its value is `riscv64`.

However, I think I'd prefer the second solution, _i.e._ to skip this test on RISC-V platforms, because I've actually attempted to build umockdev on real `riscv64gc` boards (SiFive Unmatched board, with 4*CPU, but it's performance is not so good), and this problem still exists.

martinpitt, 2022-01-21:
It is annoyingly difficult to determine the current CPU architecture in Vala (something like `#ifdef __riscv` is not passed through).

I tried to run this test case with really little CPU:
```
sudo systemd-run -p CPUQuota=1% -dt --wait sh -ec 'LD_LIBRARY_PATH=. LD_PRELOAD=libumockdev-preload.so ./test-umockdev-vala -p /umockdev-testbed-vala/detects_running_outside_testbed'
```
and even tried 0.1%, but it always passes.

So I'll just mark it as brittle test, so that it gets skipped in package builds.

XieJiSS, 2022-01-21:
Thanks for your quick response!

XieJiSS, 2022-01-21:
@martinpitt By the way, here at PLCT Lab, we are capable for providing SSH access to SiFive Unmatched boards for debug & CI use. If you are willing to, I can apply for a test account, which should be fulfilled in 5 work days.

martinpitt, 2022-01-21:
@XieJiSS : That would be nice actually, to get to the bottom of this! I don't need root privs, but I do need umockdev's build dependencies installed (in Debian/Ubuntu/Fedora etc. one would usually use schroot/mock or a podman container). My [public SSH key](https://www.piware.de/sshkey).

Thanks!

martinpitt, 2022-01-21:
Released as [0.17.6](https://github.com/martinpitt/umockdev/releases/tag/0.17.6)

XieJiSS, 2022-01-23:
@martinpitt Hi, I've send you an email with ssh configs needed to access the RISC-V board.

martinpitt, 2022-01-27:
Reopening -- while this does not block package builds any more, I'd still like to investigate what makes the test actually fail.

So far I built umockdev on @XieJiSS 's riscv64 machine, and the test works just fine. Both in default parallel as well as in `--num-processes 1` mode. I'll try a bit harder to run it in a loop, with some background activity and/or lowering CPU quota, etc.

martinpitt, 2022-01-30:
I ran the test 1000 times in a row:
```
for i in `seq 1000`; do LD_LIBRARY_PATH=b LD_PRELOAD=b/libumockdev-preload.so  b/test-umockdev-vala -p /umockdev-testbed-vala/detects_running_outside_testbed || break; done
```

I then ran the same loop 5 times in parallel as well. I stared at the code, and I really can't see a race condition there -- the two processes are properly synchronized with writing the byte through the pipe. It's just simply delivering the wrong result.

I am reasonably sure that this is *not* a race condition, but a failure that happens under some particular build conditions which are not met by my plain meson build or the Debian/Ubuntu packages. I guess I'll have to litter this thing with printfs and throw it at the Fedora koji builders, which seem to exhibit the same behaviour (at least when I checked last)

martinpitt, 2022-01-30:
The failure is gone from [Koji's s390x build](https://copr.fedorainfracloud.org/coprs/packit/martinpitt-umockdev-172/build/3260094/), presumably because that architecture gets built on real hardware now. But armhfp is still emulated, and it still [fais there](https://download.copr.fedorainfracloud.org/results/packit/martinpitt-umockdev-172/fedora-rawhide-armhfp/03260094-umockdev/builder-live.log.gz).

martinpitt, 2022-01-30:
@XieJiSS : How and where *exactly* are you building umockdev to trigger the failure? So far there is overwhelming evidence that this *only* happens in a QEMU emulated build. So far I've been unable to reproduce this on your board with the standard `meson b && meson test -C b`.

martinpitt, 2022-01-30:
I added a [30 times/seconds retry loop](https://github.com/martinpitt/umockdev/pull/172/commits/683845ae914663c82968d2ecc52411e8bd702d8d), and it 
[does not help](https://download.copr.fedorainfracloud.org/results/packit/martinpitt-umockdev-172/fedora-rawhide-armhfp/03260212-umockdev/builder-live.log.gz) -- the execve() does not fail, but it never actually gets to the ` --test-outside-testbed` evaulation. I added another debug right at the top of main(), and suspiciously the [log](https://copr-be.cloud.fedoraproject.org/results/packit/martinpitt-umockdev-172/fedora-rawhide-armhfp/03260227-umockdev/builder-live.log.gz) contains neither the top-level nor the fork/exec'ed line.

But now I finally found a bug, and it seems to fix the problem :tada: See PR #172.

commit 62466d3e (2022-01-21): Mark /umockdev-testbed-vala/detects_running_outside_testbed as brittle

This also fails in some riscv64 builds (not in Debian's or Ubuntu's,
though). This failure is hard to reproduce, it even passes with

    sudo systemd-run -p CPUQuota=1% -dt --wait sh -ec 'LD_LIBRARY_PATH=. LD_PRELOAD=libumockdev-preload.so ./test-umockdev-vala -p /umockdev-testbed-vala/detects_running_outside_testbed'

So mark it as brittle for now to skip it in normal package builds.

Fixes #169
PR #171 (2022-01-21): Mark /umockdev-testbed-vala/detects_running_outside_testbed as brittle
This also fails in some riscv64 builds (not in Debian's or Ubuntu's,
though). This failure is hard to reproduce, it even passes with

    sudo systemd-run -p CPUQuota=1% -dt --wait sh -ec 'LD_LIBRARY_PATH=. LD_PRELOAD=libumockdev-preload.so ./test-umockdev-vala -p /umockdev-testbed-vala/detects_running_outside_testbed'

So mark it as brittle for now to skip it in normal package builds.

Fixes #169
commit b5b516e3 (2022-01-21): Mark /umockdev-testbed-vala/detects_running_outside_testbed as brittle

This also fails in some riscv64 builds (not in Debian's or Ubuntu's,
though). This failure is hard to reproduce, it even passes with

    sudo systemd-run -p CPUQuota=1% -dt --wait sh -ec 'LD_LIBRARY_PATH=. LD_PRELOAD=libumockdev-preload.so ./test-umockdev-vala -p /umockdev-testbed-vala/detects_running_outside_testbed'

So mark it as brittle for now to skip it in normal package builds.

Fixes #169
commit 71d62cd0 (2022-01-30): tests: Fix /umockdev-testbed-vala/detects_running_outside_testbed

Supply a proper argv[0] to our own test program, to fix the argument
parsing. This regularly failed in architecture-emulated builds.

Also catch failures of execv().

Drop the `BRITTLE_TESTS` skipping, this should now be robust enough.

Fixes #169
closed 2022-01-30 by martinpitt