martinpitt/umockdev #195: [problem]Can I create a fake camera by using this?

Tumb1eweed, 2022-07-15:


martinpitt, 2022-07-15:
In general yes -- the source even [contains a PTP camera](https://github.com/martinpitt/umockdev/tree/main/devices/cameras) :grin: However, this is many years old, and the ioctl record most probably does not work with current libusb versions and such. But it should be possible. Did you already try and it failed, or was this just a high-level question?

martinpitt, 2022-07-15:
(I used this to run a photo import from f-spot and gphoto a couple of years ago, so it at least used to get quite far. But admittedly I haven't looked at that since then, so it may well have gotten broken.)

Tumb1eweed, 2022-07-15:
> In general yes -- the source even [contains a PTP camera](https://github.com/martinpitt/umockdev/tree/main/devices/cameras) 😁 However, this is many years old, and the ioctl record most probably does not work with current libusb versions and such. But it should be possible. Did you already try and it failed, or was this just a high-level question?

oh i just ask before trying, because i want to use to test transferring data through shared memory in ros2 nodes, and the publisher's data is from camera but i don't want to start a camera (no device), so I want to mock camera (mock xoictl particularly)

Tumb1eweed, 2022-07-15:
> (I used this to run a photo import from f-spot and gphoto a couple of years ago, so it at least used to get quite far. But admittedly I haven't looked at that since then, so it may well have gotten broken.)

i forgot to ask, is this tool supporting C++?

benzea, 2022-07-15:
There are no real requirements for the code that you are testing. The library itself is vala (i.e. C), so of course it works well from C++ if you want that.

No idea about ROS and the test.

I suppose that works by mmap'ing a device file and doing a few ioctl's on it. With umockdev, that should work, i.e.:
 1. If a read/write/ioctl happens, you can handle it yourself from the test code
 2. When mmap'ed the underlying file is mmap'ed, I think. So, I *guess* that works if you just create a big enough file on disk in the location and use it as a shared memory.

martinpitt, 2023-02-13:
This is stale and there is nothing actionable. Closing.

closed 2023-02-13 by martinpitt