martinpitt/umockdev #203: Build error with 0.17.15

z3ntu, 2022-11-24:
Since upgrading from 0.17.13 to 0.17.15 building from a tarball fails (seen with Alpine Linux packaging)
```
The Meson build system
Version: 0.64.1
Source dir: /home/builder/aports/community/umockdev/src/umockdev-0.17.15
Build dir: /home/builder/aports/community/umockdev/src/umockdev-0.17.15/output
Build type: native build

meson.build:1:0: ERROR: Command "/home/builder/aports/community/umockdev/src/umockdev-0.17.15/src/getversion.sh" failed with status 1.

A full log can be found at /home/builder/aports/community/umockdev/src/umockdev-0.17.15/output/meson-logs/meson-log.txt
```

martinpitt, 2022-11-24:
Can you please run `sh -ex src/getversion.sh`? If you build from the [release tarbal](https://github.com/martinpitt/umockdev/releases/download/0.17.15/umockdev-0.17.15.tar.xz), these ship a `.version` file and it should read it. It  could be that you umockdev-0.17.15/ has a `.git` directory, but `git` is not installed?

z3ntu, 2022-11-24:
Right, seems since basically forever the package has been built from github-generated tarball instead of release tarball - which has worked fine until the latest version.
```diff
-source="https://github.com/martinpitt/umockdev/archive/$pkgver/umockdev-$pkgver.tar.gz"
+source="https://github.com/martinpitt/umockdev/releases/download/$pkgver/umockdev-$pkgver.tar.xz"
```
Everything builds now, thanks for the pointer! https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/41721/diffs

closed 2022-11-24 by z3ntu

martinpitt, 2022-11-25:
@z3ntu : Nice! :+1: