martinpitt/umockdev #204: Please improve C example documentation

erichschroeter, 2023-01-18:
I'm trying to use umockdev for some regression tests, but I am having a difficult time. Where is `umockdev.h`? I tried manually building with `meson`, thinking that would generate the header, but that still did not produce the header (at least anywhere `find . -name umockdev.h` would find it). Am I missing something obvious?

eli-schwartz, 2023-01-19:
> I tried manually building with `meson`, thinking that would generate the header, but that still did not produce the header (at least anywhere `find . -name umockdev.h` would find it). Am I missing something obvious?

Not precisely obvious, but it is created as a side product of vala.

> Documentation on how to include umockdev using CMake would be ideal. Something like:

How about:

```cmake
find_package(PkgConfig REQUIRED)

pkg_check_modules(UMOCKDEV REQUIRED IMPORTED_TARGET GLOBAL umockdev-1.0)

add_executable(battery battery.c)
target_link_libraries(battery PkgConfig::UMOCKDEV)
```

FetchContent won't work -- that is cmake-exclusive. You can use ExternalProject instead, and define custom build/install commands, before running pkg_check_modules. (The pkg-config search path will be automatically populated with cmake's prefix path.)

martinpitt, 2023-01-19:
Note that `meson` is merely the moral equivalent of `./configure`, it does *not* build anything. See https://github.com/martinpitt/umockdev/#build-test-run for instructions how to build umockdev from source. What exactly did you do? Please copy&paste your commands and the output you got.

I don't know CMake, sorry. I am not going to create documentation for all the C build systems in the world, that would be redundant. If someone wants to use umockdev from a CMake project, they would already know how to use it. It's a standard shared library plus header file, there is nothing special about it.

eli-schwartz, 2023-01-19:
> If someone wants to use umockdev from a CMake project, they would already know how to use it.

(No. Actually, cmake users most frequently do NOT know how to use standard mechanisms such as pkg-config, because cmake discourages you from using it and suggests instead that projects write their own Turing-complete script which heuristically walks the filesystem looking for your library and header by name, and include that script in their own project. That being said I'm not suggesting you be responsible to document it.)

martinpitt, 2023-01-19:
@eli-schwartz : Ugh, I'm sorry to hear. Well, I'm happy to review PRs against the README, but I'm afraid I can't give any advice myself.

martinpitt, 2023-02-13:
Retitling and added "help-wanted". I'm not going to do this myself.