martinpitt/umockdev #206: "Environment" related crashes

hadess, 2023-02-20:
The upower test suite randomly crashes in `getenv()`: https://gitlab.freedesktop.org/upower/upower/-/issues/216

I think that one of the possible reasons is the code that creates the testbed in `src/umockdev.vala` first creates a worked thread, and then sets `UMOCKDEV_DIR`. If the worker thread reads from the environment while the main thread is still setting up the environment, that could lead to a crash.

I would recommend:
- starting the worker thread after the `checked_setenv()` call
- making a copy of the value of `UMOCKDEV_DIR` and passing it as data to the thread you're creating, so it has its own copy of the string and doesn't need to poke at the environment from another thread.

martinpitt, 2023-02-26:
Thanks for the report! I sent a fix to PR #207.

closed 2023-02-26 by martinpitt

hadess, 2023-04-15:
Would it be possible to have this be made available in a release? I can only get one in 10 CI runs to pass in upower right now.

martinpitt, 2023-04-16:
Thanks @hadess for the ping (feel free to do it ealier next time). [0.17.17](https://github.com/martinpitt/umockdev/releases/tag/0.17.17) is released upstream and Fedora koji builds are running (should be on bodhi in a few mins). I also uploaded to Debian unstable.

PR #207 (2023-02-26): umockdev: Fix race with worker thread accessing $UMOCKDEV_DIR
Set the env variable before starting the worker thread, to ensure that it sees it.

Fixes #206
commit bd8757a3 (2023-02-26): umockdev: Fix race with worker thread accessing $UMOCKDEV_DIR

Set the env variable before starting the worker thread, to ensure that
it sees it.

Fixes #206