martinpitt/umockdev #216: 32bit test fails on when built with -D_TIME_BITS=64

kabe-gh, 2023-09-07:
on 32bit environment with -D_TIME_BITS=64 -D_FILE_OFFSET_BITS-64, 
`/usr/bin/meson test` fails as below.
Attached patch seems to fix this, but a dirty hack;
maybe should left out as open issue.
```
==================================== 2/7 =====================================
test:         umockdev / umockdev
start time:   10:45:47
duration:     0.10s
result:       killed by signal 6 SIGABRT
command:      GI_TYPELIB_PATH=/home/kabe/r9builds/umockdev-r9/BUILD/umockdev-0.17.13/redhat-linux-build TOP_SRCDIR=/home/kabe/r9builds
/umockdev-r9/BUILD/umockdev-0.17.13 LD_LIBRARY_PATH=/home/kabe/r9builds/umockdev-r9/BUILD/umockdev-0.17.13/redhat-linux-build PATH=/ho
me/kabe/r9builds/umockdev-r9/BUILD/umockdev-0.17.13/redhat-linux-build:/home/kabe/.local/bin:/home/kabe/bin:/usr/share/Modules/bin:/us
r/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin G_DEBUG=fatal-warnings,fatal-criticals,gc-friendly MALLOC_CHECK_=3 MALLOC_PERTURB_=222
/home/kabe/r9builds/umockdev-r9/BUILD/umockdev-0.17.13/src/umockdev-wrapper /home/kabe/r9builds/umockdev-r9/BUILD/umockdev-0.17.13/red
hat-linux-build/test-umockdev
----------------------------------- stdout -----------------------------------
# random seed: R02S5a4656317f56f8f855170d060b47587d
1..37
# Start of umockdev-testbed tests
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.GTQTA2
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.GTQTA2
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 1 /umockdev-testbed/empty
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.YVXSA2
# DEBUG: umockdev.vala:765: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:769: umockdev_testbed_uevent: sending uevent add for device /sys/devices/extkeyboard1
Bail out! ERROR:../tests/test-umockdev.c:210:t_testbed_add_devicev: assertion failed: (device != NULL)
----------------------------------- stderr -----------------------------------
**
ERROR:../tests/test-umockdev.c:210:t_testbed_add_devicev: assertion failed: (device != NULL)
==============================================================================
```
[patch-time64.patch](https://github.com/martinpitt/umockdev/files/12549479/patch-time64.patch)

kabe-gh, 2023-09-07:
Additional patch for compiling in chroot environment invoked by setarch i686.
[patch-nobypass-selinux.patch](https://github.com/martinpitt/umockdev/files/12549519/patch-nobypass-selinux.patch)

martinpitt, 2023-09-29:
> on 32bit environment with -D_TIME_BITS=64 -D_FILE_OFFSET_BITS-64,

How exactly do you build this? The `BITS-64` is apparently a typo, but:
```
$ meson setup /tmp/t64 --buildtype debug --prefix /usr --werror -D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64
meson.build:1:0: ERROR: Unknown options: "_FILE_OFFSET_BITS, _TIME_BITS"

$ CFLAGS="-D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64" meson setup /tmp/t64 --buildtype debug --prefix /usr --werror &&  meson test -C /tmp/t64/
...
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"

```

I get the same failure with
```diff
--- meson.build
+++ meson.build
@@ -55,6 +55,9 @@ if cc.has_function('__fxstatat', prefix: '#include <sys/stat.h>')
   add_project_arguments('-DHAVE_FXSTATAT', language: 'c')
 endif
 
+add_project_arguments('-D_TIME_BITS=64', language: 'c')
+add_project_arguments('-D_FILE_OFFSET_BITS=64', language: 'c')
+
 meson.add_dist_script(srcdir / 'getversion.sh')
 
```
So I think the `CFLAGS=` approach works in general. 

I tried this in current Debian testing i386. So how exactly did you try this?

kabe-gh, 2023-09-29:
`CFLAGS="-D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64" meson . redhat-linux-build`
`meson compile -C redhat-linux-build --verbose`

emits following error as you saw:

```
In file included from /usr/include/features.h:392,
                 from /usr/include/assert.h:35,
                 from ../src/libumockdev-preload.c:31:
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |
```

kabe-gh, 2023-09-29:
So, first, add #undef _TIME_BITS in src/libumockdev-preload.c:

```
diff -up ./src/libumockdev-preload.c.t64 ./src/libumockdev-preload.c
--- ./src/libumockdev-preload.c.t64     2022-05-30 19:16:30.000000000 +0900
+++ ./src/libumockdev-preload.c 2023-09-29 21:07:37.990392403 +0900
@@ -24,6 +24,9 @@

 /* override -D_FILE_OFFSET_BITS, it breaks us */
 #undef _FILE_OFFSET_BITS
+#ifdef _TIME_BITS
+# undef _TIME_BITS
+#endif

 /* for getting stat64 */
 #define _GNU_SOURCE
```

`meson compile -C redhat-linux-build --verbose`

```
../tests/test-umockdev.c: In function ‘t_testbed_script_replay_evdev_event_framing’:
../tests/test-umockdev.c:1542:8: error: incompatible types when initializing type ‘long unsigned int’ using type ‘struct timeval’
 1542 |       {dummy, 0003, 0000, 2534},
      |        ^~~~~
../tests/test-umockdev.c:1543:8: error: incompatible types when initializing type ‘long unsigned int’ using type ‘struct timeval’
 1543 |       {dummy, 0000, 0000, 0000},
      |        ^~~~~
../tests/test-umockdev.c:1544:8: error: incompatible types when initializing type ‘long unsigned int’ using type ‘struct timeval’
 1544 |       {dummy, 0003, 0000, 2200},
      |        ^~~~~
../tests/test-umockdev.c:1545:8: error: incompatible types when initializing type ‘long unsigned int’ using type ‘struct timeval’
 1545 |       {dummy, 0000, 0000, 0000}
      |        ^~~~~
```

kabe-gh, 2023-09-29:
So fix that too:

```
diff --git a/tests/test-umockdev.c b/tests/test-umockdev.c
index e8a5462..fba19d8 100644
--- a/tests/test-umockdev.c
+++ b/tests/test-umockdev.c
@@ -1536,13 +1536,17 @@ t_testbed_script_replay_evdev_event_framing(UMockdevTestbedFixture * fixture, UN
   int fd;
   char buf[1024];

-  struct timeval dummy = {0};
+#if !defined(_TIME_BITS) && _TIME_BITS != 64
+  struct timeval dummy_timeval = {0};
+#else
+# define dummy_timeval 0,0
+#endif
   /* Simple evdev stream - x coordinate followed by SYN, times 2 */
   struct input_event dummy_events[] = {
-      {dummy, 0003, 0000, 2534},
-      {dummy, 0000, 0000, 0000},
-      {dummy, 0003, 0000, 2200},
-      {dummy, 0000, 0000, 0000}
+      {dummy_timeval, 0003, 0000, 2534},
+      {dummy_timeval, 0000, 0000, 0000},
+      {dummy_timeval, 0003, 0000, 2200},
+      {dummy_timeval, 0000, 0000, 0000}
   };

   umockdev_testbed_add_from_string(fixture->testbed,
```

kabe-gh, 2023-09-29:
You have to re-wrap every functions otherwise redirected to __ioctl_time64 et al in src/libumockdev-preload.c .
hence the large patch in https://github.com/martinpitt/umockdev/issues/216#issue-1885882960 .

commit b947c513 (2024-03-24): preload: Fix building with _FILE_OFFSET_BITS

Unsetting `_FILE_OFFSET_BITS` causes time_t to be treated as 32-bit
instead of the 64-bit that we want with `-D_TIME_BITS=64`.

Fixes #216
PR #236 (2024-03-24): Fix 64-bit time_t on 32 bit architectures
Thanks to @vorlonofportland and @liuzixing for these! Taken from [ubuntu delta](http://launchpadlibrarian.net/718550943/umockdev_0.18.0-1_0.18.0-1ubuntu5.diff.gz), package can be re-synced after landing this and making a new release.

Fixes #216
commit 5e5ee15d (2024-03-24): preload: Fix building with _FILE_OFFSET_BITS

Unsetting `_FILE_OFFSET_BITS` causes time_t to be treated as 32-bit
instead of the 64-bit that we want with `-D_TIME_BITS=64`.

Fixes #216
closed 2024-03-24 by martinpitt