martinpitt/umockdev #220: Set SELinux context

ikerexxe, 2023-12-01:
We'd like to be able to use umockdev as unprivileged users in a SELinux environment. Currently, the SELinux labels aren't set to the correct values for our application to access the files, thus making umockdev unusable in this context.

The request is to provide a way to set up the SELinux context, either by recording it in the umockdev recording, or by setting it via the API.

martinpitt, 2023-12-14:
Right, umockdev-record could put the context into the .umockdev file, and umockdev-run could set it. That has to happen via the API, so that will come first anyway.

ikerexxe, 2023-12-14:
Then, I guess we'll need an API option to set the context, and then a way to "record" and "run" to the file.

martinpitt, 2023-12-14:
Yes, that's what I'm working on now in the [selinux branch](https://github.com/martinpitt/umockdev/tree/selinux). I'll let you review the PR once it is in a consumable shape. But I'm on a sprint this week, so won't finish today; but tomorrow there's a good chance I have something.

commit d73118f5 (2023-12-14): Record and restore SELinux context for mocked /dev nodes

If libselinux is available, record the original node SELinux context
into an internal `__DEVCONTEXT` property, and restore it in
`umockdev-run`. This property can also be set via the API.

Fixes #220
commit 8bc801a8 (2023-12-15): Record and restore SELinux context for mocked /dev nodes

If libselinux is available, record the original node SELinux context
into an internal `__DEVCONTEXT` property, and restore it in
`umockdev-run`. This property can also be set via the API.

Fixes #220
PR #222 (2023-12-18): Record and restore SELinux context for mocked /dev nodes
If libselinux is available, record the original node SELinux context into an internal `__DEVCONTEXT` property, and restore it in `umockdev-run`. This property can also be set via the API.

Fixes #220
closed 2023-12-18 by martinpitt