martinpitt/umockdev #223: Help with mocking input/evdev device

sicelo, 2024-01-29:
I am working on support for (old-school) evdev-based proximity sensors in [iio-sensor-proxy](https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/merge_requests/375).

The test suite uses `umockdev`, and I have generated the required files using `umockdev-record`:
- device, https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/blob/a89df20af1d817ffc60c1df8aa5031abad10a7e6/tests/input-proximity-data/umockdev
- ioctl, https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/blob/a89df20af1d817ffc60c1df8aa5031abad10a7e6/tests/input-proximity-data/ioctl
- events, https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/blob/a89df20af1d817ffc60c1df8aa5031abad10a7e6/tests/input-proximity-data/events

Feeding that data into `evtest` is successful: 

```
nokia-n900:~/iio-sensor-proxy/builddir$ sudo umockdev-run -d ../tests/input-proximity-data/umockdev -i /dev/input/event4=../tests/input-proximity-data/ioctl -e /dev/input/event4=../tests/input-proximity-data/events -- evtest /dev/input/event4
[sudo] password for user: 
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x1 product 0x1 version 0x100
Input device name: "gpio_keys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 152 (KEY_SCREENLOCK)
    Event code 212 (KEY_CAMERA)
    Event code 528 (KEY_CAMERA_FOCUS)
  Event type 5 (EV_SW)
    Event code 9 (SW_CAMERA_LENS_COVER) state 1
    Event code 10 (SW_KEYPAD_SLIDE) state 0
    Event code 11 (SW_FRONT_PROXIMITY) state 0
    Event code 16 (?) state 1
Properties:
Testing ... (interrupt to exit)
Event: time 1706552688.289347, type 5 (EV_SW), code 11 (SW_FRONT_PROXIMITY), value 1
Event: time 1706552688.289347, -------------- SYN_REPORT ------------
Event: time 1706552693.000007, type 5 (EV_SW), code 11 (SW_FRONT_PROXIMITY), value 0
Event: time 1706552693.000007, -------------- SYN_REPORT ------------
Event: time 1706552696.670935, type 5 (EV_SW), code 11 (SW_FRONT_PROXIMITY), value 1
Event: time 1706552696.670935, -------------- SYN_REPORT ------------
^C
```

However, with `iio-sensor-proxy`, the exact same data fails on the `ioctl` call in https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/blob/e0995ca1287c6c2797f7e8b20cc5e7d3fee7ca0c/src/drv-input-proximity.c#L162-L163, as shown below:

```
nokia-n900:~/iio-sensor-proxy/builddir$ sudo G_MESSAGES_DEBUG=all umockdev-run -d ../tests/input-proximity-data/umockdev -i /dev/input/event4=../tests/input-proximity-data/ioctl -e /dev/input/event4=../tests/input-proximity-data/events -- src/iio-sensor-proxy -rv
** (umockdev-run:12069): DEBUG: 23:22:23.068: umockdev.vala:123: Created udev test bed /tmp/umockdev.HLTCI2
** (umockdev-run:12069): DEBUG: 23:22:23.071: umockdev.vala:1297: parsing device description for /devices/platform/gpio_keys/input/input4/event4
** (umockdev-run:12069): DEBUG: 23:22:23.090: umockdev.vala:1366: creating device /devices/platform/gpio_keys/input/input4/event4 (subsystem input)
** (umockdev-run:12069): DEBUG: 23:22:23.114: umockdev.vala:1430: create_node_for_device: creating pty device /tmp/umockdev.HLTCI2/dev/input/event4: got pty /dev/pts/5
** (umockdev-run:12069): DEBUG: 23:22:23.123: umockdev.vala:1451: create_node_for_device: creating ptymap symlink /tmp/umockdev.HLTCI2/dev/.ptymap/_dev_pts_5
** (umockdev-run:12069): DEBUG: 23:22:23.127: umockdev.vala:1297: parsing device description for /devices/platform/gpio_keys/input/input4
** (umockdev-run:12069): DEBUG: 23:22:23.154: umockdev.vala:1366: creating device /devices/platform/gpio_keys/input/input4 (subsystem input)
** (umockdev-run:12069): DEBUG: 23:22:23.190: umockdev.vala:1297: parsing device description for /devices/platform/gpio_keys
** (umockdev-run:12069): DEBUG: 23:22:23.192: umockdev.vala:1366: creating device /devices/platform/gpio_keys (subsystem platform)
(umockdev-run:12069): GLib-GIO-DEBUG: 23:22:23.221: _g_io_module_get_default: Found default implementation local (GLocalVfs) for *gio-vfs*
** (umockdev-run:12069): DEBUG: 23:22:23.250: umockdev.vala:1702: ScriptRunner[/dev/input/event4]: start
** (umockdev-run:12069): DEBUG: 23:22:23.262: ScriptRunner[/dev/input/event4]: read op; sleeping 0 ms
** (umockdev-run:12069): DEBUG: 23:22:23.262: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'p\xed\xb7eCj^D^@^E^@^K^@^A^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:23.263: ScriptRunner[/dev/input/event4]: read op; sleeping 0 ms
** (umockdev-run:12069): DEBUG: 23:22:23.268: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'p\xed\xb7eCj^D^@^@^@^@^@^@^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:23.269: ScriptRunner[/dev/input/event4]: read op; sleeping 4711 ms
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.309: Starting iio-sensor-proxy version 3.5
(iio-sensor-proxy:12074): GLib-GIO-DEBUG: 23:22:23.345: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.450: Found input proximity sensor at /sys/devices/platform/gpio_keys/input/input4/event4
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.463: Found device /sys/devices/platform/gpio_keys/input/input4/event4 of type proximity at Input proximity
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.550: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.618: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (iio-sensor-proxy:12074): DEBUG: 23:22:23.680: Handling driver refcounting method 'ClaimProximity' for proximity device

** (iio-sensor-proxy:12074): WARNING **: 23:22:23.693: could not determine sensor state, ioctl EVIOCGSW on /sys/devices/platform/gpio_keys/input/input4/event4 failed
** (umockdev-run:12069): DEBUG: 23:22:27.981: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'u\xed\xb7e^G^@^@^@^E^@^K^@^@^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:27.983: ScriptRunner[/dev/input/event4]: read op; sleeping 0 ms
** (umockdev-run:12069): DEBUG: 23:22:27.984: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'u\xed\xb7e^G^@^@^@^@^@^@^@^@^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:27.985: ScriptRunner[/dev/input/event4]: read op; sleeping 3670 ms
** (umockdev-run:12069): DEBUG: 23:22:31.658: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'x\xed\xb7e\xd7<^J^@^E^@^K^@^A^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:31.661: ScriptRunner[/dev/input/event4]: read op; sleeping 0 ms
** (umockdev-run:12069): DEBUG: 23:22:31.664: umockdev.vala:1712: ScriptRunner[/dev/input/event4]: read op after sleep; writing data 'x\xed\xb7e\xd7<^J^@^@^@^@^@^@^@^@^@'
** (umockdev-run:12069): DEBUG: 23:22:31.665: umockdev.vala:1757: ScriptRunner[/dev/input/event4]: end of script /tmp/evemu.JLWGI2.script, closing
** (umockdev-run:12069): DEBUG: 23:22:31.666: umockdev.vala:1746: ScriptRunner[/dev/input/event4]: not running any more, ending thread
```

On the actual hardware (on which the umockdev data was generated), the code runs fine and `iio-sensor-proxy` successfully monitors the state of the proximity sensor as expected.

I would appreciate any ideas for where I might have made a mistake, and/or how to get the above test to succeed.

benzea, 2024-01-30:
In ioctl_tree.c, I see:

  I_NAMED_SIMPLE_STRUCT_IN(EVIOCGSW(32), "EVIOCGSW", 0, ioctl_insertion_parent_stateless),

half thinking that it only accepts 32 bytes for the buffer length (in this case `SW_MAX / 8`). Maybe that should be something like:

  I_NAMED_SIMPLE_STRUCT_IN(EVIOCGSW(0), "EVIOCGSW", 32, ioctl_insertion_parent_stateless),

Instead?

sicelo, 2024-02-01:
I got it working by using `iio-sensor-proxy` itself during the ioctl recording stage, instead of `evtest` :-)

closed 2024-02-01 by sicelo