martinpitt/umockdev #226: 0.17.18: is not gcc 14.x ready

kloczek, 2024-01-31:
Looks like last version build fails with latest gcc 14.x which is now used in fedora rawhide.
<details>
<summary>Build fails with</summary>

```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ ninja -k 0
[1/17] Compiling C object umockdev-record.p/meson-generated_src_umockdev-record.c.o
FAILED: umockdev-record.p/meson-generated_src_umockdev-record.c.o
/usr/bin/gcc -Iumockdev-record.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ umockdev-record.p/meson-generated_src_umockdev-record.c.o -MF umockdev-record.p/meson-generated_src_umockdev-record.c.o.d -o umockdev-record.p/meson-generated_src_umockdev-record.c.o -c umockdev-record.p/src/umockdev-record.c
../src/umockdev-record.vala: In function ‘all_devices’:
../src/umockdev-record.vala:61:48: error: passing argument 1 of ‘_vala_array_dup1’ from incompatible pointer type [-Wincompatible-pointer-types]
   61 |     return devs.data;
      |                                                ^
      |                                                |
      |                                                void **
umockdev-record.c:418:27: note: expected ‘gchar **’ {aka ‘char **’} but argument is of type ‘void **’
../src/umockdev-record.vala:61:72: error: pointer type mismatch in conditional expression [-Wincompatible-pointer-types]
   61 |     return devs.data;
      |                                                                        ^
../src/umockdev-record.vala:61:30: note: first expression has type ‘gchar **’ {aka ‘char **’}
   61 |     return devs.data;
      |                              ^
../src/umockdev-record.vala:61:74: note: second expression has type ‘void **’
   61 |     return devs.data;
      |                                                                          ^
[2/17] Compiling C object umockdev-record.p/meson-generated_src_umockdev-pcap.c.o
FAILED: umockdev-record.p/meson-generated_src_umockdev-pcap.c.o
/usr/bin/gcc -Iumockdev-record.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ umockdev-record.p/meson-generated_src_umockdev-pcap.c.o -MF umockdev-record.p/meson-generated_src_umockdev-pcap.c.o.d -o umockdev-record.p/meson-generated_src_umockdev-pcap.c.o -c umockdev-record.p/src/umockdev-pcap.c
glib-2.0.vapi: In function ‘vala_g_array_remove_index’:
glib-2.0.vapi:6131:9: error: assignment to ‘void **’ from incompatible pointer type ‘gchar *’ {aka ‘char *’} [-Wincompatible-pointer-types]
[3/17] Compiling C object libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o
FAILED: libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o
/usr/bin/gcc -Ilibumockdev.so.0.3.0.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -fPIC -MD -MQ libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o -MF libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o.d -o libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o -c libumockdev.so.0.3.0.p/src/umockdev-pcap.c
glib-2.0.vapi: In function ‘vala_g_array_remove_index’:
glib-2.0.vapi:6131:9: error: assignment to ‘void **’ from incompatible pointer type ‘gchar *’ {aka ‘char *’} [-Wincompatible-pointer-types]
[4/17] Compiling C object libumockdev.so.0.3.0.p/meson-generated_src_umockdev-ioctl.c.o
FAILED: libumockdev.so.0.3.0.p/meson-generated_src_umockdev-ioctl.c.o
/usr/bin/gcc -Ilibumockdev.so.0.3.0.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -fPIC -MD -MQ libumockdev.so.0.3.0.p/meson-generated_src_umockdev-ioctl.c.o -MF libumockdev.so.0.3.0.p/meson-generated_src_umockdev-ioctl.c.o.d -o libumockdev.so.0.3.0.p/meson-generated_src_umockdev-ioctl.c.o -c libumockdev.so.0.3.0.p/src/umockdev-ioctl.c
../src/umockdev-ioctl.vala: In function ‘umockdev_ioctl_client_class_init’:
../src/umockdev-ioctl.vala:376:90: error: passing argument 5 of ‘g_signal_new’ from incompatible pointer type [-Wincompatible-pointer-types]
  376 |         GLib.Signal.@new("handle-ioctl", typeof(IoctlClient), GLib.SignalFlags.RUN_LAST, 0, signal_accumulator_true_handled, null, null, typeof(bool), 0);
      |                                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                          |
      |                                                                                          gboolean (*)(GSignalInvocationHint *, GValue *, GValue *, void *) {aka int (*)(struct _GSignalInvocationHint *, struct _GValue *, struct _GValue *, void *)}
In file included from /usr/include/glib-2.0/gobject/gobject.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:31,
                 from /usr/include/glib-2.0/glib-object.h:24,
                 from libumockdev.so.0.3.0.p/src/umockdev-ioctl.c:102:
/usr/include/glib-2.0/gobject/gsignal.h:308:66: note: expected ‘GSignalAccumulator’ {aka ‘int (*)(struct _GSignalInvocationHint *, struct _GValue *, const struct _GValue *, void *)’} but argument is of type ‘gboolean (*)(GSignalInvocationHint *, GValue *, GValue *, void *)’ {aka ‘int (*)(struct _GSignalInvocationHint *, struct _GValue *, struct _GValue *, void *)’}
  308 |                                              GSignalAccumulator  accumulator,
      |                                              ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
../src/umockdev-ioctl.vala:377:89: error: passing argument 5 of ‘g_signal_new’ from incompatible pointer type [-Wincompatible-pointer-types]
  377 |         GLib.Signal.@new("handle-read", typeof(IoctlClient), GLib.SignalFlags.RUN_LAST, 0, signal_accumulator_true_handled, null, null, typeof(bool), 0);
      |                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                         |
      |                                                                                         gboolean (*)(GSignalInvocationHint *, GValue *, GValue *, void *) {aka int (*)(struct _GSignalInvocationHint *, struct _GValue *, struct _GValue *, void *)}
[... cut ...]
In file included from /usr/include/glib-2.0/glib.h:52,
                 from libumockdev.so.0.3.0.p/src/umockdev.c:46:
/usr/include/glib-2.0/glib/ghash.h:114:61: note: expected ‘GHFunc’ {aka ‘void (*)(void *, void *, void *)’} but argument is of type ‘void (*)(const void *, const void *, void *)’
  114 |                                             GHFunc          func,
      |                                             ~~~~~~~~~~~~~~~~^~~~
ninja: build stopped: cannot make progress due to previous errors.
```
</details>

martinpitt, 2024-01-31:
Ah, that looks very similar to commit 539af0b4f36a40d1f85529cbeef401667e64e7f3 . [The log](https://github.com/martinpitt/umockdev/actions/runs/7691028592/job/20955708817) says "gcc 13.2.1", but maybe they already enabled some options which gcc 14 now does by default.

There isn't much that I can do here on the umockdev side, aside from making the errors non-fatal. Vala doesn't have `const`, and I suppose valac needs some updates to build with gcc 14 without all these warnings.

So I'll look into adjusting the warnings.

kloczek, 2024-01-31:
> Ah, that looks very similar to commit [539af0b](https://github.com/martinpitt/umockdev/commit/539af0b4f36a40d1f85529cbeef401667e64e7f3) . [The log](https://github.com/martinpitt/umockdev/actions/runs/7691028592/job/20955708817) says "gcc 13.2.1", but maybe they already enabled some options which gcc 14 now does by default.

That is not gona happen. clang/llvm is going to change that warning to error in upcoming 18.x as well because it causes many issues (sometimes even CVSs) on porting code from 32 to 64 bits.

martinpitt, 2024-02-01:
@kloczek I'm afraid I can't parse your recent answer. My latest comment wasn't a question, just an observation.

Anyway, umockdev does run CI on rawhide, and e.g. the [most recent run](https://github.com/martinpitt/umockdev/actions/runs/7724603820/job/21057076941) passed with gcc-14.0.1-0.3.fc40.x86_64 . How exactly did you build this? Or did you use a newer gcc version which isn't in Rawhide yet?

kloczek, 2024-02-01:
https://github.com/martinpitt/umockdev/actions/runs/7724603820/job/21057076941#step:3:1342
In that line you are executing custom meson command instead that one which is used during regular %meson rpm macro which adds LTO, hardening and `-Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3` to the $CFLAGS.
You can read about `-D_FORTIFY_SOURCE=3` on https://developers.redhat.com/articles/2022/09/17/gccs-new-fortification-level#1__a_new_builtin_provides_enhanced_buffer_size_detection

martinpitt, 2024-03-02:
Sorry, that "in that line" log reference is gone. I take it you mean the various `meson setup` calls in https://github.com/martinpitt/umockdev/blob/main/tests/run-dnf . But the [spec file build rule](https://github.com/martinpitt/umockdev/blob/main/packaging/umockdev.spec#L37) uses the standard `%meson` macros, Fedora Rawhide has had gcc 14 for a while, and everything seems happy -- PRs run the "proper" rpm build in packit, and of course the real Fedora packages too. `run-dnf` also has a "build with fortify" step whose output you can see [here](https://github.com/martinpitt/umockdev/actions/runs/8121044074/job/22198949766#step:3:7711).

Note that the fatal warnings were fallout from meson 0.14, not gcc. These were fixed yesterday in PR #229. Fedora rawhide does not yet have meson 0.14 (it's still in development), but you may have it from somewhere?

I take it this is/wasa a duplicate of #228. But of course if you see a scenario where the build fails, please yell here (or open a new issue). Thanks!

kloczek, 2024-03-02:
Any chance to make release with fix .. soon? 🤔

kloczek, 2024-03-02:
Just tested 0.17.18 + current master patches (up to f8fb070a) and build still fails
<details>
<summary>Build fails with</summary>

```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ ninja -j 0 -k 0
[1/17] Compiling C object umockdev-record.p/meson-generated_src_umockdev-record.c.o
FAILED: umockdev-record.p/meson-generated_src_umockdev-record.c.o
/usr/bin/gcc -Iumockdev-record.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ umockdev-record.p/meson-generated_src_umockdev-record.c.o -MF umockdev-record.p/meson-generated_src_umockdev-record.c.o.d -o umockdev-record.p/meson-generated_src_umockdev-record.c.o -c umockdev-record.p/src/umockdev-record.c
../src/umockdev-record.vala: In function ‘all_devices’:
../src/umockdev-record.vala:64:9: error: assignment to ‘gchar **’ {aka ‘char **’} from incompatible pointer type ‘void **’ [-Wincompatible-pointer-types]
   64 |     return devs.steal();
      |         ^
[2/17] Compiling C object umockdev-record.p/meson-generated_src_umockdev-pcap.c.o
FAILED: umockdev-record.p/meson-generated_src_umockdev-pcap.c.o
/usr/bin/gcc -Iumockdev-record.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ umockdev-record.p/meson-generated_src_umockdev-pcap.c.o -MF umockdev-record.p/meson-generated_src_umockdev-pcap.c.o.d -o umockdev-record.p/meson-generated_src_umockdev-pcap.c.o -c umockdev-record.p/src/umockdev-pcap.c
glib-2.0.vapi: In function ‘vala_g_array_remove_index’:
glib-2.0.vapi:6139:9: error: assignment to ‘void **’ from incompatible pointer type ‘gchar *’ {aka ‘char *’} [-Wincompatible-pointer-types]
[3/17] Compiling C object libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o
FAILED: libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o
/usr/bin/gcc -Ilibumockdev.so.0.3.0.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -fPIC -MD -MQ libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o -MF libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o.d -o libumockdev.so.0.3.0.p/meson-generated_src_umockdev-pcap.c.o -c libumockdev.so.0.3.0.p/src/umockdev-pcap.c
glib-2.0.vapi: In function ‘vala_g_array_remove_index’:
glib-2.0.vapi:6139:9: error: assignment to ‘void **’ from incompatible pointer type ‘gchar *’ {aka ‘char *’} [-Wincompatible-pointer-types]
[4/17] Compiling C object umockdev-record.p/meson-generated_src_umockdev-ioctl.c.o
FAILED: umockdev-record.p/meson-generated_src_umockdev-ioctl.c.o
/usr/bin/gcc -Iumockdev-record.p -I. -I.. -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -w -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -MD -MQ umockdev-record.p/meson-generated_src_umockdev-ioctl.c.o -MF umockdev-record.p/meson-generated_src_umockdev-ioctl.c.o.d -o umockdev-record.p/meson-generated_src_umockdev-ioctl.c.o -c umockdev-record.p/src/umockdev-ioctl.c
../src/umockdev-ioctl.vala: In function ‘umockdev_ioctl_client_class_init’:
../src/umockdev-ioctl.vala:376:90: error: passing argument 5 of ‘g_signal_new’ from incompatible pointer type [-Wincompatible-pointer-types]
  376 |         GLib.Signal.@new("handle-ioctl", typeof(IoctlClient), GLib.SignalFlags.RUN_LAST, 0, signal_accumulator_true_handled, null, null, typeof(bool), 0);
      |                                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
In file included from /usr/include/glib-2.0/glib.h:52,
                 from libumockdev.so.0.3.0.p/src/umockdev.c:46:
/usr/include/glib-2.0/glib/ghash.h:114:61: note: expected ‘GHFunc’ {aka ‘void (*)(void *, void *, void *)’} but argument is of type ‘void (*)(const void *, const void *, void *)’
  114 |                                             GHFunc          func,
      |                                             ~~~~~~~~~~~~~~~~^~~~
ninja: build stopped: cannot make progress due to previous errors.
```
</details>

kloczek, 2024-03-02:
Just in case I'm removing all `-Wno-error=` from main meson.build.
Masing all those errors by ignore them is not good approach).

kloczek, 2024-03-02:
> Do you mean you get these errors when you remove them (well, that's expected then) or you _also_ tried to remove them?

Yes.
I'm doing that to have full visibility of additional warning flags in rpm packaging procedure in which I can easy able swich those flags on/off.

> Actually "fixing" the warnings would mean to rewrite umockdev from Vala to something else. That's a huge undertaking (and far beyond what I could do in the little time that I have to maintain this project).

Yes but at lease one ot those warnings as you've mentioned that is not related to vala.
Vala is not pain in the a*s  😞 
Gnome 46 GA is approaching and its C code generator still currently not able generate clean code. Currently into master is merged (IMO JFDI) patch which adds tons of masking those issues pragmas)

martinpitt, 2024-03-03:
OK, now that I know what you mean, I retitle/reopen this. I'm going to ignore all the `unused-*` warnings -- they are "mostly harmless", that's just Vala → C code conversion, and the C compiler optimizes these away anyway. Also, there's not much that I can do about `discarded-qualifiers` -- the Vala language doesn't have `const`.

But I'll see if something can be done about the two interesting and also potentially bad ones: `incompatible-pointer-types` and `pointer-sign`.

martinpitt, 2024-03-03:
`pointer-sign` is a bug in the Vala bindings, I just reported https://gitlab.gnome.org/GNOME/vala/-/issues/1529. I'll add a comment to meson.build.

martinpitt, 2024-03-03:
And finally, for `incompatible-pointer-types` I get a lot of junk in fully auto-generated code, like
```
umockdev.c:12310:6: note: in expansion of macro ‘g_once_init_enter’
umockdev.c: In function ‘umockdev_socket_server_get_type’:
/usr/include/glib-2.0/glib/gatomic.h:131:5: error: argument 2 of ‘__atomic_load’ discards ‘volatile’ qualifier [-Werror=incompatible-pointer-types]
  131 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ^~~~~~~~~~~~~
```

or
```
umockdev.c: In function ‘umockdev_testbed_get_type’:
/usr/include/glib-2.0/glib/gatomic.h:131:5: error: argument 2 of ‘__atomic_load’ discards ‘volatile’ qualifier [-Werror=incompatible-pointer-types]
  131 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ^~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:262:7: note: in expansion of macro ‘g_atomic_pointer_get’
  262 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
umockdev.c:10106:6: note: in expansion of macro ‘g_once_init_enter’
```

The ones where umockdev has a handle on at least in principle are these:
```
./../source/src/umockdev-record.vala:64:9: error: assignment to ‘gchar **’ {aka ‘char **’} from incompatible pointer type ‘void **’ [-Werror=incompatible-pointer-types]
   64 |     return devs.steal();
      |         ^
```

The Vala code is correct:
```vala
    var devs = new GenericArray<string>();
    devices_from_dir("/sys/devices", ref devs);
    return devs.steal();
```

and the [template declaration](https://valadoc.org/glib-2.0/GLib.GenericArray.steal.html) says that it  returns a properly typed list.  So Vala is missing a type cast in its generated code.

------

```
../../source/src/umockdev-ioctl.vala:376:90: error: passing argument 5 of ‘g_signal_new’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  376 |         GLib.Signal.@new("handle-ioctl", typeof(IoctlClient), GLib.SignalFlags.RUN_LAST, 0, signal_accumulator_true_handled, null, null, typeof(bool), 0);
      |                                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                          |
      |                                                                                          gboolean (*)(GSignalInvocationHint *, GValue *, GValue *, void *) {aka int (*)(struct _GSignalInvocationHint *, struct _GValue *, struct _GValue *, void *)}
In file included from /usr/include/glib-2.0/gobject/gobject.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:31:
/usr/include/glib-2.0/gobject/gsignal.h:308:66: note: expected ‘GSignalAccumulator’ {aka ‘int (*)(struct _GSignalInvocationHint *, struct _GValue *, const struct _GValue *, void *)’} but argument is of type ‘gboolean (*)(GSignalInvocationHint *, GValue *, GValue *, void *)’ {aka ‘int (*)(struct _GSignalInvocationHint *, struct _GValue *, struct _GValue *, void *)’}
  308 |                                              GSignalAccumulator  accumulator,
      |                                              ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~

```
(and two similar ones for the other two signal handlers, and another group in line 763ff.)

That is really confusing: Both [vala](https://valadoc.org/gobject-2.0/GLib.SignalAccumulator.html) and [C](https://docs.gtk.org/gobject/callback.SignalAccumulator.html) declare the accumulator return value as boolean, and so do the actual C headers:
```
/usr/include/glib-2.0/gobject/gsignal.h:typedef gboolean (*GSignalAccumulator)	(GSignalInvocationHint *ihint,
```
where does the `int` return value come from???

The other difference is that the `handler_return` argument is `const` in C, but it's not in the Vala declaration. And trying to declare it as `const` is a syntax error.

-----

```
../../source/src/umockdev-ioctl.vala: In function ‘umockdev_ioctl_base_unregister_all’:
../../source/src/umockdev-ioctl.vala:848:41: error: passing argument 2 of ‘g_hash_table_foreach_remove’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  848 |             listeners.foreach_remove((key, val) => {
      |                                         ^~~~~~~~~~~~         
      |                                         |
[... 17 lines cut ...]
```

I don't see how to fix this. It's autogenerated code from a lambda, and `const void*` isn't a thing in Vala.

------

```
./../source/tests/test-umockdev-vala.vala: In function ‘t_mt_uevent’:
../../source/tests/test-umockdev-vala.vala:937:30: error: passing argument 1 of ‘g_udev_client_new’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  937 |   var gudev = new GUdev.Client ({"pci"});
      |                              ^~~~~~
      |                              |
      |                              gchar ** {aka char **}
In file included from /usr/include/gudev-1.0/gudev/gudev.h:14,
                 from test-umockdev-vala.p/tests/test-umockdev-vala.c:29:
/usr/include/gudev-1.0/gudev/gudevclient.h:74:77: note: expected ‘const gchar * const*’ {aka ‘const char * const*’} but argument is of type ‘gchar **’ {aka ‘char **’}
   74 | GUdevClient *g_udev_client_new                         (const gchar* const *subsystems);
      |                                                         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
```

[... cut ...] ck and a hard place here -- it can't *always* generate these as `const`, as a lot of existing C APIs don't use `const` where they should. Also, the C compiler is overly picky here -- if an API expects const, but you give it a writable array, that should be perfectly fine.

----

So in conclusion, I didn't find a single place to actually fix, and these are just Vala limitations. See also e.g. https://gitlab.gnome.org/GNOME/vala/-/issues/1119 where upstream gave up on `-Wincompatible-pointer-type`

martinpitt, 2024-03-03:
I sent PR #232 which quiesces the `unused-*` bits and adds some comments/references, but I don't see what else to do here, sorry.

closed 2024-03-03 by martinpitt

kloczek, 2024-03-03:
> And finally, for `incompatible-pointer-types` I get a lot of junk in fully auto-generated code, like

Do you have any idea how to solve those issues in vala code to generate clean code? 🤔

PR #232 (2024-03-03): Hide C compiler warnings about unused items, explain some others
These are numerous, but harmless -- that's just how Vala → C generation works. It takes attention away from the more interesting warnings.

Add explanation and bug reference for the others.

Fixes #226
commit 1c0098ee (2024-03-03): Hide C compiler warnings about unused items, explain some others

These are numerous, but harmless -- that's just how Vala → C generation
works. It takes attention away from the more interesting warnings.

Add explanation and bug reference for the others.

Fixes #226