martinpitt/umockdev #228: Build failing when using meson 1.4.0rc1

DimStar77, 2024-02-27:
In order to test the impact of Meson 1.4.0 , I upgraded a test repository of openSUSE Tumbleweed to this new version,

Out of the builds, umockdev is failing (seems its failing similar to what that HACK commit addresses, which was done in Gentoo? Did Gentoo update to a 1.4 pre-release of meson?)

I get plenty of these errors:

```
[   12s] umockdev-record.p/src/umockdev-record.c:1141:22: error: unused variable ‘_line_collection_size_’ [-Werror=unused-variable]
[   12s]  1141 |                 gint _line_collection_size_ = 0;
[   12s]       |                      ^~~~~~~~~~~~~~~~~~~~~~
```

Simply going back to meson 1.3.2 makes umockdev build pass (CC @jpakkane / meson release maintainer)

(full build log can be found at https://build.opensuse.org/build/GNOME:Next/openSUSE_Factory/x86_64/umockdev/_log )

DimStar77, 2024-02-27:
git bisected this on the meson side to https://github.com/mesonbuild/meson/commit/5f659af870011e74299d1455a65c2cd5f5ace51f - aka https://github.com/mesonbuild/meson/pull/12597

martinpitt, 2024-02-28:
Oooh thanks @DimStar77 for figuring that out! :100: I first saw this in the Gentoo build, and suspected gcc 14 or related, but couldn't find a meaningful difference. So I applied commit 539af0b4f36a40d1f85529cbeef401667e64e7f3.

So I suppose that should just become the new default. This makes sense at last!

DimStar77, 2024-02-28:
I first saw this in the Gentoo build, and suspected gcc 14 or related, but couldn't find a meaningful difference. 

gentoo backported that vala fix from meson 1.4 into their package with https://github.com/gentoo/gentoo/pull/34141

DimStar77, 2024-02-28:
> So I suppose that should just become the new default. This makes sense at last!

on openSUSE, I added this to umockdev.spec:
`export CFLAGS="-Wno-error=unused-function -Wno-error=unused-value -Wno-error=unused-variable %{optflags}"`

closed 2024-03-01 by martinpitt

thesamesam, 2024-03-03:
By the way, feel free to CC me if there's anything like this which shows up for Gentoo which I might be able to give insight on. Happy to help!

martinpitt, 2024-03-03:
Thanks @thesamesam It seems to be quite happy now, aside from the ugliness of #226.. (but that's not Gentoo specific at all).

PR #229 (2024-03-01): Make vala C code errors non-fatal
meson 1.4 stopped hiding valac's warnings [1], which fails compilation
due to our `-Werror=` settings. There's not much that we can do at this
point, so stop making them non-fatal.

This reverts the `gentoo-hacks` option from commit 539af0b4f36. Turns
out that this wasn't a compiler change in Gentoo after all, but that
Gentoo backported [2] that meson change at that time.

Many thanks to Dominique Leuenberger @DimStar77 for tracking down the
root cause!

Fixes #228

[1] https://github.com/mesonbuild/meson/commit/5f659af870011e
[2] https://github.com/gentoo/gen [... cut]
commit 7977c859 (2024-03-01): Make vala C code errors non-fatal

meson 1.4 stopped hiding valac's warnings [1], which fails compilation
due to our `-Werror=` settings. There's not much that we can do at this
point, so stop making them non-fatal.

This reverts the `gentoo-hacks` option from commit 539af0b4f36. Turns
out that this wasn't a compiler change in Gentoo after all, but that
Gentoo backported [2] that meson change at that time.

Many thanks to Dominique Leuenberger @DimStar77 for tracking down the
root cause!

Fixes #228

[1] https://github.com/mesonbuild/meson/commit/5f659af870011e
[2] https://github.com/gentoo/ge [... cut]