martinpitt/umockdev #230: nix test is failing

martinpitt, 2024-03-02:
3 weeks ago, the nix test has [started to fail](https://github.com/martinpitt/umockdev/actions/workflows/tests.yml):
```
meson.build:2:11: ERROR: Command `/build/source/src/getversion.sh` failed with status 1.
```

This is annoying to debug: with `DEBUG=1 tests/run-nix` I can attach to the container with `podman exec -itl bash`, but the log says
```
To attach install cntr and run the following command as root:

   cntr attach -t command cntr-/nix/store/3j73y1b58xhcrffl213ksc672vacij95-umockdev-0.17.18
```

but that doesn't actually work:
```
cannot mount filesystem: Cannot raise file descriptor limit: EPERM: Operation not permitted
```

So I went on with some `set -x` debugging, and it shows 
```
+ git -C /build/source describe
/build/source/src/getversion.sh: line 4: git: not found
```
which isn't obvious -- `git` **is** installed in the container, just somehow not in the build environment. For a long time, [adding it as a build dependency](https://github.com/martinpitt/umockdev/blob/7977c8599316e010bb40591386df7458297513c0/tests/run-nix#L31) with `buildInputs = attrs.buildInputs ++ [ pkgs.git ];` worked, but somehow it seems to have stopped. 

@jtojnar @flokli do you know what's going on here and how to debug this? Possibly also how to provide a sane interactive debug environment in that "inner" build container, so that this can be debugged more interactively?

martinpitt, 2024-03-02:
I also tried `nix-env -i cntr`, but that doesn't seem to do anything useful or obvious.

I added commit f8fb070a378fe7883982774abd2d3b122ae86632 which fixes meson log printing. The [log](https://github.com/martinpitt/umockdev/actions/runs/8121012054/job/22198884014#step:3:606) now shows the error properly. I still can't make sense of it, though.

jtojnar, 2024-03-02:
Sounds like a downfall of https://github.com/NixOS/nixpkgs/commit/58762560bc3e8843bdfcada1b04ebcd019c5529e. If a dependency is expected at build-time, it should go to `nativeBuildInputs`.

closed 2024-03-03 by martinpitt