martinpitt/umockdev #237: remove_device does not trigger "remove" uevent

bobhenz-jabil, 2024-04-12:
First of all... this package is awesome for testing!

I would like to request an enhancement though...

While `add_device()` triggers an "add" uevent. Currently `remove_device()` doesn't trigger a "remove" uevent. I find that I have to first call `testbed.uevent(syspath, "remove")` then call `testbed.remove_device(syspath)`. This doesn't feel quite right to me since I don't have to do that when I'm adding a device.

In addition, the comment for `remove_device()` says "Note that this will also remove all child devices". So it would be _really_ nice if a "remove" uevent was generated for the children as well when a device is removed. (Similar to what happens on hardware.)

Again, I appreciate your work here. Thank you.

martinpitt, 2024-04-18:
You are right, it should trigger a remove uevent, for symmetry and also to better model what happens on a real system. I suppose nobody complained so far as that's a lot less common than adding :grin: 

I'll try to work on this over the weekend, but if you want to give it a try, please do!

bobhenz-jabil, 2024-04-19:
I might be able to look at it next week, I think the difficulty will be in generating uevents for the children but I need to understand the code better to understand how to do that.

bobhenz-jabil, 2024-04-23:
Just to state what I see as the difficulties here (I might be wrong on some of this):
1. A device can be added with the name `foo/bar/baz`, or 3 devices could be added called `foo`, `bar`, `baz`. I think the resulting directory structure would be the same right?
2. The remove event needs to include the attributes and properties that the add event contained.

Therefore, I have a design proposal:
 A. A hidden file (e.g. `.umockdev`) is added into the directory of each added device. The existence of this file would mean a device was added at this level.
B.  When a parent device is removed, the directory structure of all subdirectories under it would be walked (recursively) and any directory found with a file of that name in it, would have a uevent('remove') call made for it.
C. I don't know if the attributes and properties are already stored in the directory of devices, but if not (or if it's easier) the contents of the `.umockdev` file could be the properties and attributes sent with the add-event.

@martinpitt: Does this sound like a good approach?

bobhenz-jabil, 2024-04-24:
Great! I will try to come up with a branch that implements this using the `uevent` file over the next week.

bobhenz-jabil, 2024-04-24:
Actually I was able to get some time to work on this today (https://github.com/martinpitt/umockdev/pull/238). I am not familiar with .vala so please excuse any non-valaic ways of doing things. The following _does_ seem to work though (under the minimal testing I ran).

martinpitt, 2024-10-03:
This was fixed in #238

closed 2024-10-03 by martinpitt