martinpitt/umockdev #248: (Question) Redox OS support

hardBSDk, 2024-06-03:
Your project is amazing!!

Do you if it's technically possible to implement something similar on [Redox OS](https://redox-os.org) ?

Probably your project is Linux-specific, but can you give me the steps to replicate this on Redox OS?

martinpitt, 2024-06-03:
I'm afraid I have absolutely no idea about RedoxOS -- /proc and /dev are still somewhat covered by POSIX, but /sys is Linux specific I'm afraid. I can't give you detailled steps, other than: learn how Redox OS manages hardware devices (the moral equivalents of /sys, udev, and kernel uevents), how applications use them (API for enumeration and changes), and how to write a preload library to simulate all of that.

Perhaps RedoxOS uses the same APIs as Linux, then there's a chance that umockdev will actually work. But otherwise this is an entirely new project.

hardBSDk, 2024-06-03:
@martinpitt Thanks a lot for your answer!

closed 2024-06-03 by hardBSDk