martinpitt/umockdev #260: 0.19.0 - not compile with Clang: undefined symbol: __readlinkat_alias

AngryPenguinPL, 2024-12-28:
I'm trying to compile umockdev 0.19.0 using Clang 19.1.6 on OpenMandriva Cooker and I'm getting the compilation error below.
It's worth noting that previous versions of umockdev 0.18.x compile correctly with Clang.
Version 0.19.0 also compiles correctly with GCC.

```
 [38/55] /usr/bin/cc  -o libumockdev-preload.so.0.0.0 libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o libumockdev-preload.so.0.0.0.p/src_debug.c.o libumockdev-preload.so.0.0.0.p/src_utils.c.o libumockdev-preload.so.0.0.0.p/src_ioctl_tree.c.o -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,-soname,libumockdev-preload.so.0 -Os -fomit-frame-pointer -g3 -gdwarf-4 -Wstrict-aliasing=2 -pipe -Wformat -Werror=format-security -fstack-protector-all --param=ssp-buffer-size=4 -m64 -mtune=generic -flto -Wl,-O2 -Wl,--no-undefined -flto -Os -fomit-frame-pointer -g3 -gdwarf-4 -Wstrict-aliasing=2 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fstack-protector-all --param=ssp-buffer-size=4 -m64 -mtune=generic -flto -Wl,--start-group -ldl -lpthread -Wl,--end-group
DEBUG util.py:463:  FAILED: libumockdev-preload.so.0.0.0 
DEBUG util.py:463:  /usr/bin/cc  -o libumockdev-preload.so.0.0.0 libumockdev-preload.so.0.0.0.p/src_libumockdev-preload.c.o libumockdev-preload.so.0.0.0.p/src_debug.c.o libumockdev-preload.so.0.0.0.p/src_utils.c.o libumockdev-preload.so.0.0.0.p/src_ioctl_tree.c.o -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,-soname,libumockdev-preload.so.0 -Os -fomit-frame-pointer -g3 -gdwarf-4 -Wstrict-aliasing=2 -pipe -Wformat -Werror=format-security -fstack-protector-all --param=ssp-buffer-size=4 -m64 -mtune=generic -flto -Wl,-O2 -Wl,--no-undefined -flto -Os -fomit-frame-pointer -g3 -gdwarf-4 -Wstrict-aliasing=2 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fstack-protector-all --param=ssp-buffer-size=4 -m64 -mtune=generic -flto -Wl,--start-group -ldl -lpthread -Wl,--end-group
DEBUG util.py:463:  ld.lld: error: undefined symbol: __readlinkat_alias
DEBUG util.py:463:  >>> referenced by unistd.h:104 (/usr/include/bits/unistd.h:104)
DEBUG util.py:463:  >>>               libumockdev-preload.so.0.0.0.lto.o:(__readlinkat_chk)
DEBUG util.py:463:  cc: error: linker command failed with exit code 1 (use -v to see invocation)
```

martinpitt, 2024-12-28:
Hello @AngryPenguinPL ! To bisect, the last commit that touched the preload lib was commit 284dcf1f6948645f61b75214c3d934912ff722bc , but that is already in 0.18.4 and really didn't change anything that would touch include files or readlinkat(). The previous one was commit 8cb6255eb858bf3f33ab72e9456a3bd77abde3d5 which was a bit more intrusive, but also not related to readlink, and already in 0.18.3 (from May).

So I take it that 0.18.3 or .4 also fail to build in your environment now, and it's due to some change in the build env, not a code change? (Note that there of course could still be a bug, but it's not a recent regression then)

AngryPenguinPL, 2024-12-29:
I re-checked the older version and it does indeed not compile. You are right.

The question remains, what is to blame? Clang 19.1.6?

martinpitt, 2024-12-30:
Thanks for confirming! It's not really a matter of "blaming", it's probably a combination of clang and glibc changes that the preload doesn't get along with -- I haven't yet encountered `__readlinkat_alias()`.

I tested a build on Fedora 41 x86_64 with clang-19.1.5-1.fc41.x86_64 and glibc-2.40-12.fc41.x86_64 and that works. So I can't easily reproduce/investigate this.

martinpitt, 2025-01-05:
[Debian has clang 19.1.6](https://tracker.debian.org/pkg/llvm-toolchain-19) (Fedora 41 only has 19.1.5). But `CC=clang meson setup c && meson test -C c` also works. This is also glibc 2.40, similar to F41.

What glibc version are you using? Do you have a reproducer in a container?

RossComputerGuy, 2025-01-19:
This is reproducible with Nixpkgs / NixOS via running `nix build github:NixOS/nixpkgs#pkgsLLVM.umockdev`. We've got glibc-2.40.