martinpitt/umockdev #267: Better handling of udev_device_get_is_initialized

benzea, 2025-03-20:
So, device are initialised if they have (valid?) entries in `/run/udev/data/$ID` or if a uevent was received. Now, umockdev never creates entries in `/run/udev/data/$ID`, wo device are considered "not initialised" unless the uevent was processed by the client.

On the one hand, I can abuse this for tests in libusb (though maybe I should not …):
  https://github.com/libusb/libusb/pull/1625

On the other hand, this is just broken. umockdev should probably just write out the database file when the ("add") uevent is fired. But, it would be useful if it then also added API to prevent the uevent from being fired so that one can test the enumeration corner case.

martinpitt, 2025-03-26:
> it would be useful if it then also added API to prevent the uevent from being fired so that one can test the enumeration corner case.

I was thinking of `add_devicev_no_uevent()`, but that is internal indeed. We could make this public, or add a new `add_device_options()` or so, or perhaps look at a `$UMOCKDEV_DIR/no-uevent` inhibitor file.

But OOI, this does feel like an abuse, as you can't do this in real life -- if you want to test the static coldplug enumeration, the test should set up /sys first and then start the lib/app?

benzea, 2025-03-26:
Well, there are two parts, right? We should write the `/run/udev/data/` entries so that coldplug works correctly.

The other part would be whether we should permit testing the coldplug race condition. Maybe it is overkill to support that.